esphome/issues #4252: Checking the state of current_operation for time-based cover doesn't work

Plawasan, 2023-02-26:
### The problem

I'm trying to create a time-based cover and then do some rudimentary checking of the cover's operation in a stop script, so that I can either send the reverse command when it's moving or use a double click when the cover is already stopped to tell the controller to send it to a pre-set position.

Complete yaml below, here are the relevant bits.

 I've set up a time-based cover:

```
cover:
  - platform: time_based
    id: tb_cover
    name: "LR Side Cover"
    has_built_in_endstop: true
    open_duration: 20s    
    close_duration: 10s

    open_action:
      - button.press: button_single_up

    close_action:
      - button.press: button_single_down

    stop_action:
      - script.execute: stop
```
and the appropriate script to handle the stop action:

```
script: 
 - id: stop
    then:
      - if:
          condition:
            lambda: |-
              if (id(tb_cover).current_operation == CoverOperation::COVER_OPERATION_IDLE) {
                return true;
              } else if (id(tb_cover).current_operation == CoverOperation::COVER_OPERATION_OPENING) {
                return false;
              } else if (id(tb_cover).current_operation == CoverOperation::COVER_OPERATION_CLOSING) {
                return false;
              }            
          then:
            - button.press: button_double_down
          else:
            - button.press: button_single_down
```
The logic obviously isn't complete however I'm stuck already at distinguishing between any of the potential current_operation states, the code above (taken directly from ESPHome docs) always returns true and therefore always executes the double_down button.

Here's what that looks like in logs:

```
[20:01:08][D][cover:170]: 'LR Side Cover' - Publishing:
[20:01:08][D][cover:173]:   Position: 25%
[20:01:08][D][cover:186]:   Current Operation: OPENING
[20:01:08][D][switch:017]: 'ESP Cover LR Side Up' Turning OFF.
[20:01:08][D][switch:056]: 'ESP Cover LR Side Up': Sending state OFF
[20:01:09][D][cover:170]: 'LR Side Cover' - Publishing:
[20:01:09][D][cover:173]:   Position: 30%
[20:01:09][D][cover:186]:   Current Operation: OPENING   <------- Cover is opening
[20:01:10][D][cover:076]: 'LR Side Cover' - Setting
[20:01:10][D][cover:080]:   Command: STOP                      <------- Pushed STOP in the web controls
[20:01:10][D][button:013]: 'ESP Cover LR Side Double Down' Pressed.        <----------- Double click is pressed - should be single
```

It's entirely possible I'm missing something completely obvious... 

Thanks!

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:
  esp_name: ESP Cover LR Side #Device Name
  esp_hostname: esp-cover-lr-side
  push_delay: 80ms
  ip_address: 192.168.5.105
  ip_gw: 192.168.5.1
  ip_netmask: 255.255.255.0

esphome:
  name: ${esp_hostname}

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: ${ip_address}
    gateway: ${ip_gw}
    subnet: ${ip_netmask}
  ap:
    ssid: "${esp_name} Hotspot"
    password: !secret ap_password

ota:
  password: !secret ota_password

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

captive_portal:

web_server:
  port: 80
  include_internal: true 

switch:
  - platform: gpio
    id: switch_up
    pin: 25
    name: "${esp_name} Up"
    internal: true  

  - platform: gpio
    id: switch_down
    pin: 26
    name: "${esp_name} Down"
    internal: true
 
binary_sensor:
  - platform: gpio
    id: ext_button_up
    pin: 
      number: 27
      mode:
        input: true
        pulldown: true 
    name: "${esp_name} Ext Up"
    on_press:
      then:
        - switch.turn_on: switch_up
    on_release:
      then:
        - switch.turn_off: switch_up

  - platform: gpio
    id: ext_button_down
    pin: 
      number: 14
      mode:
        input: true
        pulldown: true 
    name: "${esp_name} Ext Down"
    on_press:
      then:
        - switch.turn_on: switch_down
    on_release:
      then:
        - switch.turn_off: switch_down

script:
  - id: single_down
    mode: queued
    then:
      - switch.turn_on: switch_down
      - delay: ${push_delay}
      - switch.turn_off: switch_down
      - delay: ${push_delay}      

  - id: single_up
    mode: queued
    then:
      - switch.turn_on: switch_up
      - delay: ${push_delay}
      - switch.turn_off: switch_up
      - delay: ${push_delay}    

  - id: stop
    then:
      - if:
          condition:
            lambda: |-
              if (id(tb_cover).current_operation == CoverOperation::COVER_OPERATION_IDLE) {
                return true;
              } else if (id(tb_cover).current_operation == CoverOperation::COVER_OPERATION_OPENING) {
                return false;
              } else if (id(tb_cover).current_operation == CoverOperation::COVER_OPERATION_CLOSING) {
                return false;
              }            
          then:
            - button.press: button_double_down
          else:
            - button.press: button_single_down
 
button:
  - platform: template
    name: "${esp_name} Single Down" 
    id: button_single_down
    on_press:
      - script.execute: single_down

  - platform: template
    name: "${esp_name} Double Down" 
    id: button_double_down
    on_press:
      - repeat:
          count: 2
          then:
            - script.execute: single_down

  - platform: template
    name: "${esp_name} Triple Down" 
    id: button_triple_down
    on_press:
      - repeat:
          count: 3
          then:
            - script.execute: single_down

  - platform: template
    name: "${esp_name} Quad Down" 
    id: button_quad_down
    on_press:
      - repeat:
          count: 4
          then:
            - script.execute: single_down   

  - platform: template
    name: "${esp_name} Single Up" 
    id: button_single_up
    on_press:
      - script.execute: single_up

  - platform: template
    name: "${esp_name} Double Up" 
    id: button_double_up
    on_press:
      - repeat:
          count: 2
          then:
            - script.execute: single_up  

  
cover:
  - platform: time_based
    id: tb_cover
    name: "LR Side Cover"
    has_built_in_endstop: true
    open_duration: 20s    
    close_duration: 10s

    open_action:
      - button.press: button_single_up

    close_action:
      - button.press: button_single_down

    stop_action:
      - script.execute: stop
```

Plawasan, 2023-02-27:
For the record: as I've learned from Jesse by the time the script is executed current_operation is already set to IDLE. I've tried to move the logic of the script into the definition of the cover, made no difference.

If the current_operation value is set BEFORE executing the stop_action part of the definition then is there a different/better way to implement the conditional logic of the stop switch?

xprofiler, 2024-01-12:
I have the same problem. My Warema cover controller has two switches. When I switch UP, then the cover is moving upwards. To stop the cover I have to switch down ones. Another switch down let the cover moving downwards. So when I issue a STOP command, then I have to know if the cover is moving up or down to send the corresponding counter command.
I have reviewed the source code. When I understand it correctly, the current_operation is set before the stop_action is called, so we cannot use the state, but there is already a property called "last_operation_", see:
https://github.com/esphome/esphome/blob/4cc17dac0dff22ef476aa7b98abf17b675749379/esphome/components/time_based/time_based_cover.h#L50C8-L50C8
This is holding the information we need, but it is currently a protected property, so it cannot be used in a lamda call. Maybe it would be sufficient to have a pulic property to access the last operation in the lamda statement.
```yaml
cover:
  - platform: time_based
    name: "Sunblind AZ"
    id: sunblind_az
    open_action:
      - button.press: az_h
    open_duration: 34s
    close_action:
      - button.press: az_r
    close_duration: 32.5s
    stop_action: 
      - lambda: !lambda |-
          if (id(sunblind_az).last_operation == CoverOperation::COVER_OPERATION_OPENING) {
            // Cover is currently opening
            id(az_r).press();
          } else if (id(sunblind_az).last_operation == CoverOperation::COVER_OPERATION_CLOSING) {
            // Cover is currently closing
            id(az_h).press();
          }
    has_built_in_endstop: true
```
Maybe I could create a pull request for this small change in the source?

Plawasan, 2024-01-12:
I would definitely appreciate it, I had to resort to a somewhat ugly solution where I run a check every second to update the status of the cover in a separate variable which I then check for status, which I'm sure is a horrible way to solve this but it works...

```
globals:
   - id: cover_status
     type: int
     restore_value: no
     initial_value: '0' # 0-IDLE, 1-OPENING, 2-CLOSING

interval:
  - interval: 1s
    then:
      - script.execute: update_cover_status

script:
  - id: update_cover_status
    then:
      - lambda: |-
          if (id(${esp_entity_id}).current_operation == CoverOperation::COVER_OPERATION_IDLE) {
            id(cover_status) = 0;
          } else if (id(${esp_entity_id}).current_operation == CoverOperation::COVER_OPERATION_OPENING) {
            id(cover_status) = 1;
          } else if (id(${esp_entity_id}).current_operation == CoverOperation::COVER_OPERATION_CLOSING) {
            id(cover_status) = 2;
          } 
```

I honestly need to rewrite the whole thing but it works, so it goes on the "when I have the time" list..

closed 2024-04-25 by jesserockz

Plawasan, 2024-05-10:
@xprofiler - thanks for implementing this, I tried to use get_last_operation() in my cover control code, it works well when the cover transitions from moving to stopped however I'm hitting an issue where if the cover is stopped and you hit stop again - see below:

Cover is stopped/fully open, then to closed (via HA), then I hit stop (debug is output from my script that simply logs the status of current and last position):

```
[16:54:53][W][main:235]: >>>>>>>>>>>>>> CURRENT OPERATION: IDLE
[16:54:53][W][main:279]: >>>>>>>>>>>>>> LAST OPERATION: OPENING
[16:54:55][D][cover:076]: 'Cover Office' - Setting
[16:54:55][D][cover:084]:   Position: 0%
[16:54:55][D][button:010]: 'Cover Office Single Down' Pressed.        <---------------- Down pressed in HA
[16:54:55][W][main:1070]: >>>>>>>>>>>>> Button DOWN pushed
[16:54:55][W][main:257]: >>>>>>>>>>>>>> CURRENT OPERATION: CLOSING
[16:54:55][W][main:290]: >>>>>>>>>>>>>> LAST OPERATION: CLOSING
[16:54:55][W][main:351]: >>>>>>>>>>>>>> Running Script Single Down
[16:54:55][D][switch:012]: 'Cover Office Down' Turning ON.
[16:54:55][D][switch:055]: 'Cover Office Down': Sending state ON
[16:54:55][D][cover:170]: 'Cover Office' - Publishing:
[16:54:55][D][cover:173]:   Position: 100%
[16:54:55][D][cover:186]:   Current Operation: CLOSING
[16:54:55][D][switch:016]: 'Cover Office Down' Turning OFF.
[16:54:55][D][switch:055]: 'Cover Office Down': Sending state OFF
[16:54:56][D][cover:170]: 'Cover Office' - Publishing:
[16:54:56][D][cover:173]:   Position: 97%
[16:54:56][D][cover:186]:   Current Operation: CLOSING
[16:54:57][D][cover:170]: 'Cover Office' - Publishing:
[16:54:57][D][cover:173]:   Position: 94%
[16:54:57][D][cover:186]:   Current Operation: CLOSING
[16:54:58][D][cover:170]: 'Cover Office' - Publishing:
[16:54:58][D][cover:173]:   Position: 91%
[16:54:58][D][cover:186]:   Current Operation: CLOSING
[16:55:01][D][cover:080]:   Command: STOP                                <---------------- STOP pressed in HA
[16:55:01][W][main:235]: >>>>>>>>>>>>>> CURRENT OPERATION: IDLE
[16:55:01][W][main:290]: >>>>>>>>>>>>>> LAST OPERATION: CLOSING
[16:55:01][D][button:010]: 'Cover Office Single Up' Pressed.
[16:55:01][W][main:1201]: >>>>>>>>>>>>>>>>>> Button UP pushed
[16:55:01][W][main:336]: >>>>>>>>>>>>>> Running Script Single Up
[16:55:01][D][switch:012]: 'Cover Office Up' Turning ON.
[16:55:01][D][switch:055]: 'Cover Office Up': Sending state ON
[16:55:01][D][cover:170]: 'Cover Office' - Publishing:
[16:55:01][D][cover:173]:   Position: 82%
[16:55:01][D][cover:186]:   Current Operation: IDLE
[16:56:43][W][main:235]: >>>>>>>>>>>>>> CURRENT OPERATION: IDLE
[16:56:43][W][main:290]: >>>>>>>>>>>>>> LAST OPERATION: CLOSING
```
OK so now I'm in a state where Current Operation is set to IDLE and last is CLOSING. Now I need to check in my stop script what was the cover doing, before I hit stop again (basically the same issue I originally raised) - I would therefore expect that BOTH current and LAST operation will be set to IDLE (since I've stopped the cover twice now) but that is not the case, LAST is never updated to IDLE.

I noticed that setting last position for IDLE is missing in void TimeBasedCover::start_direction_(CoverOperation dir) (https://github.com/esphome/esphome/tree/dev/esphome/components/time_based/time_based_cover.cpp):

```
  switch (dir) {
    case COVER_OPERATION_IDLE:
      trig = this->stop_trigger_;
      break;
    case COVER_OPERATION_OPENING:
      this->last_operation_ = dir;
      trig = this->open_trigger_;
      break;
    case COVER_OPERATION_CLOSING:
      this->last_operation_ = dir;
      trig = this->close_trigger_;
      break;
    default:
      return;
```

but judging whether that's what's actually needed here is far beyond my paygrade..

Thanks!

mion00, 2024-05-26:
I encountered a similar issue: my cover has a built-in control unit, whose operation is via two push buttons, one for "up" and the other for "down" directions.
If the cover is moving "up", pressing "up" again stops it, so if the cover is stopped I must avoid executing the `stop_action` (that would start a movement again).

I don't know anything about the source code, but after a quick check, `last_operation_` was added in [#1809](https://github.com/esphome/esphome/pull/1809) to implement a "toggle" functionality, and internally it is only used in the following line:
https://github.com/esphome/esphome/blob/dev/esphome/components/time_based/time_based_cover.cpp#L70

Looking at the code however I think it is a misnomer: as of now it represents the __last direction of travel__, used by the `toggle` action to know if its was previously _opening_ or _closing_.

I have a few strategies to solve this conundrum:
- adding the missing switch-case outlined above so that the `last_operation_` attribute accurately represents the history of the cover, as the name implies, and we can build logic on top of it to avoid triggering on repeated stop actions
- refactor the `toggle` code to use an additional (private?) attribute for its own logical reasoning (named `last_moving_operation`?) that takes the same meaning of the current  `last_operation_`, to avoid breaking its logic.

OR:
- introduce a configuration option to the timed_cover that disables calling the `stop_action` if the state is currently IDLE (this way the logic is built-in the core, without needing additional conditions built with lambdas or scripts) -> I don't fully know the ramification of this change with the rest of the platform, and if it might introduce unexpected results.

xprofiler, 2024-05-27:
@Plawasan I can confirm the Issue, when I double tap on the STOP button, than at first the blind stops and with the second tap it is moving in the opposit direction. I have to debug this to check why this is happening, but I think @mion00 is pointing in the right direction.
Currently I have no time budget left for this. If I find out the problem, I will create a pull request.

saibotma, 2024-05-30:
I think this issue should be reopened, then.

mion00, 2024-06-01:
After quite some time to understand how the ESPHome code works, I opened a PR with a fix, opting to follow the first strategy I outlined above.
It works locally on my ESP32 board.

If anyone is interested in testing it, just add the following to your configuration file:

```yaml
# Use fixed timed_cover from Github PR
external_components:
  source: github://pr#6847
  components: [time_based]
```

Now the `stop_action` can correctly identify if it was invoked a second time.
```
16:22:53][D][cover:076]: 'Tapparella' - Setting
[16:22:53][D][cover:080]:   Command: STOP
[16:22:53][D][blinds_stop:291]: Stop request, current_operation: 0, last_operation: 1
[16:22:53][D][blinds_stop:132]: Pressing the reverse button to stop
[16:22:53][D][switch:012]: 'switch_close' Turning ON.
[16:22:53][D][switch:055]: 'switch_close': Sending state ON
[16:22:53][D][cover:170]: 'Tapparella' - Publishing:
[16:22:53][D][cover:173]:   Position: 89%
[16:22:53][D][cover:186]:   Current Operation: IDLE
[16:22:53][D][switch:016]: 'switch_close' Turning OFF.
[16:22:53][D][switch:055]: 'switch_close': Sending state OFF
[16:22:57][D][cover:076]: 'Tapparella' - Setting
[16:22:57][D][cover:080]:   Command: STOP
[16:22:57][D][blinds_stop:291]: Stop request, current_operation: 0, last_operation: 0
[16:22:57][D][blinds_stop:128]: Cover is already stopped!
```

Plawasan, 2024-06-05:
Finally got a chance to test this today, I still need to update the control logic for my blinds but "your part" now works as expected, thanks!

xprofiler, 2024-06-10:
I can confirm that it also works with my warema controller. Good work @mion00

saibotma, 2024-06-27:
The PR @mion00 is talking about: https://github.com/esphome/esphome/pull/6847

Plawasan, 2024-08-21:
Any idea what's keeping this from being merged? The workaround with using the update external component still works even with 2024.8.0 but I'm wary of every major upgrade rendering it unusable (I admit I have a problem of updating even things that work as expected :)

scaytrase, 2024-08-22:
Got here while trying to setting up lambdas for my cover. Implementing this makes statement that feedback cover is drop-in replacement for time_based cover invalid (which really confused me)