esphome/issues #4255: esp_spiram_get_size() not getting the correct PSRAM size

AndreasBVB, 2023-02-28:
### The problem

`  - platform: template
    name: PSRAM SIZE
    lambda: return esp_spiram_get_size()/1024/1024;
    entity_category: diagnostic
    unit_of_measurement: MB
    update_interval: 30s`

Not sure why it mostly reports 8MB, even when the WROVER module is 4MB or 16MB. Only on a ESP32 CAM knockoff it correctly reported 2MB as expected.

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

WROVER-E

### Component causing the issue

camera

pmatil, 2024-12-26:
I have a N8R2 board which has only 2 MB of PSRAM. ESPHome says the board has 8 MB of it when initializing display component. With esp-idf platform, PSRAM is not detected at all.

I can't get a full boot log for some reason. It stops after this (it does boot just fine and ESPHome loads succesfully):

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x28dc
entry 0x403c98c0
```

YAML:
```
esphome:
  name: esphome-web-d6bff8
  friendly_name: ESP32-S3 N8R2
  min_version: 2024.11.0
  name_add_mac_suffix: false
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
    board_build.flash_mode: dio
    board_upload.maximum_ram_size: 524288
  project: 
    name: esphome.web
    version: dev

# Enable PSRAM (above) with N8R2 version of ESP32-WROOM-1 which has 8 MB of flash and 2 MB of PSRAM
esp32:
  board: esp32-s3-devkitc-1
  variant: ESP32S3
  framework:
    type: arduino
  flash_size: 8MB

psram:
  mode: quad
  speed: 40MHz

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

spi:
  clk_pin: GPIO12
  mosi_pin: GPIO11
  miso_pin: GPIO13

display:
  - platform: ili9xxx
    model: ili9341
    dc_pin: GPIO14
    reset_pin: GPIO21
    cs_pin: GPIO10
    auto_clear_enabled: False
    invert_colors: false
    show_test_card: false

lvgl:
  pages:
    - id: main_page
      widgets:
        - label:
            align: CENTER
            text: 'Hello World!'
```

Boot log:
```
[I][app:102]: Project esphome.web version dev
[C][logger:185]: Logger:
[C][logger:186]:   Level: DEBUG
[C][logger:188]:   Log Baud Rate: 115200
[C][logger:189]:   Hardware UART: USB_CDC
[C][spi:064]: SPI bus:
[C][spi:065]:   CLK Pin: GPIO12
[C][spi:066]:   SDI Pin: GPIO13
[C][spi:067]:   SDO Pin: GPIO11
[C][spi:072]:   Using HW SPI: SPI
[C][ili9xxx:094]: ili9xxx
[C][ili9xxx:094]:   Rotations: 0 °
[C][ili9xxx:094]:   Dimensions: 240px x 320px
[C][ili9xxx:095]:   Width Offset: 0
[C][ili9xxx:096]:   Height Offset: 0
[C][ili9xxx:102]:   Color mode: 16bit
[C][ili9xxx:111]:   Data rate: 40MHz
[C][ili[C][ili9xxx:114]:   CS Pin: GPIO10
[C][ili9xx[C][ili9xxx:120]:   Mirror_y: NO
[C][ili9xxx:121]:   Invert colors: NO
[C][ili9xxx:126]:   Update Interval: 1.0s
[C][psram:020]: PSRAM:
[C][psram:021]:   Available: YES
[C][psram:024]:   Size: 8191 KB
[C][lvgl:086]: LVGL:
[C][lvgl:087]:   Display width/height: 240 x 320
[C][lvgl:088]:   Rotation: 0
[C][lvgl:089]:   Draw rounding: 2
[C][captive_portal:089]: Captive Portal:
[C][mdns:116]: mDNS:
[C][mdns:117]:   Hostname: esphome-web-d6bff8
[C][safe_mode:018]: Safe Mode:
[C][safe_mode:020]:   Boot considered successful after 60 seconds
[C][safe_mode:021]:   Invoke after 10 boot attempts
[C][safe_mode:023]:   Remain in safe mode for 300 seconds
[C][improv_serial:032]: Improv Serial:
[I][safe_mode:041]: Boot seems successful; resetting boot loop counter
```

clydebarrow, 2025-01-09:
> I have a N8R2 board which has only 2 MB of PSRAM.

Maybe it's not actually an N8R2 board - if it reports 8MB of PSRAM then it will likely be octal, not quad. Try that and see what happens.

> I can't get a full boot log for some reason.

Are you using the USB port or UART0? S3's default to `hardware_uart: USB_SERIAL_JTAG` but if you are using UART0 then you will get exactly the log symptoms you describe.

It would be helpful to say exactly what board it is, rather than just saying what module is on it.

pmatil, 2025-01-09:
> Maybe it's not actually an N8R2 board - if it reports 8MB of PSRAM then it will likely be octal, not quad. Try that and see what happens.
> 
It may be. It actually works with octal. I've tested this with a test program that dumps the free memory info to console and according to that the board has 8 MB. So it seems to be mislabeled. Lucky me. I guess if it reports 8 MB it actually has it?

> 
> Are you using the USB port or UART0? S3's default to `hardware_uart: USB_SERIAL_JTAG` but if you are using UART0 then you will get exactly the log symptoms you describe.
> 
Yes for "prepare for first use" I used the USB port and that works but then provisioning will fail and I have to switch to the other port. Even with that it seemed to be hit and miss but will eventually succeed. Default config will be incomplete though (using esphome-web because of reasons) maybe because of the wrong port used initially. Even if using the JTAG port I can't get more boot log than that.

I'm planning to make a post about these boards and my findings in the PSRAM thread on HA community as this is offtopic for this thread.

FYI the board is ESP32-S3 DevkitC from VCC-GND (or a clone): https://github.com/vcc-gnd/YD-ESP32-S3 Available on AE.

clydebarrow, 2025-01-09:
If it works with octal PSRAM it's definitely not 2MB PSRAM, they are quad only. Check the label on the module - see below.

I'm not sure what you mean by "provisioning" - if that's referring to configuring the WiFi via Improv, get rid of that and hard-code your WiFi credentials via a secrets file. The boilerplate config installed by "first use" is not much use for anything beyond exactly that.

[image]

closed 2025-01-09 by clydebarrow

pmatil, 2025-01-09:
> If it works with octal PSRAM it's definitely not 2MB PSRAM, they are quad only. Check the label on the module - see below.
> 
Yes the module is labeled N8R2 so that is why it's a bit confusing. But better this way, I got more PSRAM that I paid for.