esphome/issues #4258: Modbus write number fails

17844, 2023-02-28:
### The problem

Trying to write to modbus device fails

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Example YAML snippet

```yaml
substitutions:
  systemName: "IVT"

esphome:
  name: rs485
  platform: ESP8266
  board: nodemcu

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  id: mod_bus
  tx_pin: D3
  rx_pin: D4
  baud_rate: 9600
  parity: even

modbus:
  send_wait_time: 1000ms
#  flow_control_pin: 5
  id: modbus1

modbus_controller:
  - id: ivt
    ## the Modbus device addr
    address: 2
    modbus_id: modbus1
    setup_priority: -10
    command_throttle: 20ms
  - id: ivtpower
    ## the Modbus device addr
    address: 0x1
[... 577 lines cut ...]
[17:32:09][D][main:290]: Modbus Number incoming value = 2.000000
[17:32:09][V][modbus.number:039]: Value overwritten by lambda
[17:32:09][V][modbus.number:050]: Modbus Number write raw: 0014
[17:32:09][D][number:012]: 'Sätt Värmekurva': Sending state 2.000000
[17:32:09][V][modbus_controller:035]: Sending next modbus command to device 2 register 0x00 count 0
[17:32:09][V][modbus:220]: Modbus write raw: 00.14
[17:32:09][V][modbus_controller:486]: Command sent 0 0x0 0
[17:32:09][V][modbus_controller:035]: Sending next modbus command to device 2 register 0x00 count 0
[17:32:09][V][modbus:220]: Modbus write raw: 00.14
[17:32:09][V][modbus_controller:486]: Command sent 0 0x0 0
[17:32:09][V][modbus_controller:035]: Sending next modbus command to device 2 register 0x00 count 0
[17:32:09][V][modbus:220]: Modbus write raw: 00.14
[17:32:09][V][modbus_controller:486]: Command sent 0 0x0 0
[17:32:09][V][modbus_controller:035]: Sending next modbus command to device 2 register 0x00 count 0
[17:32:09][V][modbus:220]: Modbus write raw: 00.14
[17:32:09][V][modbus_controller:486]: Command sent 0 0x0 0
[17:32:09][V][modbus_controller:035]: Sending next modbus command to device 2 register 0x00 count 0
[17:32:09][V][modbus:220]: Modbus write raw: 00.14
[17:32:09][V][modbus_controller:486]: Command sent 0 0x0 0
[17:32:09][D][modbus_controller:029]: Modbus command to device=2 register=0x00 countdown=0 no response received - removed from send queue
```

### Additional information

When trying to write a value to number "Sätt Värmekurva", it fails. See last part of log. No Response. reading data from same register is no problem. It has worked in earlier versions but I am not sure when it stopped working.

erlendmlhub, 2023-04-08:
I am stuck in same position as you when trying to write a value to a number. Did you find it out?

cfr-mir, 2023-12-11:
I have been stuck there for a while. Documentation might be better on this particular topic.
After a bit of browsing the c++ code for modbus controller. It seems that you can do three things, but only one thing at a time.

1. Return the value you want to write. DO NOT push anything to "payload".
2. push the entire raw modbus frame to the payload vector and return whetever comes into mind, like 42.
3. If you don't need to modify the number, when writing.Leave out the "write_lambda:" section. Just declare register_type: as "holding".

This is what I read in the code. I didn't have the time to test it yet.

The example provided in the documentation, will definitely never hit any Modbus target. However, the text above the example, actually does give a decent overview.

I will try it out later this week. If I am wrong, I will write another commant.