esphome/issues #4261: Update fails with ld terminated with signal 9

creativetags, 2023-03-01:
### The problem

I upgraded ESPHome from 2022.2.6 to 2023.2.4 and now it fails to update or install to my ESP32s. 

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

esp32dev

### Anything in the logs that might be useful for us?

```txt
Linking /data/elf/.pioenvs/elf/firmware.elf
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
```

ssieb, 2023-03-01:
That's almost certainly running out of RAM.  What are you running esphome on?

creativetags, 2023-03-02:
Thanks, I'm running Home Assistant on a Raspberry Pi 3B with 1GB RAM. Is that no longer supported?

creativetags, 2023-03-06:
Seems unfortunate when it worked with earlier versions and the Pi 3b is officially supported by Home Assistant.

ssieb, 2023-03-06:
Home Assistant has much lower RAM requirements than a compiler and linker.

closed 2023-03-07 by creativetags

geoffrey-vl, 2023-06-13:
same issue here. Is there a list of minimal hardware specs?

croessi, 2023-08-21:
I had the same issue and solved it by increasing the swap with this script, which can be executed in the terminal addon - however this will be reverted after restart:

```shell
if [[ ! -f /backup/_swap.swap ]]
then 
	fallocate -l 2G ./_swap.swap
	mkswap ./_swap.swap
	chmod 0600 ./_swap.swap
	swapon ./_swap.swap
elif [[ ! $(dmesg|grep _swap.swap) = *swap\ on* ]]
then
	swapon /backup/_swap.swap; 
fi
``` 
You could add this script via the system.d addon to execute it every time the sytem is started.
Additionally I recommend to monitor memory usage with these additional sensors:

```
- platform: systemmonitor
  resources:
    - type: disk_use_percent
      arg: /config
    - type: memory_use_percent
    - type: swap_use_percent
```

fsiek, 2023-09-11:
Sad to see, that only power hungy Raspis are supported. Would recommend to disable the installation of EspHome on non Raspi4 systems or show a warning about the short commings.

grumete, 2024-01-09:
Utilizar un pc con esphome en linea de comandos y la dashboard. Es más sencillo. Y dejemos a la pobre raspy para que ejecute el programa ya compilado.
O por lo menos es eso lo que yo haré.
Estaba asustado por que no pudiera el esp32 con un miniprogramilla.