esphome/issues #4264: Compile error with 2023.2.4 on HA for platform ESP32 with fatal error: sdkconfig.h: No such file or directory

lambu76, 2023-03-02:
### The problem

Compile fails on platform ESP32 by using HomeAssistant add on v. 2022.3.0

/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.0

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: prova
  platform: ESP32
  board: nodemcu-32s
  
ota:

captive_portal:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0

# Enable Home Assistant API
api:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/prova.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing prova (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
Compiling /data/prova/.pioenvs/prova/src/esphome/components/api/api_connection.cpp.o
Compiling /data/prova/.pioenvs/prova/src/esphome/components/api/api_frame_helper.cpp.o
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/wchar.h:6,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/string:40,
                 from src/esphome/components/socket/socket.h:2,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/prova/.pioenvs/prova/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/wchar.h:6,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/cwchar:44,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/postypes.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/char_traits.h:40,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/string:40,
                 from src/esphome/components/socket/socket.h:2,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/prova/.pioenvs/prova/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 4.44 seconds ==========================
```

### Additional information

YAML reported is a very easy example to reproduce the problem I have.

Problem is present since release ESPHome 2023.x.x of the add-on

lambu76, 2023-03-02:
Already tried mutliple time "Clean Build Files" option

randybb, 2023-03-02:
Remove integration and add it back. Your yaml files are stored outside of the docker container (addon) so you don't need to worry about loosing configs.

lambu76, 2023-03-02:
confirmed....by removing the addon and readding the problem is solved.

YAML file not lost and once re-added the addon all items are already inside...ready to be compiled succesfully.

thx

tracestep, 2023-03-02:
Uninstalling and reinstalling the addon did not work for me, I still have the same error. Cleaned build files and uninstalled twice.

tracestep, 2023-03-03:
Solved, i had a specific (probably older) `platform_version`  specified in my YAML file under `esp32/framework`. Removing the line (after I rad removed and reinstalled the addon) fixed the error.