esphome/issues #4265: restore_value not working on ESP32

jeroenrnl, 2023-03-02:
### The problem

I have an ESP32 to keep track of an S0 energy meter. Because this is a pulse-based meter, I want the ESP to track the totals. 

I have set it up to store the raw counter in flash and restore it during boot. But it seems to lose it every time it is rebooted.

### Which version of ESPHome has the issue?

v2023.2.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Example YAML snippet

```yaml
esphome:
  name: laadpaal-dev
  on_boot:
    priority: 700   # I have tried various values between -10 and 800
    then:
      - pulse_counter.set_total_pulses:
          id: dev_counter
          value: !lambda return id(dev_total_pulses);

globals:
  - id: dev_total_pulses
    type: int
    restore_value: yes
esp32:
  board: nodemcu-32s
  framework:
    type: arduino

preferences:
  flash_write_interval: 60s
sensor:
  - platform: pulse_counter
    use_pcnt: True
    pin: 5
    icon: "mdi:flash"
    name: "DEV Laadpaal Raw counter"
    id: dev_counter
    total:
      id: dev_counter_total
      name: 'DEV Laadpaal Raw Total'
    on_value:
      then:
        - globals.set:
            id: dev_total_pulses
            value: !lambda 'return x;'
```

### Anything in the logs that might be useful for us?

```txt
[14:23:59][D][sensor:127]: 'DEV Laadpaal Raw counter': Sending state 120.00800 pulses/min with 2 decimals of accuracy
[14:23:59][D][pulse_counter:180]: 'DEV Laadpaal Raw counter': Total : 1977 pulses
[14:23:59][D][sensor:127]: 'DEV Laadpaal Raw Total': Sending state 1977.00000 pulses with 0 decimals of accuracy
[14:24:03][D][esp32.preferences:113]: Saving 1 preferences to flash...
[14:24:03][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed

[... reset ...]

WARNING laadpaal-dev.roosnl.com: Connection error occurred: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for laadpaal-dev
WARNING Disconnected from API
INFO Successfully connected to laadpaal-dev
[14:26:23][D][pulse_counter:174]: 'DEV Laadpaal Raw counter': Retrieved counter: 99.86 pulses/min
[14:26:23][D][sensor:127]: 'DEV Laadpaal Raw counter': Sending state 99.86089 pulses/min with 2 decimals of accuracy
[14:26:23][D][pulse_counter:180]: 'DEV Laadpaal Raw counter': Total : 186 pulses
[14:26:23][D][sensor:127]: 'DEV Laadpaal Raw Total': Sending state 186.00000 pulses with 0 decimals of accuracy
```

jeroenrnl, 2023-07-01:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

WHAT?

ssieb, 2024-11-11:
@jeroenrnl you aren't saving the total value, you're saving the pulse counter value.
@Dimitry-Muzi if you have the same problem, then there's the answer.  But it's probably not the same.  Come ask on the esphome discord server.

closed 2024-11-11 by ssieb

jeroenrnl, 2024-11-11:
> @jeroenrnl you aren't saving the total value, you're saving the pulse counter value.

Ehm...no?

`
globals:
  - id: dev_**total**_pulses
    type: int
    restore_value: yes
`
Any way, I don't care anymore, I ditched ESPhome for Tasmota, where it doesn't take 1 and a half year to give a useless reply to a big report.

ssieb, 2024-11-11:
```yaml
sensor:
  - platform: pulse_counter
    use_pcnt: True
    pin: 5
    icon: "mdi:flash"
    name: "DEV Laadpaal Raw counter"
    id: dev_counter
    total:
      id: dev_counter_total
      name: 'DEV Laadpaal Raw Total'
    on_value:
      then:
        - globals.set:
            id: dev_total_pulses
            value: !lambda 'return x;'
```
Note where you're setting the global from.  The `on_value` of the main sensor, not the total.  There's no bug, except in your config.

jeroenrnl, 2024-11-13:
the on_value triggers the globals.set whenever it changes, that part works. Anyway, I could have looked into it if you would have replied one and a half year ago. Now it's pointless. I've removed it.

ssieb, 2024-11-13:
You're still not getting it.  That part doesn't work because you're doing it wrong.  If you wanted a a quick answer, you should have asked on the discord server.  But whatever, have fun with Tasmota.

jeroenrnl, 2024-11-13:
No, I don't get it. I don't get why it wouldn't work this way, but I certainly don't get why I would install some proprietary chat program instead of just using issues for what they are meant for: reporting issues. I also don't get why it is so hard to appreciate the fact that someone took time to report an issue, it's just closed after a ridiculous amount of time. Mind you, you didn't give any hint on what was doing wrong until I answered to your close message, this is just a terrible way to interact with your users.

Also, I just took another look at the documentation and there is no way I could have found what you are telling me in there, because frankly, the documentation is really bad.