esphome/issues #4268: mics_4514 not reporting data

villasenor, 2023-03-02:
### The problem

I want to use the `mics_4514` from DFRobot to detect various gasses. I have it hooked up to a firebeetle 2 ESP32-E, and it all seems to be recognized, but I'm not seeing it register any actual detection of gasses in response to those gasses being present. I know there is a warm-up time, but this is the case even 24 hours after it has been online.

[image]

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

firebeetle 2

### Component causing the issue

mics_4514

### Example YAML snippet

```yaml
esphome:
  name: room-esp32
  friendly_name: room-esp32

esp32:
  board: esp32dev

# Enable logging
logger:

i2c:

# Enable Home Assistant API
api:
  encryption:
    key: "password"

ota:
  password: "password"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "room-Esp32 Fallback Hotspot"
    password: "password"

captive_portal:

sensor:
  - platform: mics_4514
    nitrogen_dioxide:
      name: Nitrogen Dioxide
      unit_of_measurement: PPM
      accuracy_decimals: 2
    carbon_monoxide:
      name: Carbon Monoxide
      unit_of_measurement: PPM
      accuracy_decimals: 2
    hydrogen:
      name: Hydrogen
      unit_of_measurement: PPM
      accuracy_decimals: 2
    ethanol:
      name: Ethanol
      unit_of_measurement: PPM
      accuracy_decimals: 2
    methane:
      name: Methane
      unit_of_measurement: PPM
      accuracy_decimals: 2
    ammonia:
      name: Ammonia
      unit_of_measurement: PPM
      accuracy_decimals: 2
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/room-esp32.yaml...
INFO Starting log output from room-esp32.local using esphome API
INFO Successfully connected to room-esp32.local
[09:26:59][I][app:102]: ESPHome version 2023.2.4 compiled on Feb 26 2023, 20:29:59
[09:26:59][C][wifi:504]: WiFi:
[09:26:59][C][wifi:362]:   Local MAC: 08:3A:J1:14:2A:DC
[09:26:59][C][wifi:363]:   SSID: [redacted]
[09:26:59][C][wifi:364]:   IP Address: 192.168.1.2
[09:26:59][C][wifi:366]:   BSSID: [redacted]
[09:26:59][C][wifi:367]:   Hostname: 'room-esp32'
[09:26:59][C][wifi:369]:   Signal strength: -76 dB ▂▄▆█
[09:26:59][C][wifi:373]:   Channel: 11
[09:26:59][C][wifi:374]:   Subnet: 255.255.255.0
[09:26:59][C][wifi:375]:   Gateway: 192.168.1.1
[09:26:59][C][wifi:376]:   DNS1: 1.1.1.1
[09:26:59][C][wifi:377]:   DNS2: 0.0.0.0
[09:26:59][C][logger:293]: Logger:
[09:26:59][C][logger:294]:   Level: DEBUG
[09:26:59][C][logger:295]:   Log Baud Rate: 115200
[09:26:59][C][logger:296]:   Hardware UART: UART0
[09:26:59][C][i2c.arduino:052]: I2C Bus:
[09:26:59][C][i2c.arduino:053]:   SDA Pin: GPIO21
[09:26:59][C][i2c.arduino:054]:   SCL Pin: GPIO22
[09:26:59][C][i2c.arduino:055]:   Frequency: 50000 Hz
[09:26:59][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[09:26:59][I][i2c.arduino:068]: Results from i2c bus scan:
[09:26:59][I][i2c.arduino:074]: Found i2c device at address 0x75
[09:26:59][C][mics_4514:035]: MICS 4514:
[09:26:59][C][mics_4514:036]:   Address: 0x75
[09:26:59][C][mics_4514:037]:   Update Interval: 60.0s
[09:26:59][C][mics_4514:038]:   Nitrogen Dioxide 'Nitrogen Dioxide'
[09:26:59][C][mics_4514:038]:     State Class: 'measurement'
[09:26:59][C][mics_4514:038]:     Unit of Measurement: 'PPM'
[09:26:59][C][mics_4514:038]:     Accuracy Decimals: 2
[09:26:59][C][mics_4514:039]:   Carbon Monoxide 'Carbon Monoxide'
[09:26:59][C][mics_4514:039]:     State Class: 'measurement'
[09:26:59][C][mics_4514:039]:     Unit of Measurement: 'PPM'
[09:26:59][C][mics_4514:039]:     Accuracy Decimals: 2
[09:26:59][C][mics_4514:040]:   Methane 'Methane'
[09:26:59][C][mics_4514:040]:     State Class: 'measurement'
[09:26:59][C][mics_4514:040]:     Unit of Measurement: 'PPM'
[09:26:59][C][mics_4514:040]:     Accuracy Decimals: 2
[09:26:59][C][mics_4514:041]:   Ethanol 'Ethanol'
[09:26:59][C][mics_4514:041]:     State Class: 'measurement'
[09:26:59][C][mics_4514:041]:     Unit of Measurement: 'PPM'
[09:26:59][C][mics_4514:041]:     Accuracy Decimals: 2
[09:26:59][C][mics_4514:042]:   Hydrogen 'Hydrogen'
[09:26:59][C][mics_4514:042]:     State Class: 'measurement'
[09:26:59][C][mics_4514:042]:     Unit of Measurement: 'PPM'
[09:26:59][C][mics_4514:042]:     Accuracy Decimals: 2
[09:26:59][C][mics_4514:043]:   Ammonia 'Ammonia'
[09:26:59][C][mics_4514:043]:     State Class: 'measurement'
[09:26:59][C][mics_4514:043]:     Unit of Measurement: 'PPM'
[09:26:59][C][mics_4514:043]:     Accuracy Decimals: 2
[09:26:59][C][captive_portal:088]: Captive Portal:
[09:26:59][C][mdns:108]: mDNS:
[09:26:59][C][mdns:109]:   Hostname: room-esp32
[09:26:59][C][ota:093]: Over-The-Air Updates:
[09:26:59][C][ota:094]:   Address: room-esp32.local:3232
[09:26:59][C][ota:097]:   Using Password.
[09:26:59][C][api:138]: API Server:
[09:26:59][C][api:139]:   Address: room-esp32.local:6053
[09:26:59][C][api:141]:   Using noise encryption: YES
```

### Additional information

I have a FireBeetle 2 ESP32-E - the `firebeetle32` board type does not work (won't build), and the `dfrobot_firebeetle2_esp32s3` board type doesn't work with i2c (esp32s3 isn't supported...but I think this board is an ESP32-E anyway). Because of this, I use the `esp32dev` board type. The gas sensor board is the Gravity: MEMS Gas Sensor (CO, Alcohol, NO2 & NH3) - I2C - MiCS-4514.

sjwright, 2023-04-18:
I'm in the same situation. My board is a Wemos D1 Mini Compatible ESP32.

```
[23:40:21][I][app:102]: ESPHome version 2023.3.2 compiled on Apr 18 2023, 23:39:42
[23:40:21][C][wifi:504]: WiFi:
[23:40:21][C][wifi:362]:   Local MAC: 9C:9C:1F:C5:65:20
[23:40:21][C][wifi:363]:   SSID: [redacted]
[23:40:21][C][wifi:364]:   IP Address: [redacted]
[23:40:21][C][wifi:366]:   BSSID: [redacted]
[23:40:21][C][wifi:367]:   Hostname: 'gas-sensor'
[23:40:21][C][wifi:369]:   Signal strength: -55 dB ▂▄▆█
[23:40:21][C][wifi:373]:   Channel: 1
[23:40:21][C][wifi:374]:   Subnet: 255.255.255.0
[23:40:21][C][wifi:375]:   Gateway: [redacted]
[23:40:21][C][wifi:376]:   DNS1: [redacted]
[23:40:21][C][wifi:377]:   DNS2: 0.0.0.0
[23:40:21][C][logger:293]: Logger:
[23:40:21][C][logger:294]:   Level: DEBUG
[23:40:21][C][logger:295]:   Log Baud Rate: 0
[23:40:21][C][logger:296]:   Hardware UART: UART0
[23:40:22][C][i2c.arduino:052]: I2C Bus:
[23:40:22][C][i2c.arduino:053]:   SDA Pin: GPIO21
[23:40:22][C][i2c.arduino:054]:   SCL Pin: GPIO22
[23:40:22][C][i2c.arduino:055]:   Frequency: 50000 Hz
[23:40:22][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[23:40:22][I][i2c.arduino:068]: Results from i2c bus scan:
[23:40:22][I][i2c.arduino:074]: Found i2c device at address 0x75
[23:40:22][C][uptime.sensor:031]: Uptime Sensor 'Gas Sensor Node Uptime'
[23:40:22][C][uptime.sensor:031]:   Device Class: 'duration'
[23:40:22][C][uptime.sensor:031]:   State Class: 'total_increasing'
[23:40:22][C][uptime.sensor:031]:   Unit of Measurement: 's'
[23:40:22][C][uptime.sensor:031]:   Accuracy Decimals: 0
[23:40:22][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[23:40:22][C][mics_4514:035]: MICS 4514:
[23:40:22][C][mics_4514:036]:   Address: 0x75
[23:40:22][C][mics_4514:037]:   Update Interval: 20.0s
[23:40:22][C][mics_4514:038]:   Nitrogen Dioxide 'Gas Sensor Nitrogen Dioxide'
[23:40:22][C][mics_4514:038]:     State Class: 'measurement'
[23:40:22][C][mics_4514:038]:     Unit of Measurement: 'ppm'
[23:40:22][C][mics_4514:038]:     Accuracy Decimals: 2
[23:40:22][C][mics_4514:039]:   Carbon Monoxide 'Gas Sensor Carbon Monoxide'
[23:40:22][C][mics_4514:039]:     State Class: 'measurement'
[23:40:22][C][mics_4514:039]:     Unit of Measurement: 'ppm'
[23:40:22][C][mics_4514:039]:     Accuracy Decimals: 2
[23:40:22][C][mics_4514:040]:   Methane 'Gas Sensor Methane'
[23:40:22][C][mics_4514:040]:     State Class: 'measurement'
[23:40:22][C][mics_4514:040]:     Unit of Measurement: 'ppm'
[23:40:22][C][mics_4514:040]:     Accuracy Decimals: 2
[23:40:22][C][mics_4514:041]:   Ethanol 'Gas Sensor Ethanol'
[23:40:22][C][mics_4514:041]:     State Class: 'measurement'
[23:40:22][C][mics_4514:041]:     Unit of Measurement: 'ppm'
[23:40:22][C][mics_4514:041]:     Accuracy Decimals: 2
[23:40:22][C][mics_4514:042]:   Hydrogen 'Gas Sensor Hydrogen'
[23:40:22][C][mics_4514:042]:     State Class: 'measurement'
[23:40:22][C][mics_4514:042]:     Unit of Measurement: 'ppm'
[23:40:22][C][mics_4514:042]:     Accuracy Decimals: 2
[23:40:22][C][mics_4514:043]:   Ammonia 'Gas Sensor Ammonia'
[23:40:22][C][mics_4514:043]:     State Class: 'measurement'
[23:40:22][C][mics_4514:043]:     Unit of Measurement: 'ppm'
[23:40:22][C][mics_4514:043]:     Accuracy Decimals: 2
[23:40:22][C][status:034]: Status Binary Sensor 'Gas Sensor Node Status'
[23:40:22][C][status:034]:   Device Class: 'connectivity'
[23:40:22][C][restart.button:017]: Restart Button 'Gas Sensor Node Restart'
[23:40:22][C][mdns:108]: mDNS:
[23:40:22][C][mdns:109]:   Hostname: gas-sensor
[23:40:22][C][ota:093]: Over-The-Air Updates:
[23:40:22][C][ota:094]:   Address: gas-sensor.local:3232
[23:40:22][C][ota:097]:   Using Password.
[23:40:22][C][api:138]: API Server:
[23:40:22][C][api:139]:   Address: gas-sensor.local:6053
[23:40:22][C][api:141]:   Using noise encryption: YES
[23:40:22][C][wifi_signal.sensor:009]: WiFi Signal 'Gas Sensor Wifi Signal'
[23:40:22][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[23:40:22][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[23:40:22][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[23:40:22][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[23:40:28][D][sensor:127]: 'Gas Sensor Carbon Monoxide': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Carbon Monoxide': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Nitrogen Dioxide': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Nitrogen Dioxide': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Methane': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Methane': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Ethanol': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Ethanol': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Hydrogen': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Hydrogen': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Ammonia': Sending state 0.00000 ppm with 2 decimals of accuracy
[23:40:28][D][sensor:127]: 'Gas Sensor Ammonia': Sending state 0.00000 ppm with 2 decimals of accuracy
```

villasenor, 2023-04-21:
@jesserockz - We discussed this [in Discord](https://discord.com/channels/429907082951524364/1084549801691054210/1084552115260752025) - I tested with a lighter, and I'm still having this issue. Looks like @sjwright is having the same issue. I don't think it is a calibration issue - something else is happening. What would be the best way to start troubleshooting this? Happy to gather data for you.

Lou-Menz, 2023-07-09:
Same here.
I2c device detected at address 0x75, but no data.

ryuzaki-ryuga, 2023-07-18:
Same as well at address 0x75 also no data. Curious I'm using a AirGradient pro kit and using the extra i2c pins on the board for the mems-4514 if anyone is in the same situation

jesserockz, 2023-07-19:
Please set logs to VERBOSE and post the line that is like:
```
   [mics_4514] Got data: %02X %02X %02X %02X %02X %02X
```

Lou-Menz, 2023-07-21:
[10:52:38][V][mics_4514:056]: Got data: 00 5F 00 EF 03 58
[10:52:38][V][sensor:043]: 'Carbon Monoxide': Received new state 0.000000
[10:52:38][D][sensor:094]: 'Carbon Monoxide': Sending state 0.00000 ppm with 2 decimals of accuracy
[10:52:38][V][sensor:043]: 'Nitrogen Dioxide': Received new state 0.000000
[10:52:38][D][sensor:094]: 'Nitrogen Dioxide': Sending state 0.00000 ppm with 2 decimals of accuracy
[10:52:38][V][sensor:043]: 'Methane': Received new state 0.000000
[10:52:38][D][sensor:094]: 'Methane': Sending state 0.00000 ppm with 2 decimals of accuracy
[10:52:38][V][sensor:043]: 'Ethanol': Received new state 0.000000
[10:52:38][D][sensor:094]: 'Ethanol': Sending state 0.00000 ppm with 2 decimals of accuracy
[10:52:38][V][sensor:043]: 'Hydrogen': Received new state 0.000000
[10:52:38][D][sensor:094]: 'Hydrogen': Sending state 0.00000 ppm with 2 decimals of accuracy
[10:52:38][V][sensor:043]: 'Ammonia': Received new state 0.000000
[10:52:38][D][sensor:094]: 'Ammonia': Sending state 0.00000 ppm with 2 decimals of accuracy
[10:52:38][W][component:204]: Component mics_4514.sensor took a long time for an operation (0.10 s).
[10:52:38][W][component:205]: Components should block for at most 20-30ms.

Coldness00, 2023-09-01:
Hello,
Any news on this topic?
I'm experiencing same issue with d1mini

owengrey, 2023-09-15:
This works for me:

i2c:
  id: bus_a
  sda: GPIO8 # ESP32-C3 MINI
  scl: GPIO9
  scan: True
 

# Example configuration entry
sensor:
  - platform: mics_4514
    i2c_id: bus_a
    address: 0x75
    nitrogen_dioxide:
      name: Nitrogen Dioxide
    carbon_monoxide:
      name: Carbon Monoxide
    hydrogen:
      name: Hydrogen
    ethanol:
      name: Ethanol
    methane:
      name: Methane
    ammonia:
      name: Ammonia
    update_interval: 60s
    
It will not show any readings unless it senses it, make sure you have some gas to try, i used carbon monoxide alarm test kit.

Lou-Menz, 2023-09-17:
> It will not show any readings unless it senses it, make sure you have some gas to try, i used carbon monoxide alarm test kit.

Thanks owengrey.

I'll get a test kit and try tit out.

ev5unleash, 2023-12-14:
Bumping this. Having almost the same issue except mine it intermittent.

I double checked the cabling and everything tests fine.

Like others, the device is always detected. However I either receive 0 for all values, 1 value for Nitrogen, or rarely all values. Resetting the ESP seems to help at times.

cjsteward07, 2024-01-20:
I'm having the same issue. After reading the [Datasheet](https://www.sgxsensortech.com/content/uploads/2014/08/0278_Datasheet-MiCS-4514.pdf) I'm wondering if adding a housing and a [filter like this](https://imgur.com/a/jiksrnD) would help. Maybe there's excessive airflow, humidity, dust, or whatever else messing with the resistance and calibration? This is the [article](https://arduino.steamedu123.com/entry/MiCS-4514-Nitrogen-Dioxide-Gas-Sensor) I pulled that picture from, it has a bit more information on the importance of blocking airflow and adding a filter. [Here's](https://arduino.steamedu123.com/entry/MiCS-4514-Nitrogen-Dioxide-Gas-Sensor) another interesting study on filter material for gas sensors.  

**Sources:**  
[Datasheet](https://www.sgxsensortech.com/content/uploads/2014/08/0278_Datasheet-MiCS-4514.pdf)  
[Additional Info](https://arduino.steamedu123.com/entry/MiCS-4514-Nitrogen-Dioxide-Gas-Sensor)  
[Filter Study](https://arduino.steamedu123.com/entry/MiCS-4514-Nitrogen-Dioxide-Gas-Sensor)

dominikaherbst, 2024-08-06:
I found this on youtube .... https://www.youtube.com/watch?v=ZAFYBPJMd2M
https://github.com/makertut/esphome-air-sensors

Unfortunately, I couldn't make it work. I cloned the git repository to HA ... validation and compiling worked, but then the esp couldn't be reached anymore ...

dominikaherbst, 2024-08-08:
Ok, I figured it out ... It is just a crappy sensor ... I switched through the addresses. After switching to 0x78, it showed again 0.00 ppm for everything, but then I put the sensor in a glass in which I had extinguished a the flame of a lighter. Everything maxed out except for NO2. I tested it with kitchen paper soaked in 99% EtOH, again - same result
[image]. Then I reduced the readout time. If you let it sit for quite a while everything seems to decrease linearly. Then I looked at the datasheet and it says that it has two sensors: An ox sensor and a red sensor. After some testing, I have the feeling that it has no sensitivity to individual gases at all. It is highly unspecific and will only detect something after giving it a blast of a gas mixture and then it maxes out on everything, because it actually cannot distinguish. So if you get a reading for one of the gases, it doesn't mean that the gas is actually present.
For the price I am very disappointed ... 
It turned out that an SGP30 sensor, which was close by, reacted much more sensitively (1/8th of the price). Of course it doesn't tell you which gas it detects, but it seems like the MiCS 4514 also doesn't do that.