esphome/issues #4269: Cannot have a component named the "Friendly Name"

jjross, 2023-03-02:
### The problem

When you add the friendly name into the configuration it’s automatically prefixed to the Home Assistant entity name and entity ID of the exposed components.  This is expected.  

The problem is that when you specify the "name" of a component to be the same as the friendly name (because you would like the component to have the same name as the friendly name) then the system will duplicate the friendly name in both the entity ID and the entity name fields.  

ESPHome should not prefix the name of a component with the friendly name if the component's name is the same as the friendly name.  

For example, if I have a simple smart plug with a single relay and I would like to use it to control a fan I might give the device the friendly name "Bedroom Fan" and name the relay "Bedroom Fan".  Currently ESPHome will prefix the name with the friendly name so the entity becomes "Bedroom Fan Bedroom Fan" with an entity ID of "switch.bedroom_fan_bedroom_fan".

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.1

### What platform are you using?

ESP8266

### Board

Sonoff S31

### Example YAML snippet

```yaml
#Sonoff S31 Lite

substitutions:
  device_name: bedroom-fan #change
  friendly_name: Bedroom Fan #change
  comment: Sonoff S31 Lite | IP 192.168.30.184
  icon: "mdi:fan"

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  comment: ${comment}

esp8266:
  board: esp01_1m

wifi:
  ssid: !secret wifi_NoT_ssid
  password: !secret wifi_NoT_password
  fast_connect: true
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: !secret wifi_ap_password

captive_portal:

# Enable logging
logger:
  baud_rate: 0  # (UART logging interferes with cse7766)

# Enable Home Assistant API
api:
  encryption: 
    key: !secret esp_home_api_encryption

ota:

time:
  - platform: homeassistant
    id: homeassistant_time

uart:
  rx_pin: RX
  baud_rate: 4800

binary_sensor:
  - platform: gpio
    id: button
    internal: true
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.toggle: relay
  - platform: status
    name: Status

switch:
  - platform: gpio
    name: ${friendly_name}
    icon: ${icon}
    pin: GPIO12
    id: relay

  - platform: restart
    name: Reboot

  - platform: safe_mode
    name: Safe Mode

status_led:
  pin:
    number: 13
    mode: OUTPUT
    inverted: true

sensor:
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s
```

ssieb, 2023-03-02:
This is a known issue, but a solution hasn't been figured out yet.

jjross, 2023-03-03:
Ok, understood. I tried to search for an existing bug report and couldn’t find one. 

Would it make sense to have the friendly name as the component name if a name is not specified?

ssieb, 2023-03-03:
That was one idea, but the problem is that makes the logging really weird.

jjross, 2023-03-03:
That reminds me, the logs don't actually reflect the prefixed name.  Is that by design or a mistake?  To me they should probably reflect what people are going to see in the HA interface.

ssieb, 2023-03-03:
Those are completely different.  HA does whatever it does, but the esphome component name is what you give it in the config.  So the fix probably needs to be in HA to drop the prefix or the name if they are both the same.  If you're looking at logs, then you already know which device you're watching.

jjross, 2023-03-06:
OK, that's fine though I think maybe that needs to be made more clear in the docs.  The behaviour of the friendly name isn't exactly made clear.

DAVe3283, 2023-03-20:
I have been running half a dozen devices with components named `""` (empty string) and they show up as expected in Home Assistant. It does look a little weird in the logs, which don't prepend the friendly name, but since you really should only have one component named the friendly name, it is still clear which component it is with the empty name.

jjross, 2023-03-20:
I hadn't considered using an empty string.  I suppose that would also
work.  I decided to just revert to not having a friendly name.

closed 2023-03-26 by jesserockz