esphome/issues #4272: Error: Could not find one of 'package.json' manifest files in the package

manelmg, 2023-03-03:
### The problem

I try to compile simple yalm and always rise the error: "Error: Could not find one of 'package.json' manifest files in the package"

I have a fresh installation with pip and last esphome version (v2023.2.4)

### Which version of ESPHome has the issue?

v2023.2.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Example YAML snippet

```yaml
esphome:
  name: ttt
  friendly_name: ttt

esp32:
  board: esp32dev
  framework:
    type: arduino
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /home/esphome/config/ttt.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ttt (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Error: Could not find one of 'package.json' manifest files in the package
```

ssieb, 2023-03-03:
Delete the `packages` directory in the `.platformio` directory and try again.  It's probably at `/home/esphome/.platformio`.  If that doesn't work, then delete the entire `.platformio` directory.

manelmg, 2023-03-06:
Done!!!
.platform directory deleted.
I have done some tests and it seems is working correctly now.

Thanks

closed 2023-03-06 by manelmg

marxtinus, 2023-04-18:
Hi, Impossible to find .platform folder in my hassos setup. 
I ended up adding older version to my yaml file : 
```
 framework:
    type: arduino
    version: 2.0.6
```
Actually this make ESPhome to redownload a "new" version of the framework. 
My guess is that my current vesion of the framework is corrupted but I found no way to delete it.