esphome/issues #4274: Fading/Transition has wrong behavior (goes darker first)

TheAndi, 2023-03-04:
### The problem

I am using custom-made PCBs with ESP-S2 to control 12V CCT strips and WS2815 strips.
After setting everything up with ESPHome and Homeassist, everything works flawlessly **except** the fading.

Every time I trigger any transition e.g. color/brightness change, the fading does not seam right. Or let me say the path from the current state (rgb/brightness) to the new state takes some detours. 

I set `default_transition_length:` to `5s` to exaggerate and study this effect.

It seems that any new state (actually, even sending the same color from the color picker in HA) leads to the strip dimming down first and then going to the new state. How can that be?

This happens for my addressable and CCT strips the same way, being more visible with the digital ones.

E.g. when dimming down the digital strips to a lower brightness, they undershoot the new brightness and then go up again.
I hope my following drawing can visualize it:

``` 
            +------------------------------------------------------+
            |                                                      |
            |old value                                             |
            |--------\                                             |
            |        |-\                                           |
            |        |  -\                                         |
 Brightness |        |    -\                                       |
            |        |      -\                                     |
            |        |        -\                   new value       |
            |        |          -\          -----------------------|
            |        |            -\      -/                       |
            |        |              -\  -/                         |
            |        |                -/                           |
            +--------|---------------------------------------------+
               new state cmd          Time
```

When the new value is very low. The strip even goes off shortly before reaching the new brightness... 
I also set `gamma_correct: 1` to not mess with anything.

It is not my hardware, as this does not arise when using WLED and its fading.

Could you please let me know what I'm doing wrong or if this is a bug?

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP32

### Board

Custom made board based on ESP-S2

### Component causing the issue

Fading in ESPHome

### Example YAML snippet

```yaml
esphome:
  name: lightbar-left

esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  password: "XXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Light-Bar-Left Fallback Hotspot"
    password: "XXX"

captive_portal:

status_led:
  pin:
    number: GPIO36

web_server:
  port: 80
    
substitutions:
  CH1_WW_WS_Data: GPIO1
  CH1_CW_WS_GND:  GPIO2
  CH2_WW_WS_Data: GPIO3
  CH2_CW_WS_GND:  GPIO4
  CH3_WW_WS_Data: GPIO5
  CH3_CW_WS_GND:  GPIO6
  CH4_WW_WS_Data: GPIO7
  CH4_CW_WS_GND:  GPIO8
  STATUS_LED:     GPIO36
  SHIFTER_DISABLE: GPIO38

# Example configuration entry
light:
  - platform: neopixelbus
    type: RGB
    variant: 800kbps
    pin: $CH1_WW_WS_Data
    method:
      type: esp32_i2s
      bus: 0
    num_leds: 216
    name: "Left Top Color"
    id: left_top_color
    gamma_correct: 1
    default_transition_length: 5s
    effects:
          - addressable_fireworks:
          - addressable_rainbow:
          - addressable_twinkle:
          - wled:
              port: 21324
          - e131:
              universe: 1
              channels: RGB

  - platform: neopixelbus
    type: RGB
    variant: 800kbps
    pin: $CH3_WW_WS_Data
    method:
      type: bit_bang
    num_leds: 216
    name: "Left Bottom Color"
    id: left_bottom_color
    gamma_correct: 1
    default_transition_length: 0s
    effects:
          - addressable_fireworks:
          - addressable_rainbow:
          - addressable_twinkle:
          - wled:
              port: 21324
          - e131:
              universe: 1
              channels: RGB

  - platform: cwww
    name: "Left Top CCT"
    cold_white: top_cct_output_cw
    warm_white: top_cct_output_ww
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    default_transition_length: 5s
    constant_brightness: False

  - platform: cwww
    name: "Left Bottom CTT"
    cold_white: bottom_cct_output_cw
    warm_white: bottom_cct_output_ww
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    constant_brightness: true

output:
  - platform: ledc
    id: top_cct_output_ww
    pin: $CH2_WW_WS_Data
  - platform: ledc
    id: top_cct_output_cw
    pin: $CH2_CW_WS_GND
  - platform: ledc
    id: bottom_cct_output_ww
    pin: $CH4_WW_WS_Data
  - platform: ledc
    id: bottom_cct_output_cw
    pin: $CH4_CW_WS_GND

switch:
  - platform: gpio
    name: "Left Top Color Power"
    id: CH1_GND
    pin: $CH1_CW_WS_GND

  - platform: gpio
    name: "Left Bottom Color Power"
    id: CH3_GND
    pin: $CH3_CW_WS_GND

wled:
e131:
  method: multicast # default: register E1.31 to Multicast group
```

bssstudio, 2024-05-21:
**Which version of ESPHome has the issue?**
2024.4.2

**What type of installation are you using?**
Home Assistant Add-on

**Which version of Home Assistant has the issue?**
2024.5.2

**What platform are you using?**
ESP8266

**Board**
Athom Led Strip Controller LS-5P

**Component causing the issue**
Fading in ESPHome - color temperature transition with gamma correction and constant brightness

I can confirm this is an issue with esp8266 software PWM - it it in fact an issue in calculating the brightness values for each channel of CCT lights when gamma correction and constant brightness are enabled.

kerberjg, 2024-12-06:
**Which version of ESPHome has the issue?**
2024.4.2

**What type of installation are you using?**
Home Assistant Core

**Which version of Home Assistant has the issue?**
2024.11.1

**What platform are you using?**
ESP32

**Board**
`esp32-c6-devkitm-1`

**Component causing the issue**
Light

I observed the same happening when using the light platform `partition` (attached to a `esp32_rmt_led_strip` component) with gamma_correct manually specified.

It doesn't happen on `esp32_rmt_led_strip` itself (when changing the color of the whole strip), just the partition. `gamma_correct` is only applied on the latter. Applying it on both changes nothing.

It only happens when `gamma_correct` is manually specified, regardless of value. Even if I set it to the same value as default (2.8) it behaves as described in the OP

kerberjg, 2024-12-06:
@oxan Saw you worked on gamma correction, could you please take a look at this?

OP's chart explains really well what's happening, but I can provide more debug material if requested