esphome/issues #4275: Changing IP to manual (with change of IP) doesn't work

christschnman, 2023-03-05:
### The problem

When I tried to set the IP to another manual IP with....

```
 # Optional manual IP
  manual_ip:
    static_ip: 192.168.0.50
    gateway: 192.168.0.1
    subnet: 255.255.255.0
```

....it didn't work because ESPhome tries to connect with the new manual IP from the config, which does not already exists. Instead of that, ESPhome should connect with the last known IP

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

IP-Adress

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/wp-stromzahler.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing wp-stromzahler (board: d1_mini_lite; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
RAM:   [====      ]  41.9% (used 34308 bytes from 81920 bytes)
Flash: [=====     ]  48.1% (used 492453 bytes from 1023984 bytes)
========================= [SUCCESS] Took 7.80 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.50
ERROR Connecting to 192.168.0.50:8266 failed: [Errno 113] No route to host
```

randybb, 2023-03-05:
You need to tell dashboard which address it has to use - use the old address. After OTA you need to remove it and then it will use the configured address.
https://esphome.io/components/esphome.html?highlight=use_address#changing-esphome-node-name

ssieb, 2023-03-05:
There is no "last known IP address".  The dashboard doesn't store any information like that.

closed 2023-03-05 by ssieb