esphome/issues #4277: Problem updating Sonoff Dual R3 via OTA

catgiggle, 2023-03-05:
### The problem

I have an issue with updating my Sonoff Dual R3 via OTA. Firstly, I created a YAML configuration (below) and tried to flash it to the Dual R3 through USB and ESPhome Web, but without success. Therefore, I compiled the configuration to a Legacy BIN file (Install > Manual download > Legacy format) and then initialized the Dual R3 through Tasmota Installer (https://tasmota.github.io/install/) and updated the firmware through the interface provided by the device (http://192.168.4.1/up) to EspHome (I uploaded the Legacy BIN file). Up to this point, the Dual R3 is working correctly, showing up in Home Assistant, providing the button and sensor according to the YAML configuration. However, when I try to update the firmware via OTA, it only causes the Dual R3 to restart (log below).

I am using ESPhome run through Docker from the official image (https://hub.docker.com/r/esphome/esphome). It's worth noting that I have many other devices and do not have any problems with updating via OTA. However, this is my first device on ESP32.

### Which version of ESPHome has the issue?

2023.2.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.11.5

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ota

### Example YAML snippet

```yaml
substitutions:
  name: esphome-dualr3-test
  prefix: esphome_dualr3_test

esphome:
  platform: esp32
  board: esp32dev
  name: ${name}

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: light
  ap:
    ssid: ${name}
    password: abc123456

logger:
  baud_rate: 0
  level: VERY_VERBOSE
ota:
api:

uart:
  tx_pin: GPIO25
  rx_pin: GPIO26
  baud_rate: 38400
  parity: EVEN
  stop_bits: 1

captive_portal:

web_server:
  port: 80

button:
  - platform: template
    name: "Template Button"
    on_press:
      - logger.log: Button Pressed

sensor:
  - platform: template
    name: "Template Sensor"
    lambda: |-
      return 42.0;
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/device-sonoff-dualr3-001.yaml...
INFO Starting log output from esphome-dualr3-test.local using esphome API
INFO Successfully connected to esphome-dualr3-test.local
[12:05:35][I][app:102]: ESPHome version 2023.2.2 compiled on Mar  5 2023, 11:08:18
[12:05:35][C][wifi:504]: WiFi:
[12:05:35][C][wifi:362]:   Local MAC: 78:E3:6D:62:28:78
[12:05:35][C][wifi:363]:   SSID: [redacted]
[12:05:35][C][wifi:364]:   IP Address: 10.0.10.118
[12:05:35][C][wifi:366]:   BSSID: [redacted]
[12:05:35][C][wifi:367]:   Hostname: 'esphome-dualr3-test'
[12:05:35][C][wifi:369]:   Signal strength: -54 dB ▂▄▆█
[12:05:35][C][wifi:373]:   Channel: 11
[12:05:35][C][wifi:374]:   Subnet: 255.255.0.0
[12:05:35][C][wifi:375]:   Gateway: 10.0.10.1
[12:05:35][C][wifi:376]:   DNS1: 10.0.10.1
[12:05:35][C][wifi:377]:   DNS2: 0.0.0.0
[12:05:35][C][logger:293]: Logger:
[12:05:35][C][logger:294]:   Level: DEBUG
[12:05:35][C][logger:295]:   Log Baud Rate: 0
[12:05:35][C][logger:296]:   Hardware UART: UART0
[12:05:35][C][uart.arduino_esp32:108]: UART Bus 1:
[12:05:35][C][uart.arduino_esp32:109]:   TX Pin: GPIO25
[12:05:35][C][uart.arduino_esp32:110]:   RX Pin: GPIO26
[12:05:35][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[12:05:35][C][uart.arduino_esp32:114]:   Baud Rate: 38400 baud
[12:05:35][C][uart.arduino_esp32:115]:   Data Bits: 8
[12:05:35][C][uart.arduino_esp32:116]:   Parity: EVEN
[12:05:35][C][uart.arduino_esp32:117]:   Stop bits: 1
[12:05:35][C][template.sensor:023]: Template Sensor 'Template Sensor'
[12:05:35][C][template.sensor:023]:   State Class: ''
[12:05:35][C][template.sensor:023]:   Unit of Measurement: ''
[12:05:35][C][template.sensor:023]:   Accuracy Decimals: 1
[12:05:35][C][template.sensor:024]:   Update Interval: 60.0s
[12:05:35][C][captive_portal:088]: Captive Portal:
[12:05:35][C][web_server:151]: Web Server:
[12:05:35][C][web_server:152]:   Address: esphome-dualr3-test.local:80
[12:05:35][C][mdns:108]: mDNS:
[12:05:35][C][mdns:109]:   Hostname: esphome-dualr3-test
[12:05:35][C][ota:093]: Over-The-Air Updates:
[12:05:35][C][ota:094]:   Address: esphome-dualr3-test.local:3232
[12:05:35][C][api:138]: API Server:
[12:05:35][C][api:139]:   Address: esphome-dualr3-test.local:6053
[12:05:35][C][api:143]:   Using noise encryption: NO
[12:05:40][D][api:102]: Accepted ::FFFF:10.0.20.100
[12:05:40][W][api.connection:083]: Home Assistant 2022.11.5 (::FFFF:10.0.20.100): Connection closed
[12:05:47][I][ota:113]: Boot seems successful, resetting boot loop counter.
[12:05:47][D][esp32.preferences:113]: Saving 1 preferences to flash...
[12:05:47][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[12:05:59][D][button:013]: 'Template Button' Pressed.
[12:05:59][D][main:252]: Button Pressed
[12:06:01][D][button:013]: 'Template Button' Pressed.
[12:06:01][D][main:252]: Button Pressed
[12:06:02][D][button:013]: 'Template Button' Pressed.
[12:06:02][D][main:252]: Button Pressed
[12:06:12][I][web_server_base:041]: OTA Update Start: esphome-dualr3-test.bin
WARNING esphome-dualr3-test.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esphome-dualr3-test.local
WARNING Disconnected from API
INFO Successfully connected to esphome-dualr3-test.local
[12:06:40][D][api:102]: Accepted ::FFFF:10.0.20.100
[12:06:40][W][api.connection:083]: Home Assistant 2022.11.5 (::FFFF:10.0.20.100): Connection closed
[12:06:56][I][web_server_base:041]: OTA Update Start: esphome-dualr3-test.bin
[12:06:56][D][web_server_base:073]: OTA in progress: 0.1%
WARNING esphome-dualr3-test.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esphome-dualr3-test.local
WARNING Disconnected from API
INFO Successfully connected to esphome-dualr3-test.local
[12:07:25][D][button:013]: 'Template Button' Pressed.
[12:07:25][D][main:252]: Button Pressed
```

james194zt, 2023-05-04:
I have similar issues to the above on the same device with OTA update, I was on Tasmota, converted perfectly fine to ESP Home but needed to change a couple of things in the script.

OTA update fails, it says OTA Update in Progress 0.1% then you get an error500 on the deviceip/update page a few seconds later. Tried manual flash as well as through the ESP Home dashboard wirelessly.

I am going to flash manually on it's pins to work around it, but has anybody worked this out?

james194zt, 2023-05-04:
I have similar issues to the above on the same device with OTA update, I was on Tasmota, converted perfectly fine to ESP Home but needed to change a couple of things in the YAML.

OTA update fails, it says OTA Update in Progress 0.1% then you get an error500 on the deviceip/update page a few seconds later. Tried manual flash as well as through the ESP Home dashboard wirelessly.

I am going to flash manually on it's pins to work around it, but has anybody worked this out?

My YAML for info

esphome:
name: sw01-kitchen
friendly_name: KITCHEN

esp32:
board: esp32dev
framework:
type: arduino

web_server:
port: 80

BL0939 Power Monitoring Configuration
Disable logging over serial
logger:
baud_rate: 0

uart:
tx_pin: GPIO25
rx_pin: GPIO26
baud_rate: 4800
parity: NONE
stop_bits: 2

sensor:

BL0939 Sensors
platform: bl0939
update_interval: 30s
voltage:
name: 'Input Voltage'
current_1:
name: 'Relay Current 1'
current_2:
name: 'Relay Current 2'
active_power_1:
name: 'Relay Active Power 1'
active_power_2:
name: 'Relay Active Power 2'
energy_1:
name: 'Relay Energy 1'
energy_2:
name: 'Relay Energy 2'
energy_total:
name: 'Relay Energy Total'
Misc Sensors
platform: wifi_signal
name: "WiFi Signal"
update_interval: 30s
platform: uptime
name: "Device Uptime"
update_interval: 120s
Relay Configuration
Port 1 WASHINGMACHINE
Port 2 TUMBLEDRYER
binary_sensor:

platform: gpio
pin:
number: GPIO00
mode: INPUT_PULLUP
inverted: True
name: "button0"
#toggle both relays on push
on_press:
switch.toggle: relay
switch.toggle: relay2
platform: gpio
pin:
number: GPIO32
mode: INPUT_PULLUP
inverted: True
name: "button1"
#toggle only relay1
on_press:
switch.toggle: relay
platform: gpio
pin:
number: GPIO33
mode: INPUT_PULLUP
inverted: True
name: "button2"
#toggle only relay2
on_press:
switch.toggle: relay2
switch:

platform: gpio
name: "WASHING MACHINE"
pin: GPIO27
id: relay
platform: gpio
name: "TUMBLE DRYER & DISHWASHER"
pin: GPIO14
id: relay2
platform: restart
name: "Restart Device"
status_led:
pin:
number: GPIO13
inverted: yes

text_sensor:

platform: version
name: "Version"

james194zt, 2023-05-04:
Update...

So I flashed ESP Home directly on, no change from the above config really just changed a name to make a difference, flashed perfectly!

Only thing I can think of is it is something to do with migrating from Tasmota to ESP Home originally. Originally I flashed from the default Sonoff firmware to Tasmota (flashed using serial interface not DIY mode), then I migrated from Tasmota to ESP Home after it had been running fine for several weeks. This time around I flashed the ESPHome binary directly over serial and it worked from both the update button AND the dashboard updating.

Problem solved for me, so might be useful if anybody else encounters it.