esphome/issues #4279: Error after clean install

pepsonEL, 2023-03-06:
### The problem

Hi. After new install HASSOS 9.5 with latest HomeAssistant version and after install clean addon ESPHome i have an error. I dont find solution:
```
Logger: homeassistant.components.esphome.dashboard
Source: helpers/update_coordinator.py:215
Integration: ESPHome (documentation, issues)
First occurred: 19:59:19 (1 occurrences)
Last logged: 19:59:19

Error requesting ESPHome Dashboard data: Cannot connect to host 127.0.0.1:64417 ssl:default [Connect call failed ('127.0.0.1', 64417)]
```

Then after connect my ESPHome devices from my network which i created in old instalation(the same versio) i also get this error:

```
Logger: homeassistant.util.logging
Source: util/logging.py:156
First occurred: 20:33:35 (10 occurrences)
Last logged: 20:33:35

Exception in setup_update_entity when dispatching 'esphome_1596a31c4e4d016840901ce707e5ac7e_on_device_update': () Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 56, in setup_update_entity async_add_entities([ESPHomeUpdateEntity(entry_data, dashboard)]) File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 90, in __init__ if coordinator.supports_update and not self._device_info.has_deep_sleep: File "/usr/src/homeassistant/homeassistant/components/esphome/dashboard.py", line 91, in supports_update raise RuntimeError("Data needs to be loaded first") RuntimeError: Data needs to be loaded first
Exception in setup_update_entity when dispatching 'esphome_73249a6d2c804630a7b67ca0cff958d9_on_device_update': () Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 56, in setup_update_entity async_add_entities([ESPHomeUpdateEntity(entry_data, dashboard)]) File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 90, in __init__ if coordinator.supports_update and not self._device_info.has_deep_sleep: File "/usr/src/homeassistant/homeassistant/components/esphome/dashboard.py", line 91, in supports_update raise RuntimeError("Data needs to be loaded first") RuntimeError: Data needs to be loaded first
Exception in setup_update_entity when dispatching 'esphome_daad0316ee59ff62613a55d3dd83c1b8_on_device_update': () Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 56, in setup_update_entity async_add_entities([ESPHomeUpdateEntity(entry_data, dashboard)]) File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 90, in __init__ if coordinator.supports_update and not self._device_info.has_deep_sleep: File "/usr/src/homeassistant/homeassistant/components/esphome/dashboard.py", line 91, in supports_update raise RuntimeError("Data needs to be loaded first") RuntimeError: Data needs to be loaded first
Exception in setup_update_entity when dispatching 'esphome_b37d670027af3d64f37ec78965a772fb_on_device_update': () Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 56, in setup_update_entity async_add_entities([ESPHomeUpdateEntity(entry_data, dashboard)]) File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 90, in __init__ if coordinator.supports_update and not self._device_info.has_deep_sleep: File "/usr/src/homeassistant/homeassistant/components/esphome/dashboard.py", line 91, in supports_update raise RuntimeError("Data needs to be loaded first") RuntimeError: Data needs to be loaded first
Exception in setup_update_entity when dispatching 'esphome_0f03a5a72ce60d52ff457bc92fbb59a7_on_device_update': () Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 56, in setup_update_entity async_add_entities([ESPHomeUpdateEntity(entry_data, dashboard)]) File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 90, in __init__ if coordinator.supports_update and not self._device_info.has_deep_sleep: File "/usr/src/homeassistant/homeassistant/components/esphome/dashboard.py", line 91, in supports_update raise RuntimeError("Data needs to be loaded first") RuntimeError: Data needs to be loaded first
```

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.1

### What platform are you using?

ESP8266

Scoff123, 2023-04-20:
I've started getting these same errors but on an existing ESPHome installation which has been stable for months. These error messages seem to have started following my recent update to thr latest Home Assistant Core version, 2023.4.5 as they were not present before this point.

ygreq, 2023-05-02:
DId you manage to fix this?

I have the exact same issue

Logger: homeassistant.components.esphome.dashboard
Source: helpers/update_coordinator.py:226
Integration: esphome (documentation, issues)
First occurred: May 2, 2023 at 10:34:54 PM (1 occurrences)
Last logged: May 2, 2023 at 10:34:54 PM

Error requesting ESPHome Dashboard data: Cannot connect to host 127.0.0.1:63211 ssl:default [Connect call failed ('127.0.0.1', 63211)]

I reinstalled ESPHome and still getting this. And this affects my setup as I cannot configure a newly discovered ESPHome node. It throws a "Unknown error occurred" whenever I try to Submit.

poleax01, 2023-06-22:
> DId you manage to fix this?
> 
> I have the exact same issue. I am running ESPHome v. 2023.4.3
> 
> Logger: homeassistant.components.esphome.dashboard Source: helpers/update_coordinator.py:226 Integration: esphome (documentation, issues) First occurred: May 2, 2023 at 10:34:54 PM (1 occurrences) Last logged: May 2, 2023 at 10:34:54 PM
> 
> Error requesting ESPHome Dashboard data: Cannot connect to host 127.0.0.1:63211 ssl:default [Connect call failed ('127.0.0.1', 63211)]
> 
> I reinstalled ESPHome and still getting this. And this affects my setup as I cannot configure a newly discovered ESPHome node. It throws a "Unknown error occurred" whenever I try to Submit.

Check your firewall.
I'm fix with
sudo ufw allow 62190