esphome/issues #4283: Adalight flickers & my9231 broken after update

Robert-K, 2023-03-06:
### The problem

I'm using the ESPHome Home Assistant with several different ESP8266 and ESP32 based devices.
After updating my devices to the latest version of ESPHome (2023.2.4), I've noticed some problems:

- The Adalight light effect on my ESP32-controlled WS2812B strip flickers when I use it to act as an ambilight (using Prismatik). The log constantly outputs the following:
  `[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=19).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=56).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=95).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=21).`
This issue does not occur when using the mood lamp or rainbow mode in Prismatik. I tried different baud rates without success.
This was not a problem before updating and I know of at least one other case of the same problem occuring in a similar setup.

- All of my my9231-based RGBW bulbs started acting weird: a few seconds after booting, the light would first stutter (transitions visibly lagged) and then turn completely unresponsive, the ESP however was still communicating normally. I was able to downgrade them to 2022.11.1 and the issue disappeared.

I unfortunately do not have a backup of my old ambilight firmware, but it used to work flawlessly.

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.3 & 2023.3.1, probably all

### What platform are you using?

ESP32

### Board

nodemcu devkit v1 & tuya hardware

### Component causing the issue

adalight, my9231

### Example YAML snippet

```yaml
# ---------- ADALIGHT SNIPPET ----------

# Disable logging over USB
logger:
  baud_rate: 0

# Adalight requires higher RX buffer size
# to operate without flickering
uart:
  baud_rate: 115200
  rx_pin: GPIO3
  rx_buffer_size: 1024

adalight:

# Please use the connection type udp on the default port and add the Data Prefix 0201.
wled:

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812X
    pin: GPIO4
    num_leds: 96
    name: "Ambilight"
    effects:
      - wled:
      - adalight:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
      - strobe:
      - random:
      - pulse:
      - flicker:

# ---------- my9231 SNIPPET ----------

my9231:
  data_pin: GPIO13
  clock_pin: GPIO15
  num_channels: 4
  num_chips: 1
  bit_depth: 8

output:
  - platform: my9231
    id: output_red
    channel: 3
  - platform: my9231
    id: output_green
    channel: 2
  - platform: my9231
    id: output_blue
    channel: 1
  - platform: my9231
    id: output_cold_white
    channel: 0

light:
  - platform: rgbw
    name: Ceiling 1
    red: output_red
    green: output_green
    blue: output_blue
    white: output_cold_white
    
    # Ensure the light turns on by default if the physical switch is actuated.
    restore_mode: ALWAYS_ON
```

### Anything in the logs that might be useful for us?

```txt
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=19).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=56).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=95).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=21).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=42).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=74).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=33).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=38).
[00:29:31][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=74).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=33).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=38).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=74).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=33).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=38).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=118).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=26).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=51).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=242).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=28).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=75).
[00:29:32][D][adalight_light_effect:085]: Frame: Invalid (size=1, first=231).
```

Fma965, 2023-05-20:
Same issue with my MY9231 based light

Fma965, 2023-08-03:
Did you ever find a fix for this? @Robert-K

Robert-K, 2023-08-03:
I downgraded to the previous version and never touched it again 🤢

Fma965, 2023-08-03:
> I downgraded to the previous version and never touched it again 🤢

hehe thanks,, i run it on HA and haven't got around to setting up a old docker container to downgrade it yet, surely there is more than just us 2 with this issue though 🤔

Robert-K, 2023-08-03:
You COULD just grab the old version from an automatic HA backup (these usually get done before every upgrade), but be careful to also back up everything before you do that. I managed to wreck my entire HA config simply by clicking the wrong thing in their backup management 😅

Fma965, 2023-08-03:
> You COULD just grab the old version from an automatic HA backup (these usually get done before every upgrade), but be careful to also back up everything before you do that. I managed to wreck my entire HA config simply by clicking the wrong thing in their backup management 😅

Haha fair point, actually i might have just found a strange fix for myself atleast, when i added a "web_server:" section to the yaml it seems to have started working as intended, strange the webserver doesn't work still but the light in HA works correctly

Robert-K, 2024-03-03:
The issue doesn't appear to be present on the latest versions. All my lights run the latest stable ESPHome version and function correctly.
Feel free to close from my side.