esphome/issues #4284: Wifi failed if fan turned off on boot

VladNord, 2023-03-06:
### The problem

Hello:)
I decided to make a growbox which has two fans with pwm control, rpm monitoring and NPN mosfets for the possibility of completely turning off.
Mosfets are: IRLZ34NPBF TO-220. Fans are 12V. Diagram attached

Actually, the problem is following: if fans are turned off during boot, Wifi will fails. Tried to use different gpios- same result. If that happens it's necessary to power on ESP with disconnected  fans then plug them back.

 thanks for the help

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2

### What platform are you using?

ESP32

### Board

nodemcu

### Example YAML snippet

```yaml
switch:
  # Fan_1_Power (MOSFET)
  - platform: gpio
   # pin: GPIO4
    pin: GPIO26
    id: fan_1_power
    internal: true

  # Fan_2_Power (MOSFET)
  - platform: gpio
#    pin: GPIO2
    pin: GPIO13
    id: fan_2_power
    internal: true
```

### Additional information

[image]

ssieb, 2023-03-06:
Can you get serial logs from boot?

VladNord, 2023-03-07:
> Can you get serial logs from boot?
Nope but I can get it.
One question, can I connect Nodemcu to laptop usb, if nodemcu already powered from another power supply?

ssieb, 2023-03-07:
Use a USB to TTL uart converter then and don't connect the +V, just GND.

VladNord, 2023-03-07:
> Use a USB to TTL uart converter then and don't connect the +V, just GND.

here is it
[image]

closed 2023-07-21 by github-actions[bot]