esphome/issues #4285: Can not set fan speed to 0

camohob, 2023-03-07:
### The problem

When I set the speed from 2 (for example) to 0, the value is sent to the output and returned back to HA unchanged (2).
Why [minimum value](https://github.com/esphome/esphome/blame/1b328da26581a201c9aaaef7be9c7910797ee2db/esphome/components/fan/fan.cpp#L42) is 1? HA supports the value 0 but bounces when I try to set it to 0 and returns it back unchanged.

[image]

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.1

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

fan

### Example YAML snippet

```yaml
fan:
  - platform: speed
    id: fan_in
    output: fan_in_output
    name: "THE Fan"
    speed_count: 3
```

camohob, 2023-07-06:
ping

ravngr, 2025-07-24:
I've run up against this when trying to implement a `platform: template` fan entity. Not sure if it matters, but I'm using MQTT exclusively to connect Home Assistant and esphome nodes.

As far as I can tell there's a discrepency/inconsistency between the documentation and implementation of fan entities in Home Assistant and/or esphome. In #6586 it's pointed out that fan speeds are zero-indexed, however Home Assistant generates zero speed commands from the UI, and these are [happily accepted by MQTT fan](https://github.com/esphome/esphome/blob/06eba96fdce7a69afc0d5ba2ca16542383768ef5/esphome/components/mqtt/mqtt_fan.cpp#L102-L105):

```python
if (speed_level >= 0 && speed_level <= this->state_->get_traits().supported_speed_count()) {
  ESP_LOGD(TAG, "New speed level %d", speed_level);
  this->state_->make_call().set_speed(speed_level).perform();
}
```

However, the base fan entity [hard clamps the speed value to be >= 1](https://github.com/esphome/esphome/blob/06eba96fdce7a69afc0d5ba2ca16542383768ef5/esphome/components/fan/fan.cpp#L45):

```python
this->speed_ = clamp(*this->speed_, 1, traits.supported_speed_count());
```

The net result is that when using the fan UI with fixed speed steps there is no possibility to distinguish between a speed of 0 (off) or 1 (low). Given the behaviour of the Home Assistant UI, I think the intent is that a zero speed should be interpreted as equivalent to turning the fan off. I don't think that's too much of a reach, it's literally supported by the UI design. Additionally in the base code for the fan entity the allowable speed is [0-100 inclusive](https://github.com/home-assistant/core/blob/4cc4bd3b9a5999a804dcd878148dcf8aadf85bf1/homeassistant/components/fan/__init__.py#L133).

[image]

I'll create a PR to allow setting zero speed.