esphome/issues #4289: NSPanel tasmota to esphome fails OTA

rfaelens, 2023-03-08:
### The problem

On an NSPanel EU version, I first flashed tasmota using the tasmota web installer.
I then downloaded a Legacy image of esphome, and flashed through the tasmota web interface.
Esphome booted succesfully.

**However, a subsequent esphome OTA update failed with Connection reset by peer at 1%.** The device logs stated "OTA update starting, 0.1%" and then also aborts with Connection reset by peer.
I had to flash esphome through _serial_ first, before OTA worked on subsequent attempts.

Probably flashing esphome through the tasmota web interface does not allow changing the partition table?
In my view, there are two issues here:
- Missing error reported by esphome to indicate the problem
- Tasmota not correctly flashing esphome through OTA (e.g. missing partition table)?

Feel free to close this issue if it should be reported at Tasmota instead.

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Sonoff NSPanel

hapklaar, 2023-07-04:
> I had to flash esphome through _serial_ first, before OTA worked on subsequent attempts.

I had the same issue, thanks for also supplying the solution.

fabiobrolo, 2023-11-24:
Have you found any workaround to fix this issue?

rfaelens, 2023-11-24:
Yes! The issue was in wifi sleep settings. Disabling those allowed the nspanel to work well

closed 2023-11-24 by rfaelens

morpheus-pl, 2023-12-31:
> Yes! The issue was in wifi sleep settings. Disabling those allowed the nspanel to work well

Where exactly do You change that ?