esphome/issues #4290: Hardcoded `.local` suffix in dashboard mDNS status check

adubkov, 2023-03-08:
### The problem

This line have hardcoded `.local` suffix, which seems to be configurable, as result when using custom `wifi.domain` dashboard`s ping doesn't seem to work:
https://github.com/esphome/esphome/blob/dev/esphome/dashboard/dashboard.py#L779

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
wifi:
  domain: .office
```

DAVe3283, 2023-04-19:
In my experience, ESPHome devices always report `.local` via mDNS. I set a custom domain as well, and can ping the devices using that custom domain, but they also are detected by Home Assistant and ESPHome with `.local`.

Is your network setup blocking mDNS? Or did you disable it in the YAML? In either case, you probably just need to set your ESPHome add-on to `Use ping for status` (Settings → Add-ons → ESPHome → Configuration (might need to `Show unused optional configuration options`). That works for me when the network setup blocks mDNS (e.g. ESPHome devices and Home Assistant are on a different VLAN with strict firewall rules between them).

jesserockz, 2023-04-19:
mDNS is `.local` nothing else. A `domain` is entirely different and is `DNS` / router controlled

bdraco, 2023-11-13:
Only `.local` is supported for mdns

closed 2023-11-13 by bdraco