esphome/issues #4292: replacing character with a line break "\n" crashing ESPhome

1liminal1, 2023-03-09:
### The problem

Not sure if I'm doing this wrong or its not supported. Doesn't seem to be anything in the display doco. 

But my code is crashing the ESPhome from the logs.

```
[14:26:48][D][homeassistant.text_sensor:017]: 'sensor.epshome_shopping_list': Got state ''Do more testing', 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing', 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
[14:26:48][D][text_sensor:067]: 'alexa_shopping_list': Sending state ''Do more testing'
 'More ESP32s''
WARNING to-do.local: Connection error occurred: [Errno 104] Connection reset by peer
```

My code for converting , to a linebreak?

```
  - platform: homeassistant
    entity_id: sensor.epshome_shopping_list
    id: alexa_shopping_list
    on_value:
      - lambda: |-
          auto text = id(alexa_shopping_list).state;
          for (size_t i = 0; i < text.length(); ++i) {
            if (text[i] == ',') {
              text[i] = '\n';
            }
          }
          id(alexa_shopping_list).publish_state(text);
          id(data_updated) = true;
      ```

### Which version of ESPHome has the issue?

v2023.2.4 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant Core 2023.3.1

### What platform are you using?

ESP32

### Board

Generic Dev Board

### Component causing the issue

text_sensor

### Example YAML snippet

```yaml
above?
```

ssieb, 2023-03-09:
Github issues are not for support requests.  Ask on the discord server or one of the forums.  You're creating an infinite loop and crashing the stack.

1liminal1, 2023-03-09:
Ok, ill close this, thanks. If you could please let me know if its posable? Ill go research, just want to know its doable, like I said, there is nothing in the doco. 

Thanks

ssieb, 2023-03-09:
My last comment here.  If you need more help, come to the discord server.
Did you read the docs? https://esphome.io/components/text_sensor/index.html

ssieb, 2023-03-11:
The display printing does not handle line breaks.

closed 2023-03-11 by ssieb

1liminal1, 2023-03-11:
Hey there, 

It does work though. I'm not a dev, I found this code in one of the forums. But it works. see the screenshot for the "To Do List"

The difference is, they are doing it for a fixed width that you define, where I want to replace a character with the line break. 

So, what do I need to do here, is this a bug or what?

Here is the post for context 

https://community.home-assistant.io/t/eink-multi-line-text/255814

Thanks 

```
        std::string s = id(alexa_to_do_list).state;

          int limit = 15;

          int space = 0;

          int i = 0;

          int line = 0;

          int y= 190;

          y= 190; // start Y

          while(i<s.length()){ //loop through string, counting all the spaces, and replacing the last one with ~ [marked by space variable] if the count exceeds limit of 35

            if(s.substr(i,1) == " "){space = i; }

            if(line>limit-1){s=s.substr(0,space)+"~"+s.substr(space+1);line = 0;}

              i++;line++;}

          size_t pos = s.find("~"); //find the first line break

          int linecount = 1; //need number of lines to store the break positions in an array

          int breakpositions[10]; //store breakpositions [the '~']

          breakpositions[0] = -1; // start at -1 cause we need to truncate the replaced characters and without this will cut off 1st character of message

          while ( pos != std::string::npos) //loop through  replacing the ~ with CR - though this doesnt matter here it will never be displayed, but need to change them to keep the loop from repeating at the start

          {

            s.replace(pos,1, "\n");

            breakpositions[linecount] = pos; //store the position of the break in an array

            pos = s.find("~"); // move forward

            linecount++; // we have a new line, count it

          }

          breakpositions[linecount] = s.length(); //set the last entry in array to the length of string for calculation below

          std::string singleline; //this will be the line we print

          i = 0;

          while (i < linecount ) {  // count through the lines

            singleline = s.substr(breakpositions[i]+1,(breakpositions[i+1]-breakpositions[i]-1)); //extract each line of text from the string - strip off the CRLF and the space.

            it.printf(10, y, id(font_small_bold), "%s", singleline.c_str()); //print it!

            y=y+30; // increment y to print properly on display

          i++;

       }
```

ssieb, 2023-03-11:
It's not a bug.  That code is how you could do the line splitting yourself in the display lambda.  Someone is looking at adding multiline support for the displays, so maybe it will get added in the near future.