esphome/issues #4293: TM1814 LED strip does not work with Neopixelbus

epiniguin, 2023-03-09:
### The problem

Hi!

I would like to control TM1814 RGBW strip with ESP32 and Neoxpixelbus.
For a test purpose I cut off 2 meters, these are 40 TM1814 chips.
For a level shifting from 3.3V to 5V I use 74HCT244.

The strip works in demo mode(internal control mode) as it does not detect an input signal.
If I turn on or off the strip from HA the first 3 LEDs blink brightly for a moment and after that the strip continues to work in demo mode.

I tried inverted mode, bit_bang, esp32_i2s and esp32_rmt methods.
Also I tried to use a simple MOSFET bidirectional level shifter. The result is the same.
I tried the last ESPHOME release and dev.

### Which version of ESPHome has the issue?

2023.2.4, 2023.3.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

NeoPixelBus

### Example YAML snippet

```yaml
esphome:
  name: led2-light

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  domain: .lan
  ssid: xxx
  password: xxx
  ap:
    ssid: xxx
    password: xxx

captive_portal:

light:
  - platform: neopixelbus
    type: "WRGB"
    variant: "tm1814"
    method:
      type: bit_bang
    pin: GPIO25
    num_leds: 40
    invert: no
    color_correct: [50%, 50%, 50%, 50%]
    name: "Bedroom 2"
```

NPluta, 2023-07-12:
I have the same Problem. Any solutions yet?

rafaelespinozap, 2024-03-01:
I will try to solve the issue.