esphome/issues #4297: ezo_pmp component failing on compile

stevewull, 2023-03-09:
### The problem

Using the Atlas Scientific Hydroponics Kit. Have three EZO pumps connected over I2C bus. My Yaml passes the validation test.  But when I try to compile and install, it gives the error message below.

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.3

### What platform are you using?

ESP32

### Board

adafruit huzzah32

### Component causing the issue

ezo_pmp

### Example YAML snippet

```yaml
i2c:
  sda: 23
  scl: 22
  scan: True
  id: bus_a

ezo_pmp:
  - id: ezo_pmp_tasA
    address: 108
    update_interval: 60s

  - id: ezo_pmp_tabB
    address: 109
    update_interval: 60s

  - id: ezo_pmp_pHdown
    address: 110
    update_interval: 60s

sensor:

  - platform: ezo
    id: rtd_ezo
    name: "Temperature"
    i2c_id: bus_a
    address: 102
    accuracy_decimals: 2
    unit_of_measurement: "°C"
    update_interval: 10s 

  - platform: ezo
    id: ph_ezo
    name: "pH"
    address: 99
    accuracy_decimals: 2
    unit_of_measurement: "pH"
    update_interval: 10s

  - platform: ezo
[... 111 lines cut ...]
               ^~~~~~~~~~~~
               is_dosing_flag_
src/esphome/components/ezo_pmp/ezo_pmp.cpp:280:17: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'is_paused_'; did you mean 'is_failed'?
       if (this->is_paused_)
                 ^~~~~~~~~~
                 is_failed
src/esphome/components/ezo_pmp/ezo_pmp.cpp:281:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'is_paused_'; did you mean 'is_failed'?
         this->is_paused_->publish_state(this->is_paused_flag_);
               ^~~~~~~~~~
               is_failed
src/esphome/components/ezo_pmp/ezo_pmp.cpp:282:17: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'dosing_mode_'; did you mean 'is_dosing_flag_'?
       if (this->dosing_mode_ && this->dosing_mode_->state != DOSING_MODE_NONE)
                 ^~~~~~~~~~~~
                 is_dosing_flag_
src/esphome/components/ezo_pmp/ezo_pmp.cpp:282:39: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'dosing_mode_'; did you mean 'is_dosing_flag_'?
       if (this->dosing_mode_ && this->dosing_mode_->state != DOSING_MODE_NONE)
                                       ^~~~~~~~~~~~
                                       is_dosing_flag_
src/esphome/components/ezo_pmp/ezo_pmp.cpp:283:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'dosing_mode_'; did you mean 'is_dosing_flag_'?
         this->dosing_mode_->publish_state(DOSING_MODE_NONE);
               ^~~~~~~~~~~~
               is_dosing_flag_
*** [/data/hydro1-tomato-station/.pioenvs/hydro1-tomato-station/src/esphome/components/ezo_pmp/ezo_pmp.cpp.o] Error 1
========================== [FAILED] Took 4.37 seconds ==========================
```

stevewull, 2023-03-11:
I notice that this was reported in issue #3832.  As per temporary fix discussed there, I added the following:

binary_sensor:
  - platform: ezo_pmp
    pump_state:
      id: pump_state
      name: Pump State      
  - platform: ezo_pmp
    is_paused:
      id: is_paused
      name: Is Paused  

The code now compiles.

V4l3n10, 2023-03-25:
Guys, 
Have any of you managed to set up a multi-ezo_pmp configuration? A configuration example would be much appreciated.

ssieb, 2023-03-25:
@V4l3n10, that is not related to this issue and github issues are not for support requests.  Ask on the discord server or one of the forums.

ssieb, 2023-03-25:
@V4l3n10, also there is literally an example right at the top of this issue...

ssieb, 2023-03-25:
This is a duplicate of https://github.com/esphome/issues/issues/3832.

closed 2023-03-25 by ssieb