esphome/issues #4298: error when want install "WiFi Signal Sensor" in esphome

lysiong, 2023-03-10:
### The problem

ESPHOME always come out error when want install "WiFi Signal Sensor"

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.3.3

### What platform are you using?

ESP32

### Board

generic

### Example YAML snippet

```yaml
INFO Reading configuration /config/esphome/esp32-proxy1.yaml...
INFO Configuration is valid!
substitutions:
  name: esp32-proxy1
  friendly_name: Bluetooth Proxy
esphome:
  name: esp32-proxy1
  friendly_name: Bluetooth Proxy
  name_add_mac_suffix: false
  project:
    name: esphome.bluetooth-proxy
    version: '1.0'
  build_path: .esphome/build/esp32-proxy1
  platformio_options: {}
  includes: []
  libraries: []
  min_version: 2023.2.4
esp32:
  board: esp32dev
  framework:
    version: 4.4.2
    sdkconfig_options: {}
    advanced:
      ignore_efuse_mac_crc: false
    source: ~3.40402.0
    platform_version: platformio/espressif32 @ 5.2.0
    type: esp-idf
  variant: ESP32
wifi:
  ap:
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  enable_btm: false
  enable_rrm: false
  networks:
  - ssid: wifi@HOME
    password: HoMe@WiFi30
    priority: 0.0
  use_address: esp32-proxy1.local
api:
[... 230 lines cut ...]
/config/esphome/.esphome/build/esp32-proxy1/src/esphome/components/api/api_connection.cpp:426: undefined reference to `esphome::sensor::Sensor::get_accuracy_decimals()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/esp32-proxy1/src/esphome/components/api/api_connection.cpp:427: undefined reference to `esphome::sensor::Sensor::get_device_class[abi:cxx11]()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/esp32-proxy1/src/esphome/components/api/api_connection.cpp:429: undefined reference to `esphome::sensor::Sensor::get_state_class()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o:(.literal._Z5setupv+0xb8): undefined reference to `vtable for esphome::wifi_signal::WiFiSignalSensor'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o:(.literal._Z5setupv+0xbc): undefined reference to `vtable for esphome::wifi_signal::WiFiSignalSensor'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o:(.literal._Z5setupv+0x104): undefined reference to `esphome::sensor::Sensor::Sensor()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o:(.literal._Z5setupv+0x108): undefined reference to `esphome::sensor::Sensor::set_device_class(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o:(.literal._Z5setupv+0x10c): undefined reference to `esphome::sensor::Sensor::set_state_class(esphome::sensor::StateClass)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o:(.literal._Z5setupv+0x110): undefined reference to `esphome::sensor::Sensor::set_unit_of_measurement(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o:(.literal._Z5setupv+0x114): undefined reference to `esphome::sensor::Sensor::set_accuracy_decimals(signed char)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o: in function `esphome::safe_mode::SafeModeButton::SafeModeButton()':
/config/esphome/.esphome/build/esp32-proxy1/src/esphome/components/safe_mode/button/safe_mode_button.h:10: undefined reference to `esphome::sensor::Sensor::Sensor()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o: in function `setup()':
/config/esphome/.esphome/build/esp32-proxy1/src/main.cpp:215: undefined reference to `esphome::sensor::Sensor::set_device_class(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/esp32-proxy1/src/main.cpp:216: undefined reference to `esphome::sensor::Sensor::set_state_class(esphome::sensor::StateClass)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/esp32-proxy1/src/main.cpp:217: undefined reference to `esphome::sensor::Sensor::set_unit_of_measurement(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/main.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()':
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:657: undefined reference to `esphome::sensor::Sensor::set_accuracy_decimals(signed char)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/esphome/core/controller.o:(.literal._ZN7esphome10Controller16setup_controllerEb+0x8): undefined reference to `esphome::sensor::Sensor::add_on_state_callback(std::function<void (float)>&&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esp32-proxy1/.pioenvs/esp32-proxy1/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/config/esphome/.esphome/build/esp32-proxy1/src/esphome/core/controller.cpp:29: undefined reference to `esphome::sensor::Sensor::add_on_state_callback(std::function<void (float)>&&)'
collect2: error: ld returned 1 exit status
*** [/data/esp32-proxy1/.pioenvs/esp32-proxy1/firmware.elf] Error 1
========================= [FAILED] Took 28.67 seconds =========================
```

ssieb, 2023-03-10:
Do a clean build.

lysiong, 2023-03-10:
> Do a clean build.

I been try before to uninstall the entire ESPhome module & delete the ESPhome folder in home assistant, restart HA & install back ESPhome but seem like problem still occur. I also try change another ESP32 board. I not sure if my code is wrong but I just follow the official documentation from the ESPhome.io website.

lysiong, 2023-03-10:
> Try the clean build option.

can I have the actual step? I not so expert in this thing. where do I find the clean build option?

lysiong, 2023-03-10:
https://esphome.github.io/bluetooth-proxies/
above is the website that I use to built the bluetooth proxy

ssieb, 2023-03-10:
That's where you got the initial install, but that's not where you're doing the compiling.  There should be a menu on the dashboard card for the device that has a "clean" option on it.

ssieb, 2023-03-10:
Yes, that.  You still get the same error after that?

lysiong, 2023-03-10:
> Yes, that. You still get the same error after that?

yes, same error come out.

lysiong, 2023-03-10:
`substitutions:
  name: "esp32-proxy1"
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: xxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s`

I add the wifi signal sensor in this particular yaml file, I not sure if I add it correctly.

rleongcs, 2023-03-30:
> Try the clean build option.

I was having similar issue with my BT Proxy after I decided to add "sensor:" to the code. No matter what I did, I got the error. Landed on this page from google search and @ssieb you saved my day ;) Now it compiles properly with no errors!

@lysiong I am not sure if the indentations has anything to do with it but below is how I have mine.
```
sensor:
- platform: uptime
  name: Uptime Sensor
```

lysiong, 2023-04-07:
> 

May I know that sensor is it add it to ESP.yaml file? maybe I add the sensor to wrong file :(
below is my esp.yaml file content. 

`substitutions:
  name: "esp32-proxy1"
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: xxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

#sensor:
  # -platform: wifi_signal
  #    name: "WiFi Signal Sensor"
  #    update_interval: 60s

sensor:
- platform: uptime
  name: Uptime Sensor`

lysiong, 2023-04-11:
@rleongcs 
I add the sensor in here
[image]
is this the correct place that I add the uptime sensor & other sensor as well?

lysiong, 2023-04-26:
Problem solved ! proper procedure is click "clean build files", then click install. It work on latest version of ESPHOME

[image]

closed 2023-04-26 by ssieb