esphome/issues #4300: After a while (ESPHome update!?) - ESP32 "Connection error occurred"

amagr0, 2023-03-10:
### The problem

With the same code, same components and same connections I started to get this error intermittently.

### Which version of ESPHome has the issue?

ESPHome 2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.2.5

### What platform are you using?

ESP32

### Board

D1 mini ESP32

### Example YAML snippet

```yaml
esphome:
  name: vmc-particles
  friendly_name: vmc-particles

esp32:
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "fo5iwa0cXPhi6ANlz/vdEM/7p1J5XmC0O2Z/C29PAUE="

ota:
  password: "358e1f3fba530dba74cfe0aa3f29e2d7"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Vmc-Particles Fallback Hotspot"
    password: "A2E7c2VwPErE"

web_server:
  local: true

captive_portal:

i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a
    
sensor:
  - platform: sen5x
    id: vmc_sen55
    pm_1_0:
      name: "VMC PM <1µm Weight concentration"
      id: vmc_pm_1_0
      accuracy_decimals: 1
    pm_2_5:
      name: "VMC PM <2.5µm Weight concentration"
      id: vmc_pm_2_5
      accuracy_decimals: 1
    pm_4_0:
      name: "VMC_PM <4µm Weight concentration"
      id: vmc_pm_4_0
      accuracy_decimals: 1
    pm_10_0:
      name: "VMC PM <10µm Weight concentration"
      id: vmc_pm_10_0
      accuracy_decimals: 1
    temperature:
      name: "VMC OUT Temperature"
      accuracy_decimals: 1
      filters:
        - offset: 0.00
    humidity:
      name: "VMC OUT Humidity"
      accuracy_decimals: 0
      filters:
        - offset: 0.00
    voc:
      name: "VMC VOC"
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    nox:
      name: "VMC NOX"
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    temperature_compensation:
      offset: 0
      normalized_offset_slope: 0
      time_constant: 0
    acceleration_mode: low
    store_baseline: true
    address: 0x69
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
WARNING vmc-particles.local: Connection error occurred: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 520, in send_message_await_response_complex
    await fut
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/dist-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 303, in _keep_alive_loop
    await self._ping()
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 628, in _ping
    await self.send_message_await_response(PingRequest(), PingResponse)
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 537, in send_message_await_response
    res = await self.send_message_await_response_complex(
  File "/usr/local/lib/python3.9/dist-packages/aioesphomeapi/connection.py", line 522, in send_message_await_response_complex
    raise TimeoutAPIError(
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for response for <class 'api_pb2.PingRequest'> after 10.0s
INFO Disconnected from ESPHome API for vmc-particles.local
WARNING Disconnected from API
INFO Successfully connected to vmc-particles.local
```

jrames, 2023-03-11:
Same issue here, with the same ESPHome and HA versions. In my case the ESP32 seems to be failing to connect to the Bluetooth device I'm attempting to control.

The interesting thing is that it runs fine if I use the ESPHome CLI for MacOS to flash and run it. But once I've flashed it, and I move it to a different power source than the computer's USB port, it starts to fail with this error. Any similarities to your experience?

amagr0, 2023-03-15:
In my situation, the ESP32 failed regardless of the power source. 
I switched to an ESP8266 and solved the problem.