esphome/issues #4305: Host Name Resolution and ESP Password Issues

ZhuDaHai, 2023-03-13:
### The problem

ESP8266 with DS18B20 sensor installed and working with Initial (local) install via PC over USB. 
Installed in remote location and was working OK. (Should have left along).
Wanted to change Host name from epshome-web-xxxxxx to more meaningful name of cabinettemp.
Set its static IP address to is MAC address in pfSense DHCP Server along with its Hostname (cabinettemp) in DNS resolver both under my custom local domain ("custom") and "local" as an alias to that hostname.
on my PC (on the name subnet) from the CMD prompt:
ping cabinettemp.custom -> 192.168.1.53
ping cabinettemp.local -> 192.168.1.53

In my esphome-web-xxxxxx.yaml, I changed:
```
esphome:
  name: cabinettemp
```
when I try to compile and upload, it reports that it cannot resolve cabinettemp.local 
My PC and HomeAssistant with ESPHome are on the name subnet (192.168.1.x)

When I add:
wifi: 
   domain: .custom

It DOES resolve cabinettemp.custom and tries to upload, but reports "ERROR ESP requests password, but no password given!"

The only passwords in the yaml file are for the "wifi:" and  "ap:" - I did NOT change either of them!

If I set the manual_ip in the ESP......yaml, it works fine: connects, uploads, runs...

So. 2 Problems:
hostname.local does not resolve.
hostname.custom resolves, but breaks with a "password required"

What is going wrong?

### Which version of ESPHome has the issue?

Current version: 2023.2.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Home Assistant 2023.3.3

### What platform are you using?

ESP8266

### Board

nodemcu esp01_1m

### Component causing the issue

Host name resolution

ZhuDaHai, 2023-03-13:
Some more info:
The sensor does NOT show up the in ESPHome Integration. So when I add it manually, I specify "cabinettemp.local" as the host and it Magically resolves and works! (Remember: cabinettemp.local does not resolve when uploading to the esp board from the ESPHome Web UI

ssieb, 2023-03-13:
If you put `.local` in DNS, you will break mdns lookups, which is where part of your problems are.  If you change the name in the config, then you have to have some other method to locate the device because it will not be found with the new name until you flash that modified config.  The easiest solution is the `use_address:` option.  If you put `.local` in your DNS, then things that properly look that up using mdns will fail (e.g. the dashboard), but anything that tries DNS for it will work.

ZhuDaHai, 2023-03-14:
OK. Used the use_address to flash the new config. Then later I changed it back to cabinettemp and let it default to .local and it found and flashed the config again. 
I think I remember having to put the .local alias in the DNS Resolver a while back (like a year or 2) just to get ESPHome to find and flash the config OTA. But that was a few releases ago - so I guess not applicable now.
Thanks. I will close this issue

closed 2023-03-14 by ZhuDaHai