esphome/issues #4307: Wavshare ePaper 2.9" Rev2.1 - Timeout while displaying image! on full refresh

AlSa-Her, 2023-03-15:
### The problem

I have one of the newer 2.9 inch V2 e-ink devices (v2 sticker with rev2.1) and I'm trying to make it work with ESPHome.
Partial update works fine (with slight haze, as to be expected), but full update of the display leaves it blinking 6 times over 5+ second.
I have been searching github for similar problems but have found none with the exact same. The only similar issue I found is:
https://github.com/esphome/issues/issues/1462 (however, my display basically works - just with an annoying full refresh)

### Which version of ESPHome has the issue?

Version: 2023.2.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

currently not connected

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4

### Component causing the issue

Waveshare 2.90inv2

### Example YAML snippet

```yaml
esphome:
  name: xbo-controller
  friendly_name: xbo-controller

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

font:
  - file: 'fonts/GoogleSans-Bold.ttf'
    id: font_googlesans_bold_50
    size: 50

spi:
  clk_pin: 33
  mosi_pin: 32

display:
  - platform: waveshare_epaper
    dc_pin: 26
    cs_pin: 25
    #busy_pin: 14
    reset_pin: 27
    model: 2.90inV2
    reset_duration: 200ms
    update_interval: 2 sec
    full_update_every: 10
    pages:
      - id: boot
        lambda: |-
          it.line(0, 0, 50, 50);
          it.print(148, 64, id(font_googlesans_bold_50), TextAlign::CENTER, "Booting...");
      - id: running
        lambda: |-
          it.line(0, 0, 50, 50);
          it.print(148, 64, id(font_googlesans_bold_50), TextAlign::CENTER, "Running...");
```

### Anything in the logs that might be useful for us?

```txt
13:25:15 | [E] | [waveshare_epaper:119] | Timeout while displaying image!
```

### Additional information

Things tried (among others):
- setting `reset_duration to: 1 / 2 / 200 / 2000ns`
- trying the modified driver component according to https://github.com/esphome/issues/issues/2334 , which just broke the device more... (grainy display after refresh with slowly fading)
```
#external_components:
#  - source:
#      type: git
#      url: https://github.com/velaar/esphome
#      ref: dev
#    components: [ waveshare_epaper, display]
```
- various `update_interval` and `full_update_every` settings, which just postpones the issue to a later time.
- using `2.90in` and `2.90in-b` as model left the device unresponsive

Setting `update_interval` and `full_update_every` very high and only refreshing with `component_update` postpones the problem but leaves shadows and is only a very annoying workaround....
I am not sure if I am just missing something stupid but would be very thankful for any help!

mischief, 2023-07-05:
i unfortunately cannot get my epaper display to work under esphome. i believe the module OP refers to is the same one i have: https://www.waveshare.com/product/displays/e-paper/epaper-2/2.9inch-e-paper-module.htm which also apparently uses the UC8151D controller.

i have found that https://github.com/ZinggJM/GxEPD2 works fine with this display under platformio with a wemos d1 mini clone, but i can't make it work in esphome.

here is my code for each test case:
- GxEPD2: https://gist.github.com/mischief/2a07386956b774e330ffc5b2f745e382
- esphome: https://gist.github.com/mischief/5e6c5bad266bdf429282ab04d4fc4bbc

the wiring for both configurations is the same; the epaper module is directly connected to my d1 mini and uses 3.3v for module vcc.

additionally, i found setting the busy pin will produce a timeout under esphome as in #1462, but not under GxEPD2.

with GxEPD2 the display shows hello world, and with esphome does not.

Kaibob2, 2023-07-05:
@mischief You have to use a custom_component to make this module work. Please read this thread here: https://github.com/esphome/issues/issues/2334 for further information

The snippet below worked until ESPhome 2023.6.0, afterwards it's gets a little tricky with the custom_component, but it's described in the above mentioned issue:

```
external_components:
  - source:
      type: git
      url: https://github.com/velaar/esphome
      ref: dev
    components: [ waveshare_epaper, display]

spi:
  clk_pin: 13
  mosi_pin: 14

display:
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 2.90inv2
    reset_duration: 2ms
    full_update_every: 200
    update_interval: 30s
    rotation: 90
```

Kaibob2, 2023-07-21:
As there are quite a few `display` related breaking changes in 2023.7.0, has anybody using the waveshare ePaper already updated to this version and can confirm that it's still working?

stephenjamieson, 2023-07-21:
It does not compile no

Kaibob2, 2023-07-21:
That's what i expected 😮‍💨

velaar, 2023-07-22:
> 

> It does not compile no

Sadly it doesn't. I'm not savvy enough to make the fix myself :( as it doesn't seem that all the breaking changes were documented.  I will update the custom component if I find a fix and will report back

Leonhard67, 2023-08-03:
häng mich hier auch mal an, bekomme das 2.9 Display auch nicht zum laufen unter ESP-Home .....
[image]

boyfromgermany, 2024-01-21:
> häng mich hier auch mal an, bekomme das 2.9 Display auch nicht zum laufen unter ESP-Home ..... ![epaperfehler](https://private-user-images.githubusercontent.com/112473930/258205391-3ff5b2e4-29cf-4b61-8f1f-2dff96436b6d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU4NDQ3NjMsIm5iZiI6MTcwNTg0NDQ2MywicGF0aCI6Ii8xMTI0NzM5MzAvMjU4MjA1MzkxLTNmZjViMmU0LTI5Y2YtNGI2MS04ZjFmLTJkZmY5NjQzNmI2ZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyMVQxMzQxMDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNGVhMTQ3OTE5NDk2Y2I0NTI5MTUyZDNkMDQ0NGVmMTRlZDUzZGY2ZTJkNzg4MjJhNWE5NDg4YTFkYWNmNTQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.vjadLYyGiUmjKWqA8QuIaM5ui_x_iUE7QGimmNWUOhU)

have you found a solution i also have the same with red