esphome/issues #4310: Restarts and unstable connection 2023.3 update

radek-kafka, 2023-03-16:
### The problem

Hi,

my M5Atom is restarting after the update to new version 2023.3 today.. I have tried multiple reboots but still same.. Lot of disconnects from WiFi, crashes etc.. 

See the logs .. 
Many thanks
Radek

### Which version of ESPHome has the issue?

2023.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2

### What platform are you using?

ESP32

### Board

M5AtomLite

### Example YAML snippet

```yaml
esphome:
  name: "atom-bluetooth-proxy"

dashboard_import:
    package_import_url: github://esphome/bluetooth-proxies/m5stack-atom-lite.yaml@main

esp32:
  board: m5stack-atom
  framework:
    type: arduino

substitutions:
  devicename: m5atom_ir
  upper_devicename: M5Atom IR
  mqtt_broker: 192.168.88.21

bluetooth_proxy:
  active: false

# Enable logging

logger:
  logs:
    homeassistant.components.bluetooth: debug
    

# Enable Home Assistant API
api:
  encryption:
    key: "removed here ="

ota:
  password: "removed"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
[... 700 lines cut ...]
[09:35:34][C][wifi:367]:   Hostname: 'atom-bluetooth-proxy'
[09:35:34][C][wifi:369]:   Signal strength: -73 dB ▂▄▆█
[09:35:34][C][wifi:373]:   Channel: 1
[09:35:34][C][wifi:374]:   Subnet: 255.255.255.0
[09:35:34][C][wifi:375]:   Gateway: 192.168.88.1
[09:35:34][C][wifi:376]:   DNS1: 192.168.88.1
[09:35:34][C][wifi:377]:   DNS2: 0.0.0.0
[09:35:34][D][wifi:527]: Disabling AP...
[09:35:34][I][mqtt:176]: Connecting to MQTT...
[09:35:34][I][mqtt:216]: MQTT Connected!
[09:36:02][W][mqtt:264]: MQTT Disconnected: TCP disconnected.
[09:36:02][I][mqtt:176]: Connecting to MQTT...
[09:36:03][D][sensor:127]: 'M5Atom IR Uptime': Sending state 55.94300 s with 0 decimals of accuracy
[09:36:21][W][mqtt:264]: MQTT Disconnected: TCP disconnected.
[09:36:21][I][mqtt:176]: Connecting to MQTT...
[09:36:21][D][sensor:127]: 'M5Atom IR WiFi Signal': Sending state -73.00000 dBm with 0 decimals of accuracy
[09:36:39][W][mqtt:264]: MQTT Disconnected: TCP disconnected.
[09:36:39][I][mqtt:176]: Connecting to MQTT...
[09:36:58][W][mqtt:264]: MQTT Disconnected: TCP disconnected.
[09:36:58][I][mqtt:176]: Connecting to MQTT...
[09:37:03][D][sensor:127]: 'M5Atom IR Uptime': Sending state 115.94300 s with 0 decimals of accuracy
[09:37:16][W][mqtt:264]: MQTT Disconnected: TCP disconnected.
[09:37:16][I][mqtt:176]: Connecting to MQTT...
[
```

kev300, 2023-03-22:
The duplicate connection seems to be unrelated.

But now I can't downgrade the ESP anymore. Clean re-compile with 2023.2.4 and reflash does not fix the instabillity. The ESP32 keeps crashing after random time frames. Sometimes 10 minutes, sometimes a few seconds. 😞

radek-kafka, 2023-03-22:
Mine is after last updates stable. I do not see messages like before. 2023.3.1 looks to me better .

ptahbe, 2023-03-29:
I'm having a similar issue. Since a certain update (don't remember which).
In ESPHome it shows online, but nothing works. I can't ping it, I get no data in HA and the web interface doesn't work.

In the logs I get this 
----- start -----
[I][mqtt:216]: MQTT Connected!

abort() was called at PC 0x401e53d7 on core 1

Backtrace:0x40083725:0x3fff56f00x400930e5:0x3fff5710 0x400986a9:0x3fff5730 0x401e53d7:0x3fff57b0 0x401e541e:0x3fff57d0 0x401e6501:0x3fff57f0 0x401e4e10:0x3fff5810 0x401e57ad:0x3fff5830 0x401e5e12:0x3fff5850 0x401e5e45:0x3fff5880 0x400dafef:0x3fff58a0 0x400db09e:0x3fff5910 0x400da7f2:0x3fff5960 0x400eea25:0x3fff59a0 0x401ff42e:0x3fff5a10 0x400ef7c6:0x3fff5a40 0x400ee7a6:0x3fff5a90 0x400ee7be:0x3fff5b00 0x401e8581:0x3fff5b20 0x401e85fd:0x3fff5b50 0x401e8de6:0x3fff5b70 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:258]: Log initialized
[C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
----- end -----

After 10 unsuccessful boot attempts I get nothing more in the logs. But sometime I get this after the 10th boot attempt :

----- start -----
[C][ota:469]: There have been 10 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[E][ota:476]: Boot loop detected. Proceeding to safe mode.
----- end -----

Then I can ping it, but nothing else works.

ESPHome is currently on 2023.3.2. I have tried it with previous versions also. I hoped it would have been fixed. I just did a clean build again, but still the same problem.

I'm using an ESP-WROOM-32 board and this is the config :
esphome:
  name: bureau
  platform: ESP32
  board: esp32dev

Is that still correct? It has worked before, but maybe something changed. I read in a post somewhere that the person changed it to edp-idf and that fixed it, but I like to be certain before I change anything.

kev300, 2023-03-29:
My ESP-WROOM-32 is now stable with 2023.3.2.
I guess there was something not re-compiled correctly. I'm using esphome in a docker container, so I removed the container, all related images and cleaned cached stuff multiple times. Including the config directory (only left the devoce yaml files there).
Then rebuilt and reflashed multiple times with 2023.2.4 until it was stable again. Then did the same with 2023.3.2 and it works now.

ptahbe, 2023-03-29:
I'm running ESPHome under Home Assistant. I already did "Clean build files" several times. Do you know how I can clean up ESPHome to re-compile everything?

kev300, 2023-03-29:
Mh no experiance with that. You could try to delete all files below config/esphome/ (except your yaml config files)
the command pio system prune was also helpfull in the docker version. But I don't know if the HA integrated version has a command line interface? As a last resort you could try to remove /reinstall the add-on

RenierM26, 2023-04-21:
Hi There,

The interval and window length does not play well with Wi-Fi. Current recommendation is to leave the interval and window on the defaults for Wifi ESP controllers.

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true  
```

Change to these defaults:

```
esp32_ble_tracker:
  scan_parameters:
    interval: 320ms
    window: 30ms
    active: true
```

Hope it helps.

THX2112, 2023-05-09:
I'm having the exact problem. Several ESP32 devices, all identical, and one stopped working after updating to latest ESPHome. No error messages. Data shows up on terminal monitor but HA can't get any data.  A clean install to a new device does nothing.

I also noticed that ESPHome no longer cleans up unused devices. When changing the name to debug, the previous ESPHome device is still in the HA logs even after deleting the ESPHome device.