esphome/issues #4311: Trying to use rf_bridge to control my novy hood

Bjoern3003, 2023-03-16:
### The problem

Hi there, I have tried a lot of time, to fix this problem by myself

I have a Novy Hood and a Novy 840029 RF Controller.

I have flashed ESP-Home on my Sonoff RF Bridge and installed it in Home-Assistant with ESP-Home Addon.

Every seems looking fine, but I can not controll my Hood with ESP-Home.

I think the protocol is not correct.

I have found another Project in FHEM, where I can see the complete Informations for the protocol: https://github.com/RFD-FHEM/RFFHEM/blob/491b4b35a430a3b83f81d29d6e7f20390c02e173/FHEM/lib/SD_ProtocolData.pm#L2370

But I don't understand, how I can translate these defines in the ESP-Home yaml code. 

Attached the Log File, when I press the light toggle Button on the Remote Control.

I have attached my yaml file, with example, that I have tried.

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.4

### What platform are you using?

ESP8266

### Board

Sonoff RF Bridge

### Example YAML snippet

```yaml
remote_receiver:
  pin: GPIO04
  dump: rc_switch
  #dump: all
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb

remote_transmitter:
  pin: GPIO05
  carrier_duty_percent: 100%

# this will log received commands, and can also transmit. Read up here:
# https://esphome.io/components/remote_transmitter.html#remote-setting-up-rf
# https://github.com/RFD-FHEM/RFFHEM/blob/491b4b35a430a3b83f81d29d6e7f20390c02e173/FHEM/lib/SD_ProtocolData.pm#L2370

button:
  - platform: template
    name: Licht Toggle
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '1011001011001011001011001011001001001011001011011011001'
          protocol: 1
          repeat:
            times: 4
            wait_time: 20ms
  - platform: template
    name: Licht Toggle v2
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010101010111010001'
          protocol: 6
          repeat:
            times: 4
            wait_time: 20ms
  - platform: template
    name: Licht Toggle v3
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '011101000100'
          protocol: 6
          repeat:
            times: 4
            wait_time: 20ms

  - platform: template
    name: Licht Toggle v4
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '10101010011101000111'
          protocol: 6
          repeat:
            times: 4
            wait_time: 20ms

  - platform: template
    name: Licht Toggle v5
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '1010101011010001'
          protocol: 6
          repeat:
            times: 4
            wait_time: 20ms

  - platform: template
    name: Licht Toggle v6
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '1010101001110111'
          protocol: 6
          repeat:
            times: 4
            wait_time: 20ms
  - platform: template
    name: Licht Toggle v7
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '0111010001'
          protocol: 6
          repeat:
            times: 4
            wait_time: 20ms
```

### Anything in the logs that might be useful for us?

```txt
[15:41:39][D][remote.rc_switch:261]: Received RCSwitch Raw: protocol=6 data='010101010111010001'
```

### Additional information

[https://github.com/RFD-FHEM/RFFHEM/blob/491b4b35a430a3b83f81d29d6e7f20390c02e173/FHEM/lib/SD_ProtocolData.pm#L2370](https://github.com/RFD-FHEM/RFFHEM/blob/491b4b35a430a3b83f81d29d6e7f20390c02e173/FHEM/lib/SD_ProtocolData.pm#L2370)

[https://github.com/RFD-FHEM/RFFHEM/issues/331](https://github.com/RFD-FHEM/RFFHEM/issues/331)

pvh0, 2023-08-28:
> ### The problem
> Hi there, I have tried a lot of time, to fix this problem by myself
> 
> I have a Novy Hood and a Novy 840029 RF Controller.
> 
> I have flashed ESP-Home on my Sonoff RF Bridge and installed it in Home-Assistant with ESP-Home Addon.
> 
> Every seems looking fine, but I can not controll my Hood with ESP-Home.
> 
> I think the protocol is not correct.
> 
> I have found another Project in FHEM, where I can see the complete Informations for the protocol: https://github.com/RFD-FHEM/RFFHEM/blob/491b4b35a430a3b83f81d29d6e7f20390c02e173/FHEM/lib/SD_ProtocolData.pm#L2370
> 
> But I don't understand, how I can translate these defines in the ESP-Home yaml code.
> 
> Attached the Log File, when I press the light toggle Button on the Remote Control.
> 
> I have attached my yaml file, with example, that I have tried.
> 
> ### Which version of ESPHome has the issue?
> 2023.3.0
> 
> ### What type of installation are you using?
> Home Assistant Add-on
> 
> ### Which version of Home Assistant has the issue?
> 2023.3.4
> 
> ### What platform are you using?
> ESP8266
> 
> ### Board
> Sonoff RF Bridge
> 
> ### Component causing the issue
> _No response_
> 
> ### Example YAML snippet
> ```yaml
> remote_receiver:
>   pin: GPIO04
>   dump: rc_switch
>   #dump: all
>   tolerance: 50%
>   filter: 250us
>   idle: 4ms
>   buffer_size: 2kb
> 
> remote_transmitter:
>   pin: GPIO05
>   carrier_duty_percent: 100%
> 
> # this will log received commands, and can also transmit. Read up here:
> # https://esphome.io/components/remote_transmitter.html#remote-setting-up-rf
> # https://github.com/RFD-FHEM/RFFHEM/blob/491b4b35a430a3b83f81d29d6e7f20390c02e173/FHEM/lib/SD_ProtocolData.pm#L2370
> 
> button:
>   - platform: template
>     name: Licht Toggle
>     on_press:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '1011001011001011001011001011001001001011001011011011001'
>           protocol: 1
>           repeat:
>             times: 4
>             wait_time: 20ms
>   - platform: template
>     name: Licht Toggle v2
>     on_press:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '010101010111010001'
>           protocol: 6
>           repeat:
>             times: 4
>             wait_time: 20ms
>   - platform: template
>     name: Licht Toggle v3
>     on_press:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '011101000100'
>           protocol: 6
>           repeat:
>             times: 4
>             wait_time: 20ms
> 
>   - platform: template
>     name: Licht Toggle v4
>     on_press:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '10101010011101000111'
>           protocol: 6
>           repeat:
>             times: 4
>             wait_time: 20ms
> 
>   - platform: template
>     name: Licht Toggle v5
>     on_press:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '1010101011010001'
>           protocol: 6
>           repeat:
>             times: 4
>             wait_time: 20ms
> 
>   - platform: template
>     name: Licht Toggle v6
>     on_press:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '1010101001110111'
>           protocol: 6
>           repeat:
>             times: 4
>             wait_time: 20ms
>   - platform: template
>     name: Licht Toggle v7
>     on_press:
>       - remote_transmitter.transmit_rc_switch_raw:
>           code: '0111010001'
>           protocol: 6
>           repeat:
>             times: 4
>             wait_time: 20ms
> ```
> 
> ### Anything in the logs that might be useful for us?
> ```
> [15:41:39][D][remote.rc_switch:261]: Received RCSwitch Raw: protocol=6 data='010101010111010001'
> ```
> 
> ### Additional information
> https://github.com/RFD-FHEM/RFFHEM/blob/491b4b35a430a3b83f81d29d6e7f20390c02e173/FHEM/lib/SD_ProtocolData.pm#L2370
> 
> [RFD-FHEM/RFFHEM#331](https://github.com/RFD-FHEM/RFFHEM/issues/331)

@Bjoern3003 Did you ever find a solution for this? Looking for the same.

Bjoern3003, 2023-08-28:
No, unfortunately not.

bart1, 2023-11-08:
Thanks to this issue and the `RFD-FHEM`, I have figured out the correct setup for my ATAG hood. I have a transmitter that reads on the back `SF01 01319004 433.92 MHz`, I think this is the same as some of the other hoods. After some hints in the various topics and playing around with a logic analyzer I have found the following settings to work for toggling the light:
```
remote_transmitter:
  pin: GPIO21
  carrier_duty_percent: 50%

button:
  - platform: template
    name: Licht Toggle
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '111000001000101110'
          protocol:
            pulse_length: 350 
            sync: [44, 1] 
            zero: [1, 2] 
            one: [2, 1] 
            inverted: True
          repeat:
            times: 10
            wait_time: 0ms
```
I use a very simple transmitter (`FS1000A`) connected to an esp. Special thanks goes out to @sidey79 in the other topic. 
I hope this is helpful for someone. I will try to decode the other signals soon

bart1, 2023-11-08:
A quick follow up for the ATAG hood these are the codes:
```
111000001000101110 -> toggle light
111000001000101110 -> repeat a 100 times to dim
111000001000101001 -> timer
111000001000101100 -> + extracting
111000001000101010 -> - extracting
111000001000101101 -> turn off
```

pvh0, 2023-11-09:
> A quick follow up for the ATAG hood these are the codes:
> 
> ```
> 111000001000101110 -> toggle light
> 111000001000101110 -> repeat a 100 times to dim
> 111000001000101001 -> timer
> 111000001000101100 -> + extracting
> 111000001000101010 -> - extracting
> 111000001000101101 -> turn off
> ```

Thanks Bart, will give it a try here with my Novy as well. It did not work with me but had inverted to false so will check again!

bart1, 2023-11-09:
@pvh0 also note it did not work for me with the wait time at 20ms, it started working when I lowered it to 0

bart1, 2023-11-09:
PS this video helped me to decode the signal (and measure if I reproduce it correctly): https://www.youtube.com/watch?v=V7TzMmDZjAY

Bjoern3003, 2023-11-09:
Interesting solution.

I have sold my Sonoff RF Bridge, because of no results.

Now I have bought Arduino 433Mhz Set to test again with your solution @bart1

pvh0, 2023-11-09:
> PS this video helped me to decode the signal (and measure if I reproduce it correctly): https://www.youtube.com/watch?v=V7TzMmDZjAY

Just ordered the logic analyser. Could you please explain how you got from the Youtube movie (Pulseview) to 
`"# 111000001000101110 -> toggle light"?  `

bart1, 2023-11-09:
> Just ordered the logic analyser. Could you please explain how you got from the Youtube movie (Pulseview) to `"# 111000001000101110 -> toggle light"? `

Using the circuit from the video (link with drawings is in the description) I made a recording while I pressed the button on the existing remote (I think I used a sampling rate of 4 MHz). In this recording I can then decode the signal manually (using the encoding from the RFFHEM issue). Using the timing feature of pulseview you can just count the intervals and each time see if a 1 or 0 is encoded. Ideally the `remote_receiver` from esphome would do  this but that does not seem to get the encoding right.

pvh0, 2023-11-13:
I also analysed the codes and these seem to be the codes for my Novy hood:

`Plus   1010101010101
Light  1010101010111010001
Min    1010101010110
On/Off 1010101010111010011`

In the meantime I found this, which reflect my codes as well:
https://github.com/abelgomez/rf-mqtt-bridge/blob/master/src/NovyController.cpp

I tried these codes with the config from Bart, but unfortunately no luck:

`remote_transmitter:
  pin: GPIO21
  carrier_duty_percent: 50%

button:
  - platform: template
    name: Licht Toggle
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '111000001000101110'
          protocol:
            pulse_length: 350 
            sync: [44, 1] 
            zero: [1, 2] 
            one: [2, 1] 
            inverted: True
          repeat:
            times: 10
            wait_time: 0ms`

Codes are sent to the Novy but no response. Will see if I can play a bit with the wait time etc...
Maybe one more check for @bart1, do you really have carrier_duty_percent: 50%? Because ESPHome documentation says Set this to 50% if you’re working with IR LEDs and to 100% if working with other things like 433MHz transmitters...

bart1, 2023-11-13:
@pvh0 I indeed hat the carrier duty 50% maybe it works better with a 100% I did not test that. It works from atleast 3 meters away for me. Did you have a look if the resulting signal emitted from the esp looked the same in the logic analyzer as the one you recorded?

renedis, 2024-01-04:
> I also analysed the codes and these seem to be the codes for my Novy hood:
> 
> `Plus 1010101010101 Light 1010101010111010001 Min 1010101010110 On/Off 1010101010111010011`
> 
> In the meantime I found this, which reflect my codes as well: https://github.com/abelgomez/rf-mqtt-bridge/blob/master/src/NovyController.cpp
> 
> I tried these codes with the config from Bart, but unfortunately no luck:
> 
> `remote_transmitter: pin: GPIO21 carrier_duty_percent: 50%
> 
> button:
> 
> * platform: template
>   name: Licht Toggle
>   on_press:
>   
>   * remote_transmitter.transmit_rc_switch_raw:
>     code: '1010101010111010001'
>     protocol:
>     pulse_length: 350
>     sync: [44, 1]
>     zero: [1, 2]
>     one: [2, 1]
>     inverted: True
>     repeat:
>     times: 10
>     wait_time: 0ms`
> 
> Codes are sent to the Novy but no response. Will see if I can play a bit with the wait time etc... Maybe one more check for @bart1, do you really have carrier_duty_percent: 50%? Because ESPHome documentation says Set this to 50% if you’re working with IR LEDs and to 100% if working with other things like 433MHz transmitters...

Your code seems one character to long?

the code is setup as: DEVICE+PREFIX+BUTTON
DEVICE = 4 characters long
PREFIX = 4 characters long
BUTTON = specific code with variable length.

In your case you try to send the light button toggle.
Which is:
1010 1010 1 0111010001 (use without spaces of course)

But it should be:
1010 1010 0111010001 (use without spaces of course)

I think you used one "1" character to much when reading https://github.com/abelgomez/rf-mqtt-bridge/blob/master/src/NovyController.cpp

Also, i think you should find out which DEVICE code your remote uses in the 10 variables.

klenaers, 2024-06-17:
@pvh0 : If you're still looking for the correct parameters for the Novy hood(based on protocol 12 of RC-switch). I use the CC1101-specific module that you find in the dev-branch of esphome. These settings work for my configuration. Maybe also for the Sonoff RF-bridge.

```
  - platform: template
    name: "Light toggle"
    on_press:
      - remote_transmitter.transmit_rc_switch_raw_cc1101:
          code: '101001010111010001'
          protocol:
            pulse_length: 320
            sync: [36,1]
            zero: [1,2]
            one: [2,1]
            inverted: true
          repeat:
            times: 10
```
And this works with duty cycle 100%
```
remote_transmitter:
  pin: GPIO22  # GDO0
  carrier_duty_percent: 100%
```

pvh0, 2024-06-18:
Thanks, cool! Couple of questions:

- How did you get the code "101001010111010001" for light?
- Can you please share the other codes, e.g. on/off?
- Can you please share how you got cc1101 working by making use of the dev branch? I'm using Docker, should I just use the image esphome:dev?

> @pvh0 : If you're still looking for the correct parameters for the Novy hood(based on protocol 12 of RC-switch). I use the CC1101-specific module that you find in the dev-branch of esphome. These [...]

Bjoern3003, 2024-06-18:
For me its compiling fine like this:

`external_components:
  # use rtttl and dfplayer from ESPHome's dev branch in GitHub
  - source:
      type: git
      url: https://github.com/gabest11/esphome
      ref: cc1101
    components: [ cc1101 ]

esphome:
  name: test-dunstabzugshaube
  friendly_name: Test Dunstabzugshaube

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "znZHlj9OBF+BdaL3a+dlqDL/rlZDpmGbmOnDj3Dusc8="

ota:
    password: "153757fc18b1a44a6719eab48b3095f9"

wifi:
  ssid: !secret wifi_ssid_smarthome
  password: !secret wifi_password_smarthome
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.20.91
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.20.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Dunstabzugshaube"
    password: "lcuUpU2rrjv4"

spi:
  clk_pin: GPIO14
  miso_pin: GPIO12
  mosi_pin: GPIO13

cc1101:
  id: transceiver
  cs_pin: GPIO15
  gdo0_pin:
    number: D1
    allow_other_uses: true
  bandwidth: 200
  frequency: 433920
  rssi:
    name: RSSI
  lqi:
    name: LQI
  temperature:
    name: Temperature

remote_transmitter:
  pin:
    number: D1  # GDO0
    allow_other_uses: true
  carrier_duty_percent: 100%

remote_receiver:
  pin:
    number: D1  # GDO0
    allow_other_uses: true
  dump:
    - rc_switch
  # Settings to optimize recognition of RF devices
  tolerance: 50%
  buffer_size: 2kb
  filter: 250us
  idle: 4ms
  on_rc_switch:
    - logger.log:
        format: "on_rc_switch: %i %llu"
        args: [x.protocol, x.code]

button:
  - platform: template
    name: "Light toggle"
    on_press:
      - remote_transmitter.transmit_rc_switch_raw_cc1101:
          code: '101001010111010001'
          protocol:
            pulse_length: 320
            sync: [36,1]
            zero: [1,2]
            one: [2,1]
            inverted: true
          repeat:
            times: 10

captive_portal:
    `

I will order an cc1101 module to test it with my novy hood.

klenaers, 2024-06-18:
@pvh0 : this is the location where you find how to combine the code for the different actions: https://github.com/SpectraCoder/ESP32_Novy_Commander/blob/main/config.example.h

NOVY_DEVICE_CODE[channelIndex] + NOVY_PREFIX + NOVY_COMMAND_LIGHT
1010 (channel 6) + 0101 + 0111010001 = 101001010111010001

Bjoern3003, 2024-06-18:
Thank you @klenaers for me now its also working without cc1101 

Its working well with this:

```
remote_transmitter:
  pin:
    number: D1  # GDO0
  carrier_duty_percent: 100%

button:
  - platform: template
    name: "Licht schalten"
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010101010111010001'
          protocol:
            pulse_length: 320
            sync: [36,1]
            zero: [1,2]
            one: [2,1]
            inverted: true
          repeat:
            times: 10
  - platform: template
    name: "Haube schalten"
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010101010111010011'
          protocol:
            pulse_length: 320
            sync: [36,1]
            zero: [1,2]
            one: [2,1]
            inverted: true
          repeat:
            times: 10
  - platform: template
    name: "Lüfter schneller"
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010101010101'
          protocol:
            pulse_length: 320
            sync: [36,1]
            zero: [1,2]
            one: [2,1]
            inverted: true
          repeat:
            times: 10
  - platform: template
    name: "Lüfter langsamer"
    on_press:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010101010110'
          protocol:
            pulse_length: 320
            sync: [36,1]
            zero: [1,2]
            one: [2,1]
            inverted: true
          repeat:
            times: 10
```

brechtvhb, 2024-09-18:
@Bjoern3003 On what board did you get this running now?

I would also like to control my novy hood, but not sure what components to order.

Bjoern3003, 2024-09-18:
ESP8266 and a cheap 433MHz Transmitter from Aliexpress.