esphome/issues #4313: ESPhome 2023.2.4 have broken the template cover component

yeeOhai, 2023-03-16:
### The problem

 Seems ESPhome 2023.2.4 have broken the template cover component:
https://esphome.io/components/cover/template.html

I have used the template cover as my garage door for a few years. After updating ESPhome from 2022.12.8 to 2023.2.4, and allowed ESPhome 2023.2.4 to update the code in the garage door d1-mini, Home Assistant no longer sees the cover state change. Log showed the ultrasonic sensor still measured the distance correctly. Replaced the ultrasonic sensor with a new one did not fix the problem.

Using a backup, restore ESPhome to 2022.12.8, then regenerate code for the  d1-mini using ESPhome 2022.12.8, Home Assistant is able to see the cover state change again and everything works again.

Now, for my own solution, I have update ESPhome 2023.2.4 again, but leave the d1-mini code at the 2022.12.8 version. 

However, this bug needs to be fix so people using the template cover component in future won't be left wasting their time troubleshooting.

### Which version of ESPHome has the issue?

2023.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.5

### What platform are you using?

ESP8266

### Board

d1-mini

### Component causing the issue

https://esphome.io/components/cover/template.html

### Example YAML snippet

```yaml
substitutions:
  devicename: garagedoor
  description: OpenGarage.io Device
  friendly_name: Garage Door
  open_distance: "65" # less than or equal in cm, Yee garage distance is 55cm
  open_close_time: "20s" # allow 20s for motion to complete
  reporting_delta: "25" # minimum 25cm change

sensor:
  - platform: ultrasonic
    trigger_pin: D6
    echo_pin: D5
    update_interval: 5s
    unit_of_measurement: cm
    accuracy_decimals: 0
    pulse_time: 10us
    name: "${friendly_name} Distance"
    id: ${devicename}_distance
    timeout: 4.0m
    # apply a filter to change the distance to cm from metre #
    filters:
      - filter_out: nan  # filter timeouts
      - multiply: 100    # from meter to cm
      - delta: ${reporting_delta}
      - sliding_window_moving_average:
          window_size: 8
          send_every: 4
          send_first_at: 4
switch:
  - platform: gpio
    id: "garage_button"
    restore_mode: RESTORE_DEFAULT_OFF
    pin: D1
    inverted: true
    name: "Garage Button"
    internal: true
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: garage_button
    
  - platform: gpio
    id: "x10_alarm_garage_sensor"
    restore_mode: RESTORE_DEFAULT_OFF
    pin: RX #D2
    inverted: true
    name: "X10 Alarm Garage Sensor"
    #internal: true

cover:
  - platform: template
    name: "${friendly_name} Control"
    id: ${devicename}_control
    device_class: garage
    # set the distance so it is equal or less than $open_distance when open
    lambda: !lambda |-
      if (id(${devicename}_distance).state <= ${open_distance}) {
        if (!(id(x10_alarm_garage_sensor).state)) {
            id(x10_alarm_garage_sensor).turn_on();} 
        return COVER_OPEN;
      } else {
        if (id(x10_alarm_garage_sensor).state) {
           id(x10_alarm_garage_sensor).turn_off();}
        return COVER_CLOSED;
      }

    open_action:
      - if:
          condition:
            lambda: "return id(${devicename}_control).position == cover::COVER_CLOSED;"
          then:
            # open garage door
            - switch.turn_on: garage_button
            #- delay: ${open_close_time}

    close_action:
      - if:
          condition:
            lambda: "return id(${devicename}_control).position == cover::COVER_OPEN;"
          then:
            # close garage door
            - switch.turn_on: garage_button
            #- delay: ${open_close_time}
```

### Additional information

in the code above, x10_alarm_garage_sensor works correctly as long as the cover state is reported correctly by the ESPhome code.

eddyeal, 2023-03-30:
Same here with the template cover, which I use for my sunscreen.
Home Assistant no longer sees the cover entity. 
I only noticed it now because the sun is starting to shine again.

ESPhome version: 2023.3.2

MikeAg03, 2023-10-23:
Any updates on this?  It has broken my garage door code as well…