esphome/issues #4314: Bluetooth Proxy compile fail

MihaT2021, 2023-03-16:
### The problem

First install after esphome found new device failed.

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.1

### What platform are you using?

ESP32-IDF

### Board

esp32 WRoom-32

### Component causing the issue

CMakeTestCCompiler.cmake

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-f46cfd
  friendly_name: Bluetooth Proxy f46cfd
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: fcIP0YGEQd/I3dQezMsEWNHZEaZpGgOGtpoFjtd15Ok=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp32-bluetooth-proxy-f46cfd.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-bluetooth-proxy-f46cfd (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing esphome/Improv @ 1.2.3
INFO Installing esphome/Improv @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.3 has been installed!
INFO Improv@1.2.3 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/esp32-bluetooth-proxy-f46cfd/.pioenvs/esp32-bluetooth-proxy-f46cfd/CMakeFiles/CMakeOutput.log".
See also "/data/esp32-bluetooth-proxy-f46cfd/.pioenvs/esp32-bluetooth-proxy-f46cfd/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/esp32-bluetooth-proxy-f46cfd/.pioenvs/esp32-bluetooth-proxy-f46cfd/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_925fa && [1/2] Building C object CMakeFiles/cmTC_925fa.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_925fa
    FAILED: cmTC_925fa 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_925fa.dir/testCCompiler.c.obj  -o cmTC_925fa   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:296 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 15.38 seconds =========================
```

deltazerorsan, 2023-03-22:
I have a similar issue, a bluetooth proxy compiles in Arduino platform but has very similar errors to above:

CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):

mweedon, 2023-04-15:
ESP32-WROOM-32UE 
DevKitC v4

`

esphome:
  name: bt-hall
  friendly_name: bt-hall

esp32:
  board: esp32dev
  framework:
      type: esp-idf
      #type: arduino

# Enable logging
logger:

# Enable Home Assistant API
# api:
#   encryption:
#     key: "0av++FMo7ZIOReNVdbKvRycZvGueLGvxNoXCBNXL9Js="

# ota:
#   password: !secret otapassword

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #power_save_mode: none
  #use_address: boiler-sound
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.15
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  
#   # Enable fallback hotspot (captive portal) in case wifi connection fails
#   ap:
#     ssid: bt-hall
#     password: bt-hall123

# binary_sensor:
#   - platform: status
#     name:  $friendlyname status

# sensor:
#   - platform: wifi_signal
#     name: $friendlyname WiFi
#     update_interval: 10s`

`INFO Reading configuration /config/esphome/esphome-web-08d9e8.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bt-hall (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/bt-hall/.pioenvs/bt-hall/CMakeFiles/CMakeOutput.log".
See also "/data/bt-hall/.pioenvs/bt-hall/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/bt-hall/.pioenvs/bt-hall/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_c929b && [1/2] Building C object CMakeFiles/cmTC_c929b.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_c929b
    FAILED: cmTC_c929b 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_c929b.dir/testCCompiler.c.obj  -o cmTC_c929b   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:296 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 11.17 seconds =========================`

Doesn't produce this error if type=arduino

Any help appreciated!!

slydiman, 2023-04-25:
The same https://github.com/esphome/issues/issues/3076