esphome/issues #4315: Component ili9xxx.display was marked as failed.

workingmanrob, 2023-03-16:
### The problem

Updated to latest stable in home assistant today - 2023.3.0 - and saw the new ili9xxx component added.

Attempt to switch from a working ili9486 config using an old PR - https://github.com/esphome/esphome/pull/1523 - to the new version failed.

This happens very early in the boot process.
```
[C][spi:023]: Setting up SPI bus...
[E][display:109]: Could not allocate buffer for display!
[E][display:109]: Could not allocate buffer for display!
[E][component:113]: Component ili9xxx.display was marked as failed.
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 1000.000000
[D][ledc.output:056]: Resolution calculated as 16
```

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.4

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

ili9xxx.display

### Example YAML snippet

```yaml
display:
  - platform: ili9xxx
    model: ILI9486
    cs_pin: 2
    dc_pin: 15
    reset_pin: 4
    rotation: 90
```

### Anything in the logs that might be useful for us?

```txt
[C][spi:023]: Setting up SPI bus...
[E][display:109]: Could not allocate buffer for display!
[E][display:109]: Could not allocate buffer for display!
[E][component:113]: Component ili9xxx.display was marked as failed.
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 1000.000000
[D][ledc.output:056]: Resolution calculated as 16
```

### Additional information

I have tried trimming the yaml down to bare minimum display config without the touchscreen or binary_sensors.

Error remained.

nielsnl68, 2023-03-16:
Hi, to use screens like the ILI9486 you need more memory then a normal esp32 has onboard.
To use this kind of displays you need to have an ESP32 module with PSRAM onboard.

workingmanrob, 2023-03-16:
hmmm - ok - well I got these esp32 boards after the 8266s I had were a miserable failure at driving the display.

I do have a working touchscreen display using the PR I linked above.  Will just keep on keeping on I guess.

You can close this if there's no hope of it ever working but the working config I have on my desk in front of me right now begs to know why it doesn't need more memory.  ¯\\_(ツ)_/¯

nielsnl68, 2023-03-16:
I did not say it does not work, i only say that you need a different esp32 to get displays like the ili9486 because of the memory needed to buffer the display.

nielsnl68, 2023-03-17:
Sorry to say, for a ili9486 you **always** need to have an esp32 with PSRAM.

Only the ili8341 and ili8342 can work's in 8bit mode without PSRAM.

I am working on a documentation change that will make this more cleary.

Zoogara, 2023-03-18:
So this is a limitation of the ESPHome implementation for these displays?  I assume the base ESPHome code uses a lot more RAM that simply coding up these devices under Arduino framework, as they work fine using no PSRAM ESP32s.

I have mine mounted on a custom board, I will either need to go back to using ILI9341 or find a WROVER board that is mostly pin compatible with a ESP32 DevkitC.

workingmanrob, 2023-03-22:
I have been able to keep my old working setup alive by using the PR from MasterTim and using esphone 2022.9.4

Mine is also mounted in a lovely little box with a battery pack attached to it and I'm not keen to rip that apart so I'll just stick with what works.

Good enough McMurray

closed 2023-03-22 by workingmanrob