esphome/issues #4316: pzemdc platform not compiling

Oleg-Sob, 2023-03-17:
### The problem

pzemdc platform not compiling

[image]

### Which version of ESPHome has the issue?

Current version: 2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2023.3.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

platform: pzemdc

ssieb, 2023-03-19:
Can you paste your config?  I agree with the change, but I'm wondering why I can't reproduce the issue.

Oleg-Sob, 2023-03-19:
esphome:
  name: pzem-017-1
#  friendly_name: pzem-017-1

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "KOyUg2KEdUPPK2GIh9IUsb5ub0kJbGvxcl8Mj3Xx37A="

ota:
  password: "fe571612ccf754a609717bee27bafad7"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  tx_pin: D1
  rx_pin: D2
  baud_rate: 9600
  stop_bits: 2

sensor:
  - platform: pzemdc
    id: pzemdc_1
    current:
      name: "PZEM-017 Current"
    voltage:
      name: "PZEM-017 Voltage"
    power:
      name: "PZEM-017 Power"
    energy:
      name: "PZEM-017 Energy"
    update_interval: 60s

button:
  - platform: template
    id: pzemdc_reset_energy
    on_press:
      then:
        - pzemdc.reset_energy: pzemdc_1

Oleg-Sob, 2023-03-19:
This config or something else

silens, 2023-03-19:
I have the same problem。
and this is my config:

substitutions:
  device_name: pzem_003

esphome:
  name: ${device_name}

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

logger:

# Enable Home Assistant API
api:
  encryption:
    key: "pxdHrmxFSCpZGi4/RptR/aHn9y1SbuQ3Oof8pYOtSQs="

ota:
  password: "644b77510133b5362983432442366d91"

wifi:
  ssid: "NETGEAR-IOT"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pzem-003 Fallback Hotspot"
    password: "Xnz0AZWPbXIP"

captive_portal:
    
web_server:
  port: 80

button:
  - platform: restart
    name: ${device_name}_reboot

time:
  - platform: sntp
    id: ${device_name}_time

uart:
  id: ${device_name}_uart
  rx_pin: 5
  tx_pin: 4
  baud_rate: 9600
  data_bits: 8
  stop_bits: 2

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${device_name}_ip
    mac_address:
      name: ${device_name}_mac

sensor:
  - platform: uptime
    name: ${device_name}_uptime
  - platform: wifi_signal
    name: ${device_name}_signal
    update_interval: 60s

  - platform: pzemdc
    current:
      name: "${device_name} Current"
    voltage:
      name: "${device_name} Voltage"
      filters:
        - multiply: 1.00
      unit_of_measurement: V
      accuracy_decimals: 2
    power:
      name: "${device_name} Power"

    update_interval: 5s

ssieb, 2023-03-19:
I figured out why it worked for me.  I had `debug:` in my uart config which must bring in the automations header.

closed 2023-03-19 by ssieb

Oleg-Sob, 2023-03-19:
And now it works?

ssieb, 2023-03-27:
This was included in the 2023.3.2 point release, so it's available now.

Oleg-Sob, 2023-03-27:
In ESPHome (dev) worked well.