esphome/issues #4317: LD2410 on esp32 - all sensors are unknown

cburghardt, 2023-03-17:
### The problem

I have connected the LD2410 to an esp32-pico-kit-v4
Unfortunately all of the sensors show up as unknown in HA.
I have tried various combinations of RX/TX pins on the board (18/23, 9/10, 16/17) but the behaviour is always identical.
Current version is GPIO9 for TX, GPIO10 for RX

### Which version of ESPHome has the issue?

2023.3.0 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32-IDF

### Board

 Espressif ESP32 ESP32-PICO-KIT V4.1SiP Development Board

### Component causing the issue

ld2410

### Example YAML snippet

```yaml
esphome:
  name: "esp32-presence-office"
  name_add_mac_suffix: False
  friendly_name: Presence Office

esp32:
  board: esp32dev
  framework:
    type: esp-idf

api:

logger:
  #baud_rate: 0  
  logs:
    sensor: INFO 
    binary_sensor: INFO

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  id: uart1
  tx_pin: GPIO9
  rx_pin: GPIO10
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  timeout: 150s
  max_move_distance : 6m
  max_still_distance: 0.75m
  g0_move_threshold: 10
  g0_still_threshold: 20
  g1_move_threshold: 10
  g1_still_threshold: 20
  g2_move_threshold: 20
  g2_still_threshold: 21
  g3_move_threshold: 30
  g3_still_threshold: 31
  g4_move_threshold: 40
  g4_still_threshold: 41
  g5_move_threshold: 50
  g5_still_threshold: 51
  g6_move_threshold: 60
  g6_still_threshold: 61
  g7_move_threshold: 70
  g7_still_threshold: 71
  g8_move_threshold: 80
  g8_still_threshold: 81  

sensor:
  - platform: ld2410
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
```

### Anything in the logs that might be useful for us?

```txt
[15:56:08][C][logger:293]: Logger:
[15:56:08][C][logger:294]:   Level: DEBUG
[15:56:08][C][logger:295]:   Log Baud Rate: 115200
[15:56:08][C][logger:296]:   Hardware UART: UART0
[15:56:08][C][logger:298]:   Level for 'sensor': INFO
[15:56:08][C][logger:298]:   Level for 'binary_sensor': INFO
[15:56:08][C][uart.idf:116]: UART Bus:
[15:56:08][C][uart.idf:117]:   Number: 1
[15:56:08][C][uart.idf:118]:   TX Pin: GPIO9
[15:56:08][C][uart.idf:119]:   RX Pin: GPIO10
[15:56:08][C][uart.idf:121]:   RX Buffer Size: 256
[15:56:08][C][uart.idf:123]:   Baud Rate: 256000 baud
[15:56:08][C][uart.idf:124]:   Data Bits: 8
[15:56:08][C][uart.idf:125]:   Parity: NONE
[15:56:08][C][uart.idf:126]:   Stop bits: 1
[15:56:08][C][ld2410:012]: LD2410:
[15:56:08][C][ld2410:014]:   HasTargetSensor 'Presence'
[15:56:08][C][ld2410:014]:     Device Class: 'occupancy'
[15:56:08][C][ld2410:015]:   MovingSensor 'Moving Target'
[15:56:08][C][ld2410:015]:     Device Class: 'motion'
[15:56:08][C][ld2410:016]:   StillSensor 'Still Target'
[15:56:08][C][ld2410:016]:     Device Class: 'occupancy'
[15:56:08][C][ld2410:019]:   Moving Distance 'Moving Distance'
[15:56:08][C][ld2410:019]:     Device Class: 'distance'
[15:56:08][C][ld2410:019]:     State Class: ''
[15:56:08][C][ld2410:019]:     Unit of Measurement: 'cm'
[15:56:08][C][ld2410:019]:     Accuracy Decimals: 0
[15:56:08][C][ld2410:020]:   Still Distance 'Still Distance'
[15:56:08][C][ld2410:020]:     Device Class: 'distance'
[15:56:08][C][ld2410:020]:     State Class: ''
[15:56:08][C][ld2410:020]:     Unit of Measurement: 'cm'
[15:56:08][C][ld2410:020]:     Accuracy Decimals: 0
[15:56:08][C][ld2410:021]:   Moving Energy 'Move Energy'
[15:56:08][C][ld2410:021]:     Device Class: 'energy'
[15:56:08][C][ld2410:021]:     State Class: ''
[15:56:08][C][ld2410:021]:     Unit of Measurement: '%'
[15:56:08][C][ld2410:021]:     Accuracy Decimals: 0
[15:56:08][C][ld2410:022]:   Still Energy 'Still Energy'
[15:56:08][C][ld2410:022]:     Device Class: 'energy'
[15:56:08][C][ld2410:022]:     State Class: ''
[15:56:08][C][ld2410:022]:     Unit of Measurement: '%'
[15:56:08][C][ld2410:022]:     Accuracy Decimals: 0
[15:56:08][C][ld2410:023]:   Detection Distance 'Detection Distance'
[15:56:08][C][ld2410:023]:     Device Class: 'distance'
[15:56:08][C][ld2410:023]:     State Class: ''
[15:56:08][C][ld2410:023]:     Unit of Measurement: 'cm'
[15:56:08][C][ld2410:023]:     Accuracy Decimals: 0
[15:56:08][C][ld2410:029]:   Firmware Version : 0.0.0000
```

WhereDoesTheCodeGo, 2023-03-17:
This just randomly popped up for me yesterday on an existing install of an emporia vue2 that had been humming along happily for over a month. No changes happened yesterday but I might have not noticed when the "failure" occurred because it initially presented with only a single circuit not reporting data until I rebooted both my HA and the vue2 at which point everything stopped reporting. I never touch the vue2 so I am suspecting something in HA changed?

Here is a diag dump from the HA integration:

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.03.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "esphome",
    "name": "ESPHome",
    "after_dependencies": [
      "zeroconf",
      "tag"
    ],
    "codeowners": [
      "@OttoWinter",
      "@jesserockz"
    ],
    "config_flow": true,
    "dependencies": [
      "bluetooth"
    ],
    "dhcp": [
      {
        "registered_devices": true
      }
    ],
    "documentation": "https://www.home-assistant.io/integrations/esphome",
    "integration_type": "device",
    "iot_class": "local_push",
    "loggers": [
      "aioesphomeapi",
      "noiseprotocol"
    ],
    "requirements": [
      "aioesphomeapi==13.5.1",
      "esphome-dashboard-api==1.2.3"
    ],
    "zeroconf": [
      "_esphomelib._tcp.local."
    ],
    "is_built_in": true
  },
  "data": {
    "config": {
      "entry_id": "d913cdca97794a0d81b56b2054fe27ea",
[... 619 lines cut ...]
          "accuracy_decimals": 0,
          "force_update": false,
          "state_class": 2,
          "last_reset_type": 0
        },
        {
          "object_id": "circuit_16_daily_energy",
          "key": 2566306509,
          "name": "Circuit 16 Daily Energy",
          "unique_id": "emporiavue2sensorcircuit_16_daily_energy",
          "disabled_by_default": false,
          "icon": "",
          "entity_category": 0,
          "device_class": "energy",
          "unit_of_measurement": "Wh",
          "accuracy_decimals": 0,
          "force_update": false,
          "state_class": 2,
          "last_reset_type": 0
        }
      ],
      "text_sensor": []
    }
  }
}

cburghardt, 2023-03-18:
OK, I got it working by swapping RX and TX pins. Apparently they need to be swapped between the esp32 and the ld2410. Now it works.

closed 2023-03-18 by cburghardt