esphome/issues #4318: RPi Pico W broken after ESPHome update to stable 2023.3.0 or late 2023.3.0beta

nepozs, 2023-03-17:
### The problem

Firmware compiled using ESPHome 2023.3.0 and some latest beta - for sure 2023.3.0beta5 (but maybe also 2023.3.0beta4 - not sure - beta versions changed too fast to be 100% sure and I was surprised, as earlier betas was working good) not working properly even if using very simple (or initial) config.

Latest stable working version 2023.2.4

Working beta versions 2023.3.0beta1 and probably (I'm not sure - don't remember) 2023.3.0beta2
with early betas internal temperature tested and it was working, so later changes must be a source of problem

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

not dependent (but 2023.3.4, 2023.3.5)

### What platform are you using?

RP2040

### Board

rpipicow (original board, not a clone)

### Component causing the issue

unknown, but even simplest (initial) config does not work

### Example YAML snippet

```yaml
esphome:
  name: rpi-pico-w
  friendly_name: rpi-pico-w

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Rpi-Pico-W Fallback Hotspot"
    password: "12345678"
```

### Anything in the logs that might be useful for us?

```txt
USB debug not possible

INFO Reading configuration /config/esphome/rpi-pico-w.yaml...
INFO Detected timezone 'Europe/Warsaw'
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
ERROR Serial port closed!
```

Supervisor log and hots logs look like RPi Pico W were in bootloop
```
23-03-17 17:46:12 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
23-03-17 17:46:12 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/113 - None
23-03-17 17:46:14 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/114 - None
23-03-17 17:46:14 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
23-03-17 17:46:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
23-03-17 17:46:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/114 - None
23-03-17 17:46:22 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/115 - None
23-03-17 17:46:22 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
23-03-17 17:46:28 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
23-03-17 17:46:28 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/115 - None
23-03-17 17:46:31 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/116 - None
23-03-17 17:46:31 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
23-03-17 17:46:37 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
23-03-17 17:46:37 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/116 - None
23-03-17 17:46:39 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/001/117 - None
23-03-17 17:46:39 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/ttyACM0 - /dev/serial/by-id/usb-Raspberry_Pi_PicoArduino_E661410403777E2D-if00
```
```
Mar 17 16:45:47 nuctyrmanda kernel: cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Mar 17 16:45:55 nuctyrmanda kernel: usb 1-1: USB disconnect, device number 111
Mar 17 16:45:55 nuctyrmanda kernel: usb 1-1: new full-speed USB device number 112 using xhci_hcd
Mar 17 16:45:55 nuctyrmanda kernel: usb 1-1: New USB device found, idVendor=2e8a, idProduct=f00a, bcdDevice= 1.00
Mar 17 16:45:55 nuctyrmanda kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 17 16:45:55 nuctyrmanda kernel: usb 1-1: Product: PicoArduino
Mar 17 16:45:55 nuctyrmanda kernel: usb 1-1: Manufacturer: Raspberry Pi
Mar 17 16:45:55 nuctyrmanda kernel: usb 1-1: SerialNumber: E661410403777E2D
Mar 17 16:45:55 nuctyrmanda kernel: cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Mar 17 16:46:03 nuctyrmanda kernel: usb 1-1: USB disconnect, device number 112
Mar 17 16:46:03 nuctyrmanda kernel: usb 1-1: new full-speed USB device number 113 using xhci_hcd
Mar 17 16:46:04 nuctyrmanda kernel: usb 1-1: New USB device found, idVendor=2e8a, idProduct=f00a, bcdDevice= 1.00
Mar 17 16:46:04 nuctyrmanda kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 17 16:46:04 nuctyrmanda kernel: usb 1-1: Product: PicoArduino
Mar 17 16:46:04 nuctyrmanda kernel: usb 1-1: Manufacturer: Raspberry Pi
Mar 17 16:46:04 nuctyrmanda kernel: usb 1-1: SerialNumber: E661410403777E2D
Mar 17 16:46:04 nuctyrmanda kernel: cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Mar 17 16:46:12 nuctyrmanda kernel: usb 1-1: USB disconnect, device number 113
Mar 17 16:46:12 nuctyrmanda kernel: usb 1-1: new full-speed USB device number 114 using xhci_hcd
Mar 17 16:46:12 nuctyrmanda kernel: usb 1-1: New USB device found, idVendor=2e8a, idProduct=f00a, bcdDevice= 1.00
Mar 17 16:46:12 nuctyrmanda kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 17 16:46:12 nuctyrmanda kernel: usb 1-1: Product: PicoArduino
Mar 17 16:46:12 nuctyrmanda kernel: usb 1-1: Manufacturer: Raspberry Pi
Mar 17 16:46:12 nuctyrmanda kernel: usb 1-1: SerialNumber: E661410403777E2D
Mar 17 16:46:12 nuctyrmanda kernel: cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Mar 17 16:46:20 nuctyrmanda kernel: usb 1-1: USB disconnect, device number 114
```
```

### Additional information

Revert to 2023.2.4 of course helps.
If I will have enough free time i will try to find latest working beta (next week probably).

TeaTimeSoon, 2023-03-18:
Same issue for me! :-(

mbirrell66, 2023-03-18:
Same issue as well

rasmus-wi, 2023-03-19:
I enabled logging on UART0, plus VERY_VERBOSE logging, and got this. Definitely looks like some kind of boot loop. 
```
[I][logger:258]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][rp2040.preferences:117]: Saving preferences to flash...
[I][logger:258]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][rp2040.preferences:117]: Saving preferences to flash...
``

nepozs, 2023-03-19:
Related issue (I've just missed it because it has title including "not rebooting", but this is in realty bootloop)
https://github.com/esphome/issues/issues/4308

jesserockz, 2023-03-21:
Duplicate of #4308

closed 2023-03-21 by jesserockz