esphome/issues #4319: xpt2046 / ili9431 - No more touchscreen input on two devices

PlantsThatsWhatsUp, 2023-03-17:
### The problem

Touchscreen no longer works. It does not register any input. There is nothing in logs, even on very_verbose. It's as if it's not being touched at all. I suspect it has to do with the updates to ili9xxx component. I will install the docker esphome when I have time and attempt to see if it running an older version fixes it.

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32

### Board

Firebeetle ESP32-E

### Component causing the issue

xpt2046 

### Example YAML snippet

```yaml
binary_sensor:
  - platform: touchscreen
    name: temphumibutton_main
    internal: true
    x_min: 17
    x_max: 100
    y_min: 29
    y_max: 84
    page_id: mainpage
    filters:
      - delayed_on: 20ms
      - delayed_off: 20ms
    on_press:
      then:
      - display.page.show: temperaturepage
      - component.update: theDisplay

...
...

touchscreen:
  platform: xpt2046
  id: theTouchscreen
  cs_pin: 4 # T_CS
  interrupt_pin: 35 #T_IRQ
  update_interval: 1s
  report_interval: 1s
  threshold: 300
  calibration_x_min: 3860
  calibration_x_max: 280
  calibration_y_min: 0
  calibration_y_max: 3860
  swap_x_y: false
  on_touch:
    ...

display:
  - platform: ili9xxx
    id: theDisplay
    model: ili9341
    cs_pin: 14
    dc_pin: 25
    reset_pin: 1
    rotation: 270
    # so its 320 X 240 
    update_interval: 2s
    pages:
         ...
```

### Anything in the logs that might be useful for us?

```txt
Nothing useful in logs, this is the only mention:

[14:07:35][C][xpt2046:141]: XPT2046:
[14:07:35][C][xpt2046:143]:   IRQ Pin: GPIO35
[14:07:35][C][xpt2046:144]:   X min: 280
[14:07:35][C][xpt2046:145]:   X max: 3860
[14:07:35][C][xpt2046:146]:   Y min: 0
[14:07:35][C][xpt2046:147]:   Y max: 3860
[14:07:35][C][xpt2046:149]:   Swap X/Y: NO
[14:07:35][C][xpt2046:150]:   Invert X: YES
[14:07:35][C][xpt2046:151]:   Invert Y: NO
[14:07:35][C][xpt2046:153]:   threshold: 300
[14:07:35][C][xpt2046:154]:   Report interval: 1000
[14:07:35][C][xpt2046:156]:   Update Interval: 1.0s
```

### Additional information

I have two substantially identical devices. They didn't both fail immediately upon update, but in the 24 hours since, have failed.  Full YAML is 2000+ lines, but can upload it if deemed required.

nielsnl68, 2023-03-18:
I'm seeing the same thing, i think. 
Before i dig into this issue, just wondering, i use a esp32s2 as controller, which on do you use @PlantsThatsWhatsUp?

PlantsThatsWhatsUp, 2023-03-18:
> I'm seeing the same thing, i think. 
> Before i dig into this issue, just wondering, i use a esp32s2 as controller, which on do you use @PlantsThatsWhatsUp?

[ESP32-E](https://www.dfrobot.com/product-2195.html)

nielsnl68, 2023-03-20:
Could you jump into Discord for a moment and help me out with some details for me, so i can narrow down the issue we having.
Because i do not know exactly when this issue started. I did not updated my display over the last month's or so.

Zoogara, 2023-03-29:
Can confirm having same issue - device worked correctly prior to 2023.3.0

NOTE: this may be relevant - the display and the touchscreen are sharing the same SPI bus.  I cannot change this, I am using the Dustin Watts TFT board.  There was a comment on one forum that the issue was resolved by using an addition SPI bus, note this is not an option for me nor should it be required.

ESPHome: 2023.3.0
Display: ILI9341
Touchscreen: XPT2046
ESP: ESP32 Devkit-C

Zoogara, 2023-03-31:
More to this - received some WROVER boards so replaced ILI9341 with ILI9488.  Touch screen now works fine...  Exactly same wiring and shared bus.  Maybe there is something going on with only some touch screen chips???  Weird.

Johncunningham918, 2023-03-31:
I could be wrong, but I believe the issue has to do with the chip selector pins (CS and T_CS) when sharing the same SPI buss specifically on the ili9341 ICs. I have 5 ili9341 touch displays, and none of them will respond to touch in ESPHome. See thread: https://community.home-assistant.io/t/ili9341-xpt2046-displays-will-not-function-correctly/553580. 

However, I was eventually able to confirm they work using Arduino IDE on an Arduino uno over a shared SPI buss, so long as both chip selector pins on the arduino were set as outputs, and pulled high before running the script. (If using the ili9341/xpt2046 on arduino, you need to use either level shift ICs or voltage dividers on all the SPI pins to bring them down to 3.3v as they are not 5v tolerant like the VCC and LED pins are)
```
// avoid chip select contention
  pinMode(TS_CS, OUTPUT);
  digitalWrite(TS_CS, HIGH);
  pinMode(TFT_CS, OUTPUT);
  digitalWrite(TFT_CS, HIGH);
```
 I don't know if the recent ili9xxx and touchscreen component updates changed this behavior or not as I didn't use the components prior to the esphome updates, but I figured it was worth sharing. I am going to give it another shot using esphome with what I learned in arduino, hopefully I will be successful this time by pulling the pins high in the config. I will post back either way to let the world know.

Johncunningham918, 2023-03-31:
I am hitting a bit of a roadblock; perhaps it is due to my lack of knowledge in esphome config. I did start probing around trying to compare values from my arduino/ili9341/xpt2046 instance against my non-working esp32/ili9341/xpt2046, but I don't know enough to say whether the behaviors of each device should match or not. But maybe some of you may find useful information below. 

This is what I saw when I looked at the CS (blue line) pin, and the T_CS (yellow line) pin while trying to run the touch calibration on the ESP32 through esphome. Touch inputs didn't change either output, and this kept repeating from when it was powered on until it was powered back off. Trying to pull the pins high and low within the esphome config also did not change this behavior.
[image]

With the probes hooked up the same way on the arduino (yellow line is T_CS pin, blue is CS pin), this is what was showing after startup. At this point, the arduino is waiting for you to press 2 points on the screen to calibrate it before starting a pong style game. 
[image]

The beginning lower frequency is when I was holding the stylus to the screen. You can see the frequency increase immediately after ending the touch input. Again, the yellow line represents the T_CS pin on the display. 
[image]

After pressing both calibration points, you then get a "start game" button to begin playing. This is what showed while it was waiting for the start game button press:
[image]

Finally, this is what it looked like while the game was in play. 
[image]

So clearly, there is a huge difference in the operations between the two setups. Like I said, this may be normal, or it might not be. Hopefully someone who knows better can comment on it. If anybody would like me to probe something else between the two, I would be happy to. Just say what you want and I will see what I can do.

wmnbg, 2023-04-02:
I can confirm having same issue - device worked correctly prior to 2023.3.0

I have two AZtouch Display and both recognize no touch anymore.
I Use a ESP32 D1 Mini and Touch IRQ is connectet to GPIO16.
The SPI-Bus is used from the ili9431-Display and the xpt2046 Touch controller.
When I pull down GPIO16 to Ground it switches to the next Screen, how I have configured.
But I got no Signal on IRQ by touching the screen.

After downgrading to ESPHome 2023.2.4 the touchpanel works again like before.

Johncunningham918, 2023-04-02:
> I can confirm having same issue - device worked correctly prior to 2023.3.0
>
> I have two AZtouch Display and both recognize no touch anymore. I Use a ESP32 D1 Mini and Touch IRQ is connectet to [...]

How did you downgrade to 2023.2.4?

wmnbg, 2023-04-03:
Hi,
in Homeassistant go to _Preferences/System/Backups_ and look for "addon_5c53de3b_esphome_2023.2.4"
Mark both Checkboxes and Click "Restore"
[image]

PlantsThatsWhatsUp, 2023-04-07:
Might be helpful to explore behaviour when running this PR that Niels prepared.
 https://github.com/esphome/esphome/pull/4596

```
external_components:
  - source: github://pr#4596
    components: [display, touchscreen, xpt2046]
    refresh: 10s
```

darlenya, 2023-08-27:
Hi, is there any update on this issue? I am Using version 2023.8.1 and had the same issue

Johncunningham918, 2023-08-27:
> Hi, is there any update on this issue? I am Using version 2023.8.1 and had the same issue

I moved my project to Arduino and haven't tried using the esp8266/esphome since. Maybe try downgrading to 2023.2.4 like the person above did, hopefully that works with your project.

pepe59, 2023-08-30:
Here, the display ILI9341 works for me, including touch on the esp32.
ESPHome version 2023.8.2.
I stopped sharing the SDO(MISO)/T-DO pins and used only the T-DO touch on the defined MISO pin in the yaml esp.
I do not use pin SDO(MISO) displays.

smarthomecircle, 2023-09-02:
@pepe59 If possible can you share the ESP Home configuration ?

pepe59, 2023-09-02:
Yes, there you have it. It still works fine.
I'm testing media player control.
```
esphome:
  name: ha-zaloha-display
  friendly_name: Display HA záloha

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  logs:
    component: ERROR

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxxxxxxxxxxxxx"
    password: "xxxxxxxxxxxxxx"

captive_portal:

spi:
 clk_pin: 18 #display SCK/touch T_CLK
 mosi_pin: 23 #display SDI(MOSI)/touch T_DIN
 miso_pin: 19 #touch T_DO

touchscreen:
  platform: xpt2046
  id: my_touchscreen
  cs_pin: 17 #touch T_CS
  interrupt_pin: 16 #touch T_IRQ
  update_interval: 50ms
  report_interval: 1s
  threshold: 400
  calibration_x_min: 283
  calibration_x_max: 3776
  calibration_y_min: 3870
  calibration_y_max: 392
  swap_x_y: False

display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: 13 #display CS
    dc_pin: 27 #display DC
    reset_pin: 33 #display RESET
    rotation: 90
    id: tft_ha_test

# Define a PWM output on the ESP32
output:
  - platform: ledc
    pin: 32 #display LED
    id: gpio_32_backlight_pwm

# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: gpio_32_backlight_pwm
    name: "Display HA test podsvícení"
    id: back_light
    restore_mode: ALWAYS_ON
```

smarthomecircle, 2023-09-02:
> Yes, there you have it. It still works fine. I'm testing media player control.
>
> ```
> esphome:
>   name: ha-zaloha-display
>   friendly_name: Display HA záloha
>
> esp32:
>   board: esp32dev
>   framework: [...]

Thanks a lot for the reply @pepe59 

I also just figured it out. My config was the same. 
As you as said before the MISO from the display should not be connected but the DO from the display must be connected to MISO pin in the ESP32. 

I had not connected the interrupt pin because I was little misguided by this comment in the thread [here](https://community.home-assistant.io/t/esphome-ili9341-touchscreen/316348/18?u=amrut) which stated this. 
>and left the T_IRQ unused

darlenya, 2023-09-04:
Hi thanks for you fast reply. I will test this on the next weekend

smarthomecircle, 2023-09-12:
Hi All, 

So I tried the config that @pepe59 provided above. It works but only when I disconnect and reconnect the interrupt pin. 

So the scenarios is like this.
I start the ESP32 with the display and touch pins connected exactly as shown in the comments in the above config. 
Now when I touch the screen it does not work. 
Now, I just disconnect and reconnect the `T_IRQ` on the display and the touch starts working. 

What could be the cause?

I am using ESPHome version 2023.8.2.

Do I need to set something like this for the interrupt pin
>  pinMode(TS_CS, OUTPUT);
  digitalWrite(TS_CS, HIGH);
  pinMode(TFT_CS, OUTPUT);
  digitalWrite(TFT_CS, HIGH);

Which was mentioned for CS pin a comments before. 
or its some kind of race condition in the code which starts working after disconnection and reconnection ( I am not sure.. Just speculating😄)

smarthomecircle, 2023-09-12:
I can now confirm that the touch panel works flawlessly after downgrading esphome to `2023.2.4` with this configuration. 

```
esphome:
  name: "lcd-with-touch-esp32"
  friendly_name: LCD-With-Touch-ESP32

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxx="

ota:
  password: "xxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip: 
    static_ip: 192.168.0.85
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lcd-Esp32 Fallback Hotspot"
    password: "vxLa8dfwFIL1"

captive_portal:

web_server:
  port: 80

spi:
 clk_pin: 18
 mosi_pin: 23
 miso_pin: 19

touchscreen:
  platform: xpt2046
  id: my_touchscreen
  cs_pin: 17
  interrupt_pin: 16
  update_interval: 50ms
  report_interval: 1s
  threshold: 400

  calibration_x_max: 3691
  calibration_x_min: 319
  calibration_y_max: 453
  calibration_y_min: 3873
  on_touch:
    - lambda: |-
          ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%d",
              id(my_touchscreen).x,
              id(my_touchscreen).y,
              id(my_touchscreen).x_raw,
              id(my_touchscreen).y_raw
              );

display:
  - platform: ili9341   # this has changed since version 2023.3.0
    model: TFT_2.4    # this has changed since version 2023.3.0
    cs_pin: 13
    dc_pin: 27
    reset_pin: 33
    rotation: 270
    lambda: |-
      auto red = Color(255, 0, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      it.fill(green);
      if (id(my_touchscreen).touched)
        it.filled_circle(id(my_touchscreen).x, id(my_touchscreen).y, 30, red);

# Define a PWM output on the ESP32
output:
  - platform: ledc
    pin: 32 #display LED
    id: gpio_4_backlight_pwm

# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: gpio_4_backlight_pwm
    name: "Display HA test podsvícení"
    id: back_light
    restore_mode: ALWAYS_ON

```
The config is the same as @pepe59 but the had to change the platform because of the downgrade. 

I had to change the display platform to the below config after downgrading. So there is definitely a bug a in the newer version.

```
display:
  - platform: ili9341   # this has changed since version 2023.3.0
    model: TFT_2.4    # this has changed since version 2023.3.0
```

Once I start using the platform ili9xxx, the touch stops working again.

jeroenterheerdt, 2024-05-31:
I recognize this is an old thread, where you ever able to make this config work on newer versions @smarthomecircle ? I too have the issue where I need to toggle the IRQ

BafS, 2024-11-19:
Thanks for those info, it took me a lot of time to figure out how to wire it with the buggy version (one shared SPI but MISO plugged to the touchscreen only). I also have the same issue with the IRQ to toggle the touchscreen.

closed 2024-12-03 by clydebarrow

CodeMartn, 2024-12-13:
The issue was closed without a commit.
Was this issue really fixed? I still have that issue with 2024.11.3