esphome/issues #4322: After updating from command line Home Assistant ESPHome addon still list as update available

erdembey, 2023-03-18:
### The problem

I am using esphome command line to update my devices. With this way i've solved the DHT11 bug which is mentioned below;
https://github.com/esphome/issues/issues/4189#issuecomment-1469631775

As a result of using command line update method, Home Assistant ESPHome gui list my devices as "update available" even though i've already updated via my pc.

### Which version of ESPHome has the issue?

v2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32

ssieb, 2023-03-18:
The dashboard doesn't know what version is actually on the device.  It only knows what it last compiled for the device.

closed 2023-03-18 by ssieb

erdembey, 2023-03-18:
This can be feature request. We can see the version via logs as far as i can see. Also, we can see the version from HA device info page.

ssieb, 2023-03-18:
It's in progress: https://github.com/esphome/esphome/pull/4429
But this issue is a duplicate of https://github.com/esphome/issues/issues/4094 anyway.