esphome/issues #4323: gpio high at boot for half a second

aura6174, 2023-03-18:
### The problem

recently flashed 2 tuya wipro smartlight to esphome firmware, everything works well but everytime i power up both smartlight it's a flick or strobe effect for enough time detect by human eye which is weird. 
Already tried early_pin_init and restore mode nothing change happen. 
Using template yaml from esphome cookbook.

found same issue someone facing: https://www.reddit.com/r/homeassistant/comments/pss2q5/esphome_smart_plugs_and_lights_turn_on/

### Which version of ESPHome has the issue?

2023.2.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

wipro-ns7001, wipro-ns9001

aura6174, 2023-03-20:
hey, esphome dev take a look at this issue, this issue causing light strike while powered on the device. which was weird for some reasons, i can't revert back to stock firmware.

Samu84, 2023-03-22:
i think you need to remove on_boot:, then put restore state

aura6174, 2023-03-27:
> i think you need to remove on_boot:, then put restore state
> 
> 

No, on_boot not creating this issue, i inspect there is early_pin_init make pin high at boot, i tried "early_pin_init: false" makes no difference.

On other project, which uses "High logic active relay" i observe relay on-off-on same behaviour. No issue on "Low logic active relay"

dd32, 2023-04-14:
Certain pins on the esp8266 may have a floating logic high before the processor fully boots, according to [this](https://rabbithole.wwwdotorg.org/2017/03/28/esp8266-gpio.html) GPIO 12, 13, and 14 may be affected by that (your green, warm white, and blue channels).

pmannk, 2023-05-02:
Some pins pull high during boot and at reset. Unfortunately thats the nature of the hardware.
Early init might help, but it's not guaranteed.
I've covered a bit more on this in the following comments elsewhere if you're curious:
https://github.com/esphome/issues/issues/3280#issuecomment-1141579560
https://github.com/esphome/issues/issues/3280#issuecomment-1144361641

However you also mention this is a Wipro NS9001. Are you sure your GPIO assignment is correct?
Looking at the blackadder templates they suggest GPIO04 is brightness and GPIO12 is colour temperature (ie, it's a color temperature light: https://esphome.io/components/light/color_temperature.html)

dgtal1, 2025-05-04:
Same here. I already soldered the entire project and my relays are turned on at boot for a short while. This is unacceptable. early_pin_init: false didn't solve it.
Has anyone found some workaround??

ssieb, 2025-05-04:
Which pin?

dgtal1, 2025-05-05:
> Which pin?

12,13,14,16.

I'm considering adding another transistor as the master for all 4 and make a RC delay for it attached from 3.3V to its base, but that will take another 0.7 V from me and then it may not be sufficient for the relays to turn on... that issue is really bad and nasty

ssieb, 2025-05-05:
You'll want to read this: https://rabbithole.wwwdotorg.org/2017/03/28/esp8266-gpio.html