esphome/issues #4324: Compile error using Fastled, Arduino framework 2.7.4 and ESPHome 2023.3.0

jsuanet, 2023-03-18:
### The problem

After upgrading ESPHome to version 2023.3 you get compile errors when using Fastled component and Arduino framework 2.7.4.

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n.a.

### What platform are you using?

ESP8266

### Board

Wemos D1-mini

### Component causing the issue

Fastled

### Example YAML snippet

```yaml
esphome:
  name: test-8266

esp8266:
  board: d1_mini
  framework:
    version: 2.7.4

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
   
api:
  password: !secret api_password

ota:
  password: !secret ota_password

logger:
  level: DEBUG

light:
  - platform: fastled_clockless
    chipset: WS2813
    pin: D5
    num_leds: 95
    rgb_order: GRB
    default_transition_length: 50ms
    id: my_led
    name: "Led-strip"
```

### Anything in the logs that might be useful for us?

```txt
root@ac55e03c37ab:/config# esphome version
Version: 2023.3.0
root@ac55e03c37ab:/config# esphome compile test-8266.yaml 
INFO Reading configuration test-8266.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing test-8266 (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing fastled/FastLED @ 3.3.2
INFO Installing fastled/FastLED @ 3.3.2
Unpacking  [####################################]  100%
Library Manager: FastLED@3.3.2 has been installed!
INFO FastLED@3.3.2 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- FastLED @ 3.3.2
Compiling .pioenvs/test-8266/lib3bf/FastLED/FastLED.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/bitswap.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/colorpalettes.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/colorutils.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/hsv2rgb.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/lib8tion.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/noise.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/platforms.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/power_mgt.cpp.o
Compiling .pioenvs/test-8266/lib3bf/FastLED/wiring.cpp.o
Compiling .pioenvs/test-8266/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/test-8266/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/test-8266/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/test-8266/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/api_frame_helper.cpp:5:0:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/test-8266/src/esphome/components/api/api_connection.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/test-8266/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/test-8266/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/test-8266/src/esphome/components/api/api_pb2.cpp.o] Error 1
============================================================================== [FAILED] Took 9.06 seconds ==============================================================================
root@ac55e03c37ab:/config#
```

shfan91, 2023-03-18:
Hey there, ran into the same issue you described above. I was able to work around this by switching from FastLED light to NeoPixelBus Light.

Here's a sample of my code
`esphome:
  name: esp-office

esp8266:
  board: nodemcuv2
  framework:
    version: recommended #2.7.4

wifi:
  ssid: !secret ssid
  password: !secret wifi
  fast_connect: true
  power_save_mode: none

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret key

ota:

i2c:
  sda: D2
  scl: D1
  scan: False

sensor:
  - platform: dht
    pin: D6
    model: dht22
    temperature:
      name: "esp_office_temp"
      id: temp
    humidity:
      name: "esp_office_humidity"
    update_interval: 30s
  - platform: bh1750
    name: "esp_office_lux"
    address: 0x23
    update_interval: 30s 
    

binary_sensor:
  - platform: gpio
    pin: D0
    name: "esp_office_motion"
    device_class: motion
    id: motion
    filters:
      - delayed_off: 60s
    on_state:
      then:
      - if:
          condition:
              binary_sensor.is_on: motion
          then:
            - light.turn_on:
                id: led1
                red: 100%
                green: 100%
                blue: 0%
                brightness: 35%
            - delay: 5s      
          else:
            - light.turn_on:
                id: led1
                red: 0%
                green: 100%
                blue: 0%
                brightness: 35%

    
light:
  - platform: neopixelbus
    variant: WS2812
    pin: D3
    num_leds: 1
    type: RGB
    name: "esp_office_led"
    id: led1`

jsuanet, 2023-03-18:
Yes, I know that switching to Neopixelbus solves the problem, however for me that's a work around.
If the fastled component is not longer working anymore, it should be declared deprecated and a "breaking change" item in the release notes. otherwise it is just a bug in the new 2023.3.0 version 😄

andyfsimon, 2023-03-20:
Same here - was planning to move to Neopixelbus but this should be reported nevertheless.

jesserockz, 2023-03-22:
lol, didnt see it until now, buit I am pretty sure this is the fix

```diff
index a4f64171f..05a7eaa4c 100644
--- a/esphome/core/helpers.h
+++ b/esphome/core/helpers.h
@@ -550,7 +550,7 @@ class Mutex {
  */
 class LockGuard {
  public:
-  LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
+  LockGuard(Mutex &mutex) : mutex_(mutex) { mutex_.lock(); }
   ~LockGuard() { mutex_.unlock(); }
 
  private:
```

closed 2023-03-22 by jesserockz

nor500, 2023-03-23:
I installed 2023.3.1 update but it didn 't make any change regarding this issue. The compile error  still remained.

jsuanet, 2023-03-23:
The PR is already merged in the dev branch and has label 2023.3.2, so you have to wait till that version is released. I tested with the dev version and with that version the problem is solved.