esphome/issues #4326: Esphome webpage keeps closing during edit, as well as devices all go offline and then online again in sync

fresnoboy, 2023-03-19:
### The problem

When trying to edit any esphome device config, the window will close after about a minute or so.  Also, all devices periodically go online and offline, even though network traffic is stable.   This started after an update to esphome 2023.3.0.

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP8266

### Board

LC tech relay board

### Anything in the logs that might be useful for us?

```txt
This message is occurring over and over 

{'Corrupt packet received while decoding name from {self.source}': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                                                    IncomingDecodeError('Corrupt packet received while decoding name from {self.source}'),
                                                                    <traceback object at 0x7fc67a59f480>),
 'index out of range': (<class 'IndexError'>,
                        IndexError('index out of range'),
                        <traceback object at 0x7fc67a4f4140>),
 'unpack requires a buffer of 10 bytes': (<class 'struct.error'>,
                                          error('unpack requires a buffer of 10 bytes'),
                                          <traceback object at 0x7fc67a4655c0>)}
```

fresnoboy, 2023-03-19:
Also, after the edit window closes, I can't seem to refresh the esphome page.  After a few secs I get the  Supervisor
Unable to fetch add-on info to start Ingress error.

Antexa, 2023-04-02:
Did you try with private navigation ? Did the trick for me... 

Just a clear cache on my web browser resolve the conflict.