esphome/issues #4327: ESPHOME and NGINX Proxy Manager SSL proxy HA addon don't play nicely

mcarbonneaux, 2023-03-19:
### The problem

In fact esphome don't accept the normal way of nginx is transmiting the browser request host and port to esphome throught ha ingress. 

by default when firewall is on front and expose the nginx with diferent port than the nginx port (is the case of nginx proxy manager ha addons is 443) this port are not transmited by default by nginx (nginx use the 443 port in place), and the esphome refuse the connection.

For example:
- firewall on port 12345 on public ip -> homeassistant nginx proxy manager port 443 on ha private ip (ex: 192.168.xx.xx the ha os ip) -> homeassistant core ui on 8123 (on local container private ip)

with the way of nginx do the 12345 port are not transmited to the backend (the backend receive 443 in place)...

I've found a "clean way" to fix this use of ha ingress with esphome using NGINX proxy manager addon (base on exchange on https://github.com/esphome/issues/issues/1035#issuecomment-697560334) by using `custom location` of `nginx proxy manager`.

i've added the location `/api/hassio_ingress/` on the `Proxy Host` to `homeassistant/api/hassio_ingress/` on `8123` port, with `http` protocol:

[image]

and i've added the custom configuration :
```
proxy_set_header Host $http_host;
```

to force to use the `hostname` **AND** `port` from the browser as transmited by using (`$http_host` nginx variable) the `Host` header received in place of the nginx variable `$host` that not containe the request port.

- see https://serverfault.com/questions/706438/what-is-the-difference-between-nginx-variables-host-http-host-and-server-na

and also configured `Websockets support` on the `proxy host` :

[image]

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP8266

zenguru84, 2023-05-26:
it's working.
now i can access esphome devices from outside network.
thank you!

HarlemSquirrel, 2023-06-11:
This worked for me after switching my setup from the "Let's Encypt" + "NGINX Home Assistant SSL proxy" add-ons to the “Nginx Proxy Manager” add-on and added the custom location and proxy_set_header directive.

henryabra, 2023-07-05:
Amazing! Worked for me :)

kellerza, 2023-07-30:
You can add Websocket support to the `/api/hassio_ingress` URI in the *NGINX Home Assistant SSL proxy* as well (source [here](https://community.home-assistant.io/t/esphome-web-ui-black-screen-for-validate-install-and-log/337578/16))

Add the following to `/share/nginx_proxy_default_ingress.conf`

```nginx
location /api/hassio_ingress {
    proxy_pass http://homeassistant.local.hass.io:8123;
    proxy_set_header Host $http_host;
    proxy_redirect http:// https://;
    proxy_http_version 1.1;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $connection_upgrade;
    proxy_set_header X-Forwarded-Host $http_host;
}
```

Configure the *NGINX Home Assistant SSL proxy* to read this new default, under **Customize**:

```yaml
active: true
default: nginx_proxy_default*.conf
servers: nginx_proxy/*.conf
```

mcarbonneaux, 2023-08-08:
i've not tested esphome with the NGINX Home Assistant SSL proxy because of other problem with it that are more simple to manage with proxy manager... is good to know  how to do it's with it !

john159753, 2023-10-09:
> You can add Websocket support to the `/api/hassio_ingress` URI in the _NGINX Home Assistant SSL proxy_ as well (source [...]

This did the trick for me, I of course needed to swap out specifics with my configuration, but this was the thing that led me down the right path.. Thanks!

henryabra, 2024-01-05:
Just wanted to add a clarification on what worked for me:
The exact steps as described by @mcarbonneaux  with one change: instead of adding in the custom configuration the value `proxy_set_header Host $http_host;` I have added `proxy_set_header X-Forwarded-Host $http_host;`

whc2001, 2024-07-06:
This used to work for me, however after I have reinstalled my server it no longer works. Not sure what's going wrong.
[image]
[image]

whc2001, 2024-07-06:
Okay seems like here is the reason:

[image]

Basically NginxProxyManager adds the custom config on the top of the default config content, making it being overridden.

1. Create the custom location config like this, also when clicking the dropdown menu take note of the proxy host ID (in my case 12):

[image]

[image]

2. Find the `data` directory of your NginxProxyManager instance, open the file `data/nginx/proxy_host/<ID>.conf` with text editor, find all lines saying `proxy_set_header Host $host;` and delete them all.

3. Restart NginxProxyManager (`docker container restart <name>`). Now the ESPHome addon console should work. DO NOT TOUCH WEBUI PROXY SETTINGS EVER AGAIN or you need to redo step 2!

[image]