esphome/issues #4328: Multiple I2C buses causes boot loop if I2C devices are not present at boot

fran6120, 2023-03-19:
### The problem

Due to communication problems with the i2c devices I have had to separate them into two buses.

My project works with battery, I make use of the deep_sleep function and therefore I need to disconnect the devices with the highest current consumption. I do this through a BJT transistor.

So far everything was working perfectly. On boot the esp32s2 (lolin s2 mini) through a GPIO, the BJT transistor "turns on" which acts as a switch, powering the I2C devices.

But now, with two I2C buses at boot, the transistor no longer "turns on" because the esp32 goes into bootloop.
If I "turn on" the BJT transistor manually to power the devices at boot or during bootloop there is no problem.

 ```yaml
esphome:
  name: estacions2
  platformio_options:
    board_build.f_cpu: 80000000L
  on_boot:
  - priority: 775 
    then:
      - output.turn_on: bc548c_switch
  on_loop:
    then:
      - script.execute: led_conexion

esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: esp-idf

api:

wifi:
  networks:
    - ssid: !secret wifi_ssid1
      password: !secret wifi_password1
      manual_ip:
        static_ip: 192.168.1.200
        gateway: 192.168.1.1
        subnet: 255.255.255.0
        dns1: 192.168.1.1
  fast_connect: true

deep_sleep:
  id: control_deep_sleep
  
i2c:
  - id: bus_0
    sda: GPIO17
    scl: GPIO21
    scan: true
    frequency: 100kHz
  - id: bus_1
    sda: GPIO38
    scl: GPIO40
    scan: true
    frequency: 100kHz

output:
  - platform: gpio 
    pin: GPIO16
    id: bc548c_switch
    inverted: false
  - platform: gpio
    pin: GPIO15
    inverted: false
    id: led

script:
  - id: led_conexion
    mode: single
    then:
      if:
        condition:
          wifi.connected:
        then:
          - output.turn_on: led
          - delay: 75ms 
          - output.turn_off: led
          - delay: 925ms
        else:
          - output.turn_off: led
```

You don't need to connect any I2C devices to the esp32, just switching from one bus to two buses causes the bootloop.

It is also not insufficient power problem since the error is reproduced with only the esp32 powered by USB.

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32-IDF

### Board

ESP32S2 (Lolin S2 Mini)

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  name: estacions2
  platformio_options:
    board_build.f_cpu: 80000000L
  on_boot:
  - priority: 775 
    then:
      - output.turn_on: bc548c_switch
  on_loop:
    then:
      - script.execute: led_conexion

esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: esp-idf

api:

wifi:
  networks:
    - ssid: !secret wifi_ssid1
      password: !secret wifi_password1
      manual_ip:
        static_ip: 192.168.1.200
        gateway: 192.168.1.1
        subnet: 255.255.255.0
        dns1: 192.168.1.1
  fast_connect: true

deep_sleep:
  id: control_deep_sleep
  
i2c:
  - id: bus_0
    sda: GPIO17
    scl: GPIO21
    scan: true
    frequency: 100kHz
  - id: bus_1
    sda: GPIO38
    scl: GPIO40
    scan: true
    frequency: 100kHz

output:
  - platform: gpio 
    pin: GPIO16
    id: bc548c_switch
    inverted: false
  - platform: gpio
    pin: GPIO15
    inverted: false
    id: led

script:
  - id: led_conexion
    mode: single
    then:
      if:
        condition:
          wifi.connected:
        then:
          - output.turn_on: led
          - delay: 75ms 
          - output.turn_off: led
          - delay: 925ms
        else:
          - output.turn_off: led
```

### Anything in the logs that might be useful for us?

```txt
From serial port:

[0;32mI (341) spi_flash: detected chip: generic[0m
[0;32mI (
abort() was called at PC 0x400ae55d on core 0

Backtrace: 0x40024c7e:0x3ffc71c0 0x4002b3fd:0x3ffc71e0 0x40031b06:0x3ffc7200 0x400ae55d:0x3ffc7270 0x40025282:0x3ffc7290 0x4011754f:0x3ffd1a60 0x400aebf7:0x3ffd1a80 0x4002c63c:0x3ffd1aa0

ELF file SHA256: 238372e0a42f6bfc

Rebooting...
```

### Additional information

From exception decoder:

Decoding stack results
0x40024c7e: panic_abort at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/panic.c line 408
0x4002b3fd: esp_system_abort at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/esp_system.c line 137
0x40031b06: abort at /config/.esphome/platformio/packages/framework-espidf/components/newlib/abort.c line 46
0x400ae55d: task_wdt_isr at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/task_wdt.c line 176
0x4011754f: esp_pm_impl_waiti at /config/.esphome/platformio/packages/framework-espidf/components/hal/esp32s2/include/hal/cpu_ll.h line 202
0x400aebf7: esp_vApplicationIdleHook at /config/.esphome/platformio/packages/framework-espidf/components/esp_system/freertos_hooks.c line 63
0x4002c63c: prvIdleTask at /config/.esphome/platformio/packages/framework-espidf/components/freertos/tasks.c line 3987

ssieb, 2023-03-19:
That looks like it's not even getting to any esphome code yet.  Although you haven't even configured the logger, so it's hard to say.  Turn on verbose logging and then paste logs.

fran6120, 2023-03-19:
The same code with one bus works, with two buses the error is reproduced.  When there is a failure, I do not have a log from the "logger" because the esp32 does not turn on, it is not even capable of setting a GPIO high.

Maybe the failure is not esphome...

ssieb, 2023-03-19:
If that is your full yaml, you don't have the logger at all.  But that script is going to cause you trouble as well.  You should not be running that from `on_loop:`.  There are much better ways to do that.

fran6120, 2023-03-20:
Trying to do tests to clarify the issue I have verified the following:

this yaml works correctly:

```yaml
esphome:
  name: estacions2
  platformio_options:
    board_build.f_cpu: 80000000L
  on_boot:
  - priority: 775
    then:
      - output.turn_on: bc548c_switch

esp32:
  board: lolin_s2_mini
  variant: ESP32S2
  framework:
    type: esp-idf

logger:
  baud_rate: 57600
  level: VERY_VERBOSE
  hardware_uart: USB_CDC

api:

ota:
  password: "9f3432252c737a4f6429e8074af07543"

wifi:
  networks:
    - ssid: !secret wifi_ssid1
      password: !secret wifi_password1
      manual_ip:
        static_ip: 192.168.1.200
        gateway: 192.168.1.1
        subnet: 255.255.255.0
        dns1: 192.168.1.1
  fast_connect: true

deep_sleep:
  id: control_deep_sleep
  
i2c:
  - id: bus_0
    sda: GPIO17
    scl: GPIO21
    scan: true
    sda_pullup_enabled: False
    scl_pullup_enabled: False
    frequency: 100kHz

output:
  - platform: gpio
    pin: GPIO16
    id: bc548c_switch
    inverted: true
```

this other causes bootloop:

```yaml
esphome:
  name: estacions2
  platformio_options:
    board_build.f_cpu: 80000000L
  on_boot:
  - priority: 775
    then:
[... 857 lines cut ...]
  esphome_version: '2023.3.0'
  compilation_time: 'Mar 20 2023, 01:27:53'
  model: 'lolin_s2_mini'
  has_deep_sleep: YES
  project_name: ''
  project_version: ''
  webserver_port: 0
  bluetooth_proxy_version: 0
  manufacturer: 'Espressif'
  friendly_name: ''
}[0m
[0;38m[VV][api.service:513]: on_list_entities_request: ListEntitiesRequest {}[0m
[0;38m[VV][api.service:055]: send_list_entities_done_response: ListEntitiesDoneResponse {}[0m
[0;38m[VV][api.service:522]: on_subscribe_states_request: SubscribeStatesRequest {}[0m
[0;38m[VV][api.service:584]: on_subscribe_homeassistant_services_request: SubscribeHomeassistantServicesRequest {}[0m
[0;38m[VV][api.service:611]: on_subscribe_home_assistant_states_request: SubscribeHomeAssistantStatesRequest {}[0m
[0;38m[VV][api.service:486]: on_ping_request: PingRequest {}[0m
[0;38m[VV][api.service:043]: send_ping_response: PingResponse {}[0m
[0;38m[VV][api.service:486]: on_ping_request: PingRequest {}[0m
[0;38m[VV][api.service:043]: send_ping_response: PingResponse {}[0m
[0;38m[VV][scheduler:225]: Running interval '' with interval=60000 last_execution=4294961846 (now=54553)[0m
[0;38m[VV][api.service:486]: on_ping_request: PingRequest {}[0m
[0;38m[VV][api.service:043]: send_ping_response: PingResponse {}[0m
```
Using arduino framework instead of esp-idf reproduces the same error but I don't have log because arduino framework doesn't support "USB CDC" as hardware uart

ssieb, 2023-03-20:
`[E][i2c.idf:254]: Recovery failed: SCL is held LOW on the I2C bus` which results in `[VV][i2c.idf:204]: TX to 08 failed: timeout` instead of `[VV][i2c.idf:201]: TX to 51 failed: not acked`.  You don't have any time information in the logs, but I suspect it's taking so long that it's triggering the WDT.  What happens if you have the single i2c using the second set of pins?

ssieb, 2023-03-20:
You could also try enabling the pullups.

fran6120, 2023-03-20:
I disable the internal pull-ups because I use 4.7kohm external ones, I have even removed the pull-up resistors that the i2c modules come with to only have a couple of resistors per bus. Even so, I have tried to enable the internal ones but the result is the same. I don't think this is the problem.

I have done more tests:

Anyway, I think the problem is linked to the use of the transistor as a switch, because if you power up the sensors directly, the esp32 turns on normally.

With a minimal configuration yaml, with a gpio to drive the transistor and two i2c buses, it is capable of turning on even without connected devices.

If I connect only one device on each bus, it still turns on, but doesn't find any devices.

If I add any more device on either of the two buses, the esp32 goes into bootloop.

If I connect all devices (7) but set i2c scan to false
the esp32 turns on, and in this situation, if i set in the yaml the 7 devices with i2c scan to false on both buses, the esp32 turns on normally most of the time, the gpio drives the transistor and the devices return data to homeassistant , but sometimes (very few) it goes into bootloop.

Maybe i2c has a preset timeout and it times out

fran6120, 2023-03-21:
> You need to figure out why it's getting `[E][i2c.idf:254]: Recovery failed: SCL is held LOW on the I2C bus` on the second i2c bus, especially since you say you have pullups. This really sounds like a hardware problem. And yes, it's timing out and triggering the WDT because the scan is taking too long when it times out on every address.

On the second i2c bus I have 4 devices vs. two devices I have on the first bus.

Also, in the first bus one of the devices is an RTC, it takes its own battery and it is the only one that is always shown when the scan option is true.

If I swap the devices on the first bus to the second bus, and the devices on the second bus to the first bus, then the `[E][i2c.idf:254]: Recovery failed: SCL is held LOW` is displayed on the first bus instead of in the second as up to now, so in principle everything indicates that the failure is from the sensors or the circuit.

I have checked the connection of the pull-up resistors of both buses and they are exactly the same.

I have been disconnecting the sensors from the second bus one by one and reviewing the log, but `[E][i2c.idf:254]: Recovery failed: SCL is held LOW` is reproduced randomly and always on the bus where those sensors are.

But doing more tests it seems that [E][i2c.idf:254]: Recovery failed: SCL is held LOW on the I2C bus only happens when the sensors are powered through the transistor driven by the GPIO. If the sensors are directly powered on both buses, "[C][i2c.idf:053]: Recovery: bus successfully recovered" is displayed, and if the scan option is set to true on both buses, all devices are displayed on the log

I don't know how i2c devices work once they are powered, but it seems as if the devices at the time of establishing communication, these, are not yet ready due to the delay in driving the transistor through the GPIO.

fran6120, 2023-03-21:
At the moment this can be solved by changing the configuration priority of the GPIO that drives the transistor in the on_boot trigger.

The i2c bus configuration priority is 1000 compared to 800 for the outputs, so enabling the GPIO with priority 1000 seems to be enough for now.

```yaml
  on_boot:
  - priority: 1000
    then:
      - lambda: |-
          gpio_pad_select_gpio(GPIO_NUM_16);
          gpio_set_direction(GPIO_NUM_16, GPIO_MODE_OUTPUT);
          gpio_set_level(GPIO_NUM_16, 1);
```

esp32 turns on normally, without failures on both i2c buses even with the scan option set to true

```
[11:02:10][C][i2c.idf:047]: I2C Bus:
[11:02:10][C][i2c.idf:048]:   SDA Pin: GPIO17
[11:02:10][C][i2c.idf:049]:   SCL Pin: GPIO21
[11:02:10][C][i2c.idf:050]:   Frequency: 400000 Hz
[11:02:10][C][i2c.idf:053]:   Recovery: bus successfully recovered
[11:02:10][I][i2c.idf:063]: Results from i2c bus scan:
[11:02:10][I][i2c.idf:069]: Found i2c device at address 0x1E
[11:02:10][I][i2c.idf:069]: Found i2c device at address 0x53
[11:02:10][I][i2c.idf:069]: Found i2c device at address 0x58
[11:02:10][I][i2c.idf:069]: Found i2c device at address 0x76
[11:02:10][C][i2c.idf:047]: I2C Bus:
[11:02:10][C][i2c.idf:048]:   SDA Pin: GPIO38
[11:02:10][C][i2c.idf:049]:   SCL Pin: GPIO40
[11:02:10][C][i2c.idf:050]:   Frequency: 400000 Hz
[11:02:10][C][i2c.idf:053]:   Recovery: bus successfully recovered
[11:02:10][I][i2c.idf:063]: Results from i2c bus scan:
[11:02:10][I][i2c.idf:069]: Found i2c device at address 0x40
[11:02:10][I][i2c.idf:069]: Found i2c device at address 0x51
```

With this I think the issue can be closed and perhaps open a "feature request" requesting some option like "very_early_init" for the outputs to configure them as priority 1000.

Thanks for the help!

ssieb, 2023-03-21:
You should have the same effect by adding `setup_priority: 1000` to the gpio output.

closed 2023-03-21 by ssieb