esphome/issues #4329: PZEM-004T - detect missing data (if power meter is not powered)

adynis, 2023-03-20:
### The problem

If (after working fine) suddenly the PZEM-004T stopps sending data (e.g.: the AC power of PZEM-004T is interrupted) then ... nothing happends: the last value remain active as if everyrhing is normal !? 
How to detect that actually data is not recent anymore ? that data is not send anymore?

I would like to see a NaN (undefined/unavailable/something else) in case of missing data, or at least to be able to detect when PZEM-004T device does not communciate anymore. Is there any solution?

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32

BrainForge, 2023-07-19:
I have the same problem, any ideas how to solve it?

LevKos, 2024-01-19:
I also faced a similar problem.
And it lies in the fact that if there is no power to the high-voltage part, it does not output anything, but in the driver it is processed that if there is nothing, then the values do not change.
I suggest making the following changes to pzemac.cpp:

void MYPZEMAC::on_modbus_data(const std::vector<uint8_t> &data) {
  if (data.size() < 20) {
    ESP_LOGW(TAG, "Invalid size for PZEM AC!");
// add start
    this->voltage_sensor_->publish_state(0);
    this->current_sensor_->publish_state(0);
    this->power_sensor_->publish_state(0);
    this->frequency_sensor_->publish_state(0);
    this->power_factor_sensor_->publish_state(0);
//add stop
    return;
  }
.....