esphome/issues #4331: ILI9xxx with ESP8266 boot loop crash

IgorYbema, 2023-03-20:
### The problem

Using the ILI9xxx on a ESP8266 board seems to boot loop / crash as soon as the ili9xxx config is added

### Which version of ESPHome has the issue?

2023.3.0 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

ili9xxx

### Example YAML snippet

```yaml
esphome:
  name: esphome-test
  friendly_name: esphome-test

esp8266:
  board: esp01_1m

logger:
  level: VERY_VERBOSE

spi:
  clk_pin: 14
  mosi_pin: 13
  miso_pin: 12

display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: 5
    dc_pin: 4
```

### Anything in the logs that might be useful for us?

```txt
SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
[I][logger:258]: Log initialized
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[C][spi:023]: Setting up SPI bus...
[VV][scheduler:062]: set_interval(name='', interval=60000, offset=28180)
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40205bce, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4023c0aa, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40205bce epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023c0aa depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd20 end: 3fffffd0 offset: 0190
3ffffeb0:  02625a00 00000001 3ffefda4 40201ff0  
3ffffec0:  02625a00 00000000 00000005 00000003  
3ffffed0:  00000003 4023c0aa 3ffefe34 4020243d  
3ffffee0:  3ffefe34 00000003 4023c0aa 4020248a  
3ffffef0:  000000ef 40204b59 00000004 4020664b  
3fffff00:  00000000 00000001 00000005 00000010  
3fffff10:  3ffefe34 3ffefe48 3ffefe34 402024c1  
3fffff20:  3fffff28 00000000 3ffefe34 40201ef7  
3fffff30:  00000000 00000003 3ffefe34 402045be  
3fffff40:  00000000 00000000 40204828 3ffee8d4  
3fffff50:  3ffee8d8 00000002 3ffee8d4 402051a5  
3fffff60:  00000000 009c1001 00000000 0000000c  
3fffff70:  3ffefdec 00000002 3ffee8d4 402040b1  
3fffff80:  00000004 00000004 3ffeff64 feefeffe  
3fffff90:  00000010 00000001 feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3ffeeb0c  
3fffffb0:  3fffdad0 00000000 3ffeeaf8 40206920  
3fffffc0:  feefeffe feefeffe 3ffe85d8 4010133d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```
```

nielsnl68, 2023-03-20:
I will check it tomorrow. 

But just to be sure. Is this the whole part of the display component. Could you test this with a simple example `lambda: ` code?

IgorYbema, 2023-03-21:
Just tried:
```
    lambda: |-
      it.line(0, 0, 50, 50);
```

Doesn't help.

Also, when adding color_pallette: GRAYSCALE (because I thought maybe memory related) I get this compiling error:
```
INFO Reading configuration /config/esphome/esphome-test.yaml...
INFO Generating C++ source...
ERROR ID uint8_t_2 is already registered
```

IgorYbema, 2023-03-21:
And yes, this is the whole yaml file. Nothing more. Also tried on other esp8266 boards (other brands, types). Always same issue.

IgorYbema, 2023-03-23:
@nielsnl68 found time to test it?

nielsnl68, 2023-03-23:
Sadly not yet. I'm trying to resolve an other issue with a touchscreen driver first atm.

workingmanrob, 2023-03-23:
Should it work at all with an 8266?

I never had any success with an 8266 driving a display.

disconn3ct, 2023-04-18:
"Same here." Prior to flashing esphome, it was working stand-alone controlling the display and reacting to the touchscreen. (My board is an old nodemcu.)

enra64, 2023-05-29:
I have the same issue, happy to provide any data if it helps. The issue seems quite deep in the code or protocol or whatever, so THB I don't think I can help with code. This is what I found: https://arduino.stackexchange.com/questions/67442/nodemcu-1-0-exception-9-fatal-exception-9loadstorealignmentcause

dosanito, 2023-08-26:
Hi,
I run into the same issue. Is there a way to use an older library or did anyone find a solution on this?
If not, is there a way to include Arduino libraries? The display is working with a sketch, but I would like to use the display in HA.

Would be happy if it works.

nielsnl68, 2023-08-26:
please provide some log messages showing when and how it is crashing.
Also i like to see your complete YAML file so i can see how you config the display.

dosanito, 2023-08-27:
Hi,
please find here the .yaml after captive portal.

color:
  - id: my_text_color
    red: 100%
    green: 100%
    blue: 100%

spi:
  clk_pin: D5
  mosi_pin: D7
  miso_pin: D6

output:
  - id: screen_backlight_pin
    platform: esp8266_pwm
    frequency: 1000 Hz
    pin: D8

light:
  - platform: monochromatic
    output: screen_backlight_pin
    name: "WZ Backlight display"
    id: screen_backlight
    restore_mode: RESTORE_DEFAULT_OFF
    on_turn_on:
      - delay: 60s
      - light.turn_off: screen_backlight

display:
  - platform: ili9xxx
    model: ILI9341
    cs_pin: D1
    dc_pin: D2
    rotation: 270
    id: screen

I tried with lambda to output something and without. As soon as the ili9xxx display is present, the issue with crash during boot happens,

Here is the complete log, Finally the devices goes into safe boot. 

As soon as I remove the display, all works fine.

[I][logger:262]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][spi:023]: Setting up SPI bus...
[C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x4022a5e2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40269f7a depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc10 end: 3fffffc0 offset: 0190
3ffffda0:  02625a00 3fff1444 3fff19ec 40211c54  
3ffffdb0:  00000003 3ffefc90 00000003 4022a649  
3ffffdc0:  02625a00 40260001 3fff18e4 00000003  
3ffffdd0:  3fff1444 40269f7d 40269f7a 40212b88  
3ffffde0:  00000003 40269f7a 3fff18e4 40212f90  
3ffffdf0:  3fff18e4 00000003 40269f7a 40212fda  
3ffffe00:  000000ef 3ffffe20 00000004 40100bc4  
3ffffe10:  00000000 00000006 3fff18e4 3fff14b0  
3ffffe20:  3fff18e4 3fff18f8 3fff18e4 40213011  
3ffffe30:  3fff1508 00000004 3fff18e4 40212ae7  
[... 609 lines cut ...]
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0007f2f0
~ld
[I][logger:262]: Log initialized
[C][ota:469]: There have been 10 suspected unsuccessful boot attempts.
[E][ota:476]: Boot loop detected. Proceeding to safe mode.
[I][app:029]: Running through setup()...
[C][wifi:038]: Setting up WiFi...
[C][wifi:048]: Starting WiFi...
[C][wifi:049]:   Local MAC: 48:3F:DA:04:BA:22
[D][wifi:425]: Starting scan...
[D][wifi:440]: Found networks:
[I][wifi:483]: - 'togobell' (5C:49:79:9F:A3:84) [redacted]▂▄▆█

dosanito, 2023-08-27:
Ahh I missed the board.
esp8266:
  board: d1_mini

dosanito, 2023-08-27:
Hi Nielsnl68,
I found an article on az-delivery website (company selling the az-touch) and they described the buffer calculation in ili9xxx in esphome is calculating display buffer memory which is too much for esp8266 device. The do not describe more details. So they recommend to use esp32.
It is in German here. 
https://www.az-delivery.de/blogs/azdelivery-blog-fur-arduino-und-raspberry-pi/terminanzeige-mit-az-touch-mod-am-esp32-mqtt-node-red-und-esphome-teil-3 

So I decided to go with esp32. If you are a core developer and have ideas to correct this in esphome, I would appreciate this, if you decide to stay as is, it would be helpful for newbies like me to see a comment that esp8266 is not supported by ili9xxx module.
But I would support if you want to test changes.

enra64, 2023-08-27:
Yeah I just went with ESP32 in the end and it worked fine. I think putting a hint that ESP8266 is not supported has by far the best cost-benefit ratio here, as the price difference is so small. If somebody really needs esp8266 later they can fix it :D

GitHubStefan13, 2023-12-14:
Same issue with ESP8266, D1-Mini
Just providing the Logs as additional help to narrow down the issue.

Here is the config File:
esphome:
  name: bugtest   
   
esp8266:
  board: d1_mini

spi:
    miso_pin: D3 # SD0
    mosi_pin: D2 # SDI
    clk_pin:  D4 # CLK

logger:
  level: DEBUG #VERBOSE
      
display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: D1
    dc_pin: D6
    reset_pin: D5

And the Stracktrace: (ESPHome 23.11.6)

[15:25:00][I][logger:326]: Log initialized
[15:25:00][I][app:029]: Running through setup()...
[15:25:00][D][spi:039]: Setting up SPI bus...
[15:25:00][D][spi_device:355]: mode 0, data_rate 40000kHz
[15:25:00]
[15:25:00]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:25:00]
[15:25:00]Exception (3):
WARNING Exception type: LoadStoreError: Processor internal physical address or data error during load or store
[15:25:00]epc1=0x4020990e epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023caca depc=0x00000000
WARNING Decoded 0x4020990e: esphome::spi::SPIDelegate::write_array(unsigned char const*, unsigned int)
[15:25:00]
[15:25:00]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:25:00]
[15:25:00]ctx: cont
[15:25:00]sp: 3ffffd50 end: 3fffffd0 offset: 0190
[15:25:00]3ffffee0:  40105cef 00000001 0000000c 401003a4
WARNING Decoded 0x40105cef: ets_timer_disarm
WARNING Decoded 0x401003a4: digitalWrite
[15:25:00]3ffffef0:  40105cb6 00000000 00000005 401003a4
WARNING Decoded 0x40105cb6: wdt_feed
WARNING Decoded 0x401003a4: digitalWrite
[15:25:00]3fffff00:  00000003 4023caca 3ffefcc4 40201fc2
WARNING Decoded 0x4023caca: ethernet_input
WARNING Decoded 0x40201fc2: esphome::ili9xxx::ILI9XXXDisplay::send_command(unsigned char, unsigned char const*, unsigned char)
[15:25:00]3fffff10:  3ffefcc4 00000003 4023caca 4020213a
WARNING Decoded 0x4023caca: ethernet_input
WARNING Decoded 0x4020213a: esphome::ili9xxx::ILI9XXXDisplay::init_lcd_(unsigned char const*)
[15:25:00]3fffff20:  000000ef 3ffeeaf0 0000000a 3ffeeaf0
[15:25:00]3fffff30:  3ffefcc4 3ffefcbc 40206b75 3ffefcbc
WARNING Decoded 0x40206b75: esp_yield
[15:25:00]3fffff40:  3ffefcc4 3ffefcd8 3ffefcc4 40202171
WARNING Decoded 0x40202171: esphome::ili9xxx::ILI9XXXILI9341::initialize()
[15:25:00]3fffff50:  4020208b 3ffefcc4 3ffefcc4 40202099
WARNING Decoded 0x4020208b: esphome::ili9xxx::ILI9XXXDisplay::setup()
WARNING Decoded 0x40202099: esphome::ili9xxx::ILI9XXXDisplay::setup()
[15:25:00]3fffff60:  00000010 00000003 3ffefcc4 40204b06
WARNING Decoded 0x40204b06: esphome::PollingComponent::call_setup()
[15:25:00]3fffff70:  3ffefc94 00000002 3ffee85c 402045c5
WARNING Decoded 0x402045c5: esphome::Application::setup()
[15:25:00]3fffff80:  00000004 00000004 3ffefe04 feefeffe
[15:25:00]3fffff90:  00000010 02625a00 00000010 feefeffe
[15:25:00]3fffffa0:  feefeffe feefeffe feefeffe 3ffeea88
[15:25:00]3fffffb0:  3fffdad0 00000000 3ffeea74 40206cac
WARNING Decoded 0x40206cac: loop_wrapper() at core_esp8266_main.cpp
[15:25:00]3fffffc0:  feefeffe feefeffe 3ffe85d8 40100fcd
WARNING Decoded 0x40100fcd: cont_wrapper
[15:25:00]<<<stack<<<
[15:25:00]
[15:25:00]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:25:00]
[15:25:00] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[15:25:00]
[15:25:00]load 0x4010f000, len 3460, room 16
[15:25:00]tail 4
[15:25:00]chksum 0xcc
[15:25:00]load 0x3fff20b8, len 40, room 4
[15:25:00]tail 4
[15:25:00]chksum 0xc9
[15:25:00]csum 0xc9
[15:25:00]v00046fd0
[15:25:00]~ld

Grey-Lancaster, 2024-01-08:
Same problem here with a D1 mini esp8266.  FWIW, I could not get an ESP32 min to work either.  I have seen an example with ili9341 that works, but the same code with ili9xxx results in this boot loop

dison4linux, 2024-02-01:
I also ran into this issue with D1-Mini (ESP8266) and ended up here after some googling...
I tried with a Lolin/Wemos D32 Pro and I had no luck there either, though the error message was a little different.
With either board it won't boot until it fails boot a few times and then tries safe mode (which I assume disables the SPI and display components)

dison4linux, 2024-02-02:
Config:
```
esphome:
  name: ili9341
  
esp32:
  board: lolin_d32_pro

wifi:
  ssid: "redacted"
  password: "redacted"
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ili9341 Fallback Hotspot"

captive_portal:

# Enable logging
logger:
# Enable Home Assistant API
api:

# Enable OTA Updates
ota:

spi:
  clk_pin: GPIO14 # 5 D5
  mosi_pin: GPIO13 # 7 D7
  miso_pin: GPIO12 # 6 D6

color:
  - id: red
    red: 100%
    green: 3%
    blue: 3%

display:
  - platform: ili9xxx
    model: ili9341
    dc_pin: GPIO15 #16 D8
    cs_pin: GPIO16 # 4 D0
    reset_pin: 1
    lambda: |-
      it.fill(Color::BLACK);
      it.print(0, 0, id(font18), id(red), TextAlign::TOP_LEFT, "Hello World!");

font:
  - file: "gfonts://Roboto"
    id: font18
    size: 18
```
Logs:
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:351]: Log initialized
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[... 117 lines cut ...]
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:351]: Log initialized
[C][ota:473]: There have been 9 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through se�ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[I][logger:351]: Log initialized
[C][ota:473]: There have been 10 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[E][ota:480]: Boot loop detected. Proceeding to safe mode.
```

nielsnl68, 2024-02-02:
could you run without the display and make sure it has to do with the display. because i do noting about the display.
Also can you set the logging level to verbose.

dison4linux, 2024-02-02:
I use this same ESP32 to drive an E-Paper display without issue, both use SPI but different pins.  I just went in and adjusted the pins for the ILI9341 to match what I use for the E-Paper display and now the ESP32 boots up okay.  So this seems to have to do with which pins are used for SPI not so much the ILI9341 display setup.  The issue is though, that this display is a sort of "shield" or "hat" meaning that the ESP plugs into it and we don't have control over which pin goes where.  There are also pins to do manual wiring but its much more convenient to use the socket for the ESP.  This is the display I'm referring to: [https://www.wemos.cc/en/latest/d1_mini_shield/tft_2_4.html](https://www.wemos.cc/en/latest/d1_mini_shield/tft_2_4.html)

nielsnl68, 2024-02-02:
I have the same display running on a S2 D1-mini. And it works fine. Just for others to compare do you mind showing which pins you now use.

dison4linux, 2024-02-02:
The pins I'm using now are below, but that's just a copy&paste from another config file for another display ... I don't actually have the display connected and working from these pins, I just wanted to validate whether or not the ESP would boot after changing pins and it does.
```
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO12
...
display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: GPIO14
    dc_pin: GPIO27
    reset_pin: GPIO33
```
But it does _not_ boot when configured to use the pins as they are laid out in the socket on the back of the display.
```
spi:
  clk_pin: GPIO14 # 5 D5
  mosi_pin: GPIO13 # 7 D7
  miso_pin: GPIO12 # 6 D6
...
display:
  - platform: ili9xxx
    model: ili9341    
    dc_pin: GPIO15 #16 D8
    cs_pin: GPIO16 # 4 D0
    reset_pin: 1
```
Can you share your config for the S2?  I might have one of those laying around I could try.

Grey-Lancaster, 2024-02-02:
This works with an esp32 D1 mini plugged straight into the back of the Lolin 2.4 TFT I don't think I got an 8266 to work

```
esphome:
  name: esp32tft3
  friendly_name: ESP32TFT3

  platform: ESP32
  board: mhetesp32minikit

spi:
 clk_pin: 18
 mosi_pin: 23
 miso_pin: 19
 
# Example configuration entry

display:
  - platform: ili9xxx
    model: ili9341
    cs_pin: 26
    dc_pin: 5 
    reset_pin: 33
```

nielsnl68, 2024-02-02:
Sure, here it is incl. touchscreen setup:

```yaml
spi:
  clk_pin: 7
  mosi_pin: 11
  miso_pin: 9

display:
- platform: ili9xxx
  id: my_display
  model: ili9341
  cs_pin: 5
  dc_pin: 12
  reset_pin: 35
  auto_clear_enabled: false
  data_rate: 40MHz
  dimensions: 320x240
  update_interval: never
  transform:
    mirror_y: false
    mirror_x: false
    swap_xy: true

touchscreen:
- platform: xpt2046
  display: my_display
  id: my_toucher
  update_interval: 50ms
  cs_pin: 18
  #interrupt_pin: 16

  threshold: 300
  calibration:
    x_min: 210
    x_max: 3890
    y_min: 170
    y_max: 3730
  transform:
    mirror_x: false
    mirror_y: true
    swap_xy: true
'''