esphome/issues #4332: Esp32 internal temperature looks Fahreinheit-ic on M5Stack Atom Lite

Roofcatbusy, 2023-03-20:
### The problem

On M5Stack Atom lite matchbox internal temerture sensor shows 73...85 degreees which is a bit odd (box is small and I can touch it easily) According to manufacturer Atom Lite is ESP32-PICO based, and following definition was used while configuring esphome for the board
```
esp32:
  board: m5stick-c
  framework:
    type: arduino
```
For me it looks like F->C conversion were not used.

### Which version of ESPHome has the issue?

2023.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.3.5

### What platform are you using?

ESP32

### Board

M5Stack Atom Lite

### Component causing the issue

internal_temperture

dd32, 2023-04-14:
The original ESP32 doesn't have a temperature sensor that's reliable.
Early revisions included one, but later revisions removed the API for it.

The ESP-S2, C3, and S3 do include temperature sensors.

You can still use it on an ESP32, but, as you found, the value may be highly inaccurate.

You'll find the F => C conversion here:
https://github.com/esphome/esphome/blob/dev/esphome/components/internal_temperature/internal_temperature.cpp#L27-L31

Mat931, 2023-06-28:
The internal temperature readings on the original ESP32 are unusually high. If you want more accurate readings you can use a linear calibration:

```yaml
sensor:
  - platform: internal_temperature
    name: "Internal Temperature"
    filters:
      - calibrate_linear:
          # Map 0.0 (from internal_temperature sensor) to 1.0 (accurate temperature measurement)
          - 0.0 -> 1.0
          - 10.0 -> 12.1
```