esphome/issues #4334: on_value_changed for the Sensor Component

Seeelefant, 2023-03-22:
### The problem

Currently there are only the on-value, on-value-range, on-raw-value and sensor-in-range-condition events for the sensor component. I would like to have something like on_value_changed for the Sensor Component, which might help to make reactions more intelligent.

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

ssieb, 2023-03-22:
Add a `delta` filter.

closed 2023-03-22 by ssieb

Seeelefant, 2023-03-22:
Hi Samuel, this is not the solution I am looking for. In my understanding each sensor has an  update_interval, with your proposal I would get a trigger after the update_interval **and** a change of the measurement-I was looking for an event driven architecture, i.e. after a change of the sensor value  an immediate trigger. What do you think?

ssieb, 2023-03-22:
The sensor only measures on the update interval.  That's what it's for.  There is no such thing as continuous measurement.