esphome/issues #4338: Problem BME680 BSEC with ESP32-S2

ldelvalleh, 2023-03-23:
### The problem

The problem comes when I compile the code where I try to configure a BME680 sensor with a Lolin Wemos D1 Mini ESP32-S2 board.

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.3.1

### What platform are you using?

ESP32

### Board

Lolin S2 mini

### Component causing the issue

bme680_bsec

### Example YAML snippet

```yaml
esphome:
  name: pruebas2

esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "SSID"
  password: "PASS"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: "PASS"

captive_portal:
    
i2c:
  scan: true

bme680_bsec:
  address: 0x77
  temperature_offset: 0
  iaq_mode: static
  sample_rate: ulp
  state_save_interval: 6h

[... 206 lines cut ...]
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-rgb-led.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-time.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\libb64\cdecode.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\libb64\cencode.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\main.cpp.o
Compiling .pioenvs\pruebas2\FrameworkArduino\stdlib_noniso.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\wiring_pulse.c.o
Compiling .pioenvs\pruebas2\FrameworkArduino\wiring_shift.c.o
Archiving .pioenvs\pruebas2\libFrameworkArduino.a
Linking .pioenvs\pruebas2\firmware.elf
c:/users/luis/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: cannot find -lalgobsec
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\pruebas2\firmware.elf] Error 1
```

ssieb, 2023-03-23:
This is a problem with using proprietary software. There is no version for the S2 and no source to compile.
https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME-680-BSEC-for-ESP32-S2/td-p/16892
There is nothing we can do about this, other than fail the config if you try to compile it for an S2.

ssieb, 2023-03-24:
https://github.com/esphome/esphome/pull/4616 gives a proper error message now if you try to compile with esp-idf or one of the ESP32 variants.

closed 2023-03-24 by ssieb

ldelvalleh, 2023-03-24:
Thank you very much, I thought so.

TamasKazsoki, 2023-03-24:
Actually there is a solution i was able to found thanks to ssieb's reply to me last day.

The original BSEC module is old and cannot handle newer devices. Bosch is developing a new module called BSEC2 which supports ESP32-S2, S3 and C3 models as well. Also it supports BME680 and the newer BME688 as well. @schluggi created a fork of the ESPHome module and also initiated changes to the ESPHome module as well. Until that is merged, you can use the new module with lolin_s2_mini.

You have to use SchLuggi 's BSEC2 module with defining it as an external component:

`
external_components:
  - source: github://Schluggi/esphome@bsec2
    components:
      - bme680_bsec
`

The defining address: 0x77 for BME680 as the default here is the using of 0x76. Fortunately, you defined it in your config

This is the other thread: [https://github.com/esphome/feature-requests/issues/2072](url)

fran6120, 2023-04-15:
The same bme680 configuration in yaml with esphome 2023.5.0-dev fails to compile both esp32-s2 and esp32-c3.

Using framework esp-idf:
```
INFO Reading configuration /config/estacion-c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing estacionc3 (board: lolin_c3_mini; framework: espidf; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Library Manager: Installing SPI
Warning! Could not find the package with 'SPI' requirements for your system 'linux_x86_64'
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/estacionc3/src/esphome/components/bme680_bsec/bme680_bsec.o
Compiling .pioenvs/estacionc3/src/esphome/components/button/button.o
Compiling .pioenvs/estacionc3/src/esphome/components/deep_sleep/deep_sleep_component.o
Compiling .pioenvs/estacionc3/src/esphome/components/esp32/core.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.cpp:1:
src/esphome/components/bme680_bsec/bme680_bsec.h:12:10: fatal error: bsec2.h: No such file or directory

***************************************************************
* Looking for bsec2.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bsec2.h"
* Web  > https://registry.platformio.org/search?q=header:bsec2.h
*
***************************************************************

 #include <bsec2.h>
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/estacionc3/src/esphome/components/bme680_bsec/bme680_bsec.o] Error 1
========================== [FAILED] Took 3.52 seconds ==========================
```

using arduino framwork:
```
INFO Reading configuration /config/estacion-c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing estacionc3 (board: lolin_c3_mini; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- BME68x Sensor library @ 1.1.40407+sha.16984f4
|-- BSEC2 Software Library @ 1.5.2400+sha.a8fe04a
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling .pioenvs/estacionc3/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
Compiling .pioenvs/estacionc3/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/estacionc3/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/estacionc3/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling .pioenvs/estacionc3/src/esphome/components/gpio/output/gpio_binary_output.cpp.o
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'virtual void esphome::bme680_bsec::BME680BSECComponent::setup()':
[... 46 lines cut ...]
                          ^~~~~~~~~~~~~
                          bsec_do_steps_m
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::load_state_()':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:462:26: error: 'bsec_set_state' was not declared in this scope
     this->bsec_status_ = bsec_set_state(state, BSEC_MAX_STATE_BLOB_SIZE, work_buffer, sizeof(work_buffer));
                          ^~~~~~~~~~~~~~
src/esphome/components/bme680_bsec/bme680_bsec.cpp:462:26: note: suggested alternative: 'bsec_set_state_m'
     this->bsec_status_ = bsec_set_state(state, BSEC_MAX_STATE_BLOB_SIZE, work_buffer, sizeof(work_buffer));
                          ^~~~~~~~~~~~~~
                          bsec_set_state_m
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::save_state_(uint8_t)':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:482:7: error: 'bsec_get_state' was not declared in this scope
       bsec_get_state(0, state, BSEC_MAX_STATE_BLOB_SIZE, work_buffer, BSEC_MAX_STATE_BLOB_SIZE, &num_serialized_state);
       ^~~~~~~~~~~~~~
src/esphome/components/bme680_bsec/bme680_bsec.cpp:482:7: note: suggested alternative: 'bsec_get_state_m'
       bsec_get_state(0, state, BSEC_MAX_STATE_BLOB_SIZE, work_buffer, BSEC_MAX_STATE_BLOB_SIZE, &num_serialized_state);
       ^~~~~~~~~~~~~~
       bsec_get_state_m
Compiling .pioenvs/estacionc3/src/esphome/components/homeassistant/binary_sensor/homeassistant_binary_sensor.cpp.o
Compiling .pioenvs/estacionc3/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
*** [.pioenvs/estacionc3/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
========================== [FAILED] Took 2.73 seconds ==========================
```

any suggestions?

ssieb, 2023-04-15:
It might not work with esp-idf, but regardless, this is not the place to discuss that.

raccettura, 2023-05-03:
[image]
Looks like BSEC 2.4 is the solution per:
https://www.bosch-sensortec.com/media/boschsensortec/downloads/bsec/bsec1_binary_size_information.pdf

maikischa, 2023-05-28:
I'm running into exact the same issue:

INFO Reading configuration /config/esp32-s3-devkit.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-s3-devkit (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- BME68x Sensor library @ 1.1.40407+sha.16984f4
|-- BSEC2 Software Library @ 1.5.2400+sha.a8fe04a
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/sensor/filter.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/sensor/sensor.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/spi/spi.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/st7735/st7735.cpp.o
Compiling .pioenvs/esp32-s3-devkit/src/esphome/components/switch/automation.cpp.o
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'virtual void esphome::bme680_bsec::BME680BSECComponent::setup()':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:23:24: error: 'bsec_init' was not declared in this scope
   this->bsec_status_ = bsec_init();
                        ^~~~~~~~~
src/esphome/components/bme680_bsec/bme680_bsec.cpp:23:24: note: suggested alternative: 'bsec_init_m'
   this->bsec_status_ = bsec_init();
                        ^~~~~~~~~
                        bsec_init_m
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::set_config_(const uint8_t*, uint32_t)':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:65:24: error: 'bsec_set_configuration' was not declared in this scope
   this->bsec_status_ = bsec_set_configuration(config, len, work_buffer, sizeof(work_buffer));
                        ^~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/bme680_bsec/bme680_bsec.cpp:65:24: note: suggested alternative: 'bsec_set_configuration_m'
   this->bsec_status_ = bsec_set_configuration(config, len, work_buffer, sizeof(work_buffer));
                        ^~~~~~~~~~~~~~~~~~~~~~
                        bsec_set_configuration_m
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::update_subscription_()':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:130:7: error: 'bsec_update_subscription' was not declared in this scope
       bsec_update_subscription(virtual_sensors, num_virtual_sensors, sensor_settings, &num_sensor_settings);
       ^~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/bme680_bsec/bme680_bsec.cpp:130:7: note: suggested alternative: 'bsec_update_subscription_m'
       bsec_update_subscription(virtual_sensors, num_virtual_sensors, sensor_settings, &num_sensor_settings);
       ^~~~~~~~~~~~~~~~~~~~~~~~
       bsec_update_subscription_m
[... 17 lines cut ...]
src/esphome/components/bme680_bsec/bme680_bsec.cpp:361:26: error: 'bsec_do_steps' was not declared in this scope
     this->bsec_status_ = bsec_do_steps(inputs, num_inputs, outputs, &num_outputs);
                          ^~~~~~~~~~~~~
src/esphome/components/bme680_bsec/bme680_bsec.cpp:361:26: note: suggested alternative: 'bsec_do_steps_m'
     this->bsec_status_ = bsec_do_steps(inputs, num_inputs, outputs, &num_outputs);
                          ^~~~~~~~~~~~~
                          bsec_do_steps_m
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::load_state_()':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:462:26: error: 'bsec_set_state' was not declared in this scope
     this->bsec_status_ = bsec_set_state(state, BSEC_MAX_STATE_BLOB_SIZE, work_buffer, sizeof(work_buffer));
[... cut ...]
       bsec_get_state(0, state, BSEC_MAX_STATE_BLOB_SIZE, work_buffer, BSEC_MAX_STATE_BLOB_SIZE, &num_serialized_state);
       ^~~~~~~~~~~~~~
src/esphome/components/bme680_bsec/bme680_bsec.cpp:482:7: note: suggested alternative: 'bsec_get_state_m'
       bsec_get_state(0, state, BSEC_MAX_STATE_BLOB_SIZE, work_buffer, BSEC_MAX_STATE_BLOB_SIZE, &num_serialized_state);
       ^~~~~~~~~~~~~~
       bsec_get_state_m
*** [.pioenvs/esp32-s3-devkit/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
========================= [FAILED] Took 70.92 seconds =========================