esphome/issues #4339: Flickering problem with AC Dimmer using a Robodyn module

Janux31, 2023-03-23:
### The problem

Hello,
I'm trying to use a Robodyn dimmer module with a LED bulb with ESP Home and Home Assistant.
The dimmable LED (Osram) is always flickring. I have tried many configurations without success.
**The same installation works fine with the "official" RBDdimmer library and the serial monitor example.**

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.0

### What platform are you using?

ESP32

### Board

AZDelivery Dev Kit v4

### Component causing the issue

ac_dimmer

### Example YAML snippet

```yaml
My yaml section:
output:
  - platform: ac_dimmer
    id: robotdyn
    gate_pin: GPIO32
    zero_cross_pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: yes
    min_power: 30%
    # init_with_half_cycle: False

light:
  - platform: monochromatic
    output: robotdyn
    name: "Lumière variable"
```

### Anything in the logs that might be useful for us?

```txt
No error in log file.
The problem is only flickering.
```

### Additional information

Note that the value of "init_with_half_cycle" is true in log file when this attribute is not set in the yaml. The doc explain that the default value is False but it seems to be True. 

I have trie INPUT and INPUT_PULLUP but there is no difference.

My installation is official HA OS and ESPHome add-on on RPi.

swbouman, 2023-04-20:
Any luck already? I’m facing the same problem suddenly after 1 year running the dimmer without a problem.

cesarpuig, 2024-01-20:
Hi @Janux31

Have you been able to make it work with a robodyn AC dimmer or another one?

Thank you