esphome/issues #4340: I2C timeout too short for PN532

tracestep, 2023-03-23:
### The problem

Connecting a PN532 NFC/RFID board via I2C to ESP32 (esp-idf) does not work. Sometimes it gets enumerated (0x24 address), but mostly not. Anyway, the PN532 component initialization never succeeds. On the very verbose logs, the bus scan returns "timeout" on 0x24, and timeouts later in initialization, even if some early communication succeeds. Reducing the bus frequency to 10kHz helps a little, but does not solve the problem. 

With a logic analyzer, I noticed PN532 does "clock stretching", up to at least 440µs. That, mentioned in the component datasheet, and within I2C specs, is longer than the default timeout on ESP32 (100µs), with esp-idf at least. It is possible that this issue affects other I2C peripherals as well. The I2C component does not provide a way to change that timeout. 

I was able to change the I2C component code to include a configuration option for changing the timeout on esp-idf to 1000µs, which solves the problem in my setup. I will make a pull-request linked to this issue.

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.03.1

### What platform are you using?

ESP32-IDF

### Board

W32-ETH-01

### Component causing the issue

i2c

### Example YAML snippet

```yaml
i2c:
  scl: GPIO32
  sda: GPIO33
  scan: True
  frequency: 50kHz

pn532_i2c:
  update_interval: 1s
```

### Anything in the logs that might be useful for us?

```txt
[VV][i2c.idf:173]: 0x23 TX 
[VV][i2c.idf:210]: TX to 23 failed: not acked
[VV][i2c.idf:173]: 0x24 TX 
[VV][i2c.idf:213]: TX to 24 failed: timeout
[VV][i2c.idf:173]: 0x25 TX 
[VV][i2c.idf:210]: TX to 25 failed: not acked
...
[C][pn532:018]: Setting up PN532...
[VV][i2c.idf:173]: 0x24 TX 0000FF02FED4022A00
[V][pn532:246]: Reading ACK...
[VV][i2c.idf:149]: 0x24 RX 01
[VV][i2c.idf:149]: 0x24 RX 010000FF00FF00
[V][pn532:257]: ACK valid: YES
[V][pn532_i2c:043]: Reading response
[VV][i2c.idf:149]: 0x24 RX 01
[VV][i2c.idf:149]: 0x24 RX 010000FF06FAD5
[V][pn532:262]: Sending NACK for retransmit
[VV][i2c.idf:173]: 0x24 TX 0000FFFF0000
[VV][i2c.idf:213]: TX to 24 failed: timeout
[V][pn532_i2c:049]: Reading response of length 5
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[VV][i2c.idf:149]: 0x24 RX 00
[V][pn532_i2c:032]: Timed out waiting for readiness from PN532!
[D][pn532_i2c:051]: No response data
[E][pn532:029]: Error getting version
[E][component:113]: Component pn532 was marked as failed.
```

### Additional information

Changing the timeout to 1000µs (probably less) solves the issue.

tracestep, 2023-07-27:
Just made a new commit to the pull-request

guillempages, 2024-01-10:
Any updates here? I've also encountered this issue, and I see that the proposed PR was closed without merging :-(

nicosans, 2024-01-24:
This option could interest me, as I bet I have encountered the same issue

closed 2024-04-09 by jesserockz