esphome/issues #4342: Display on M5Stack core no longer works (ili9342 / ili9xxx)

ax42, 2023-03-23:
### The problem

Following error shows soon after boot:
`[E][display:109]: Could not allocate buffer for display!`

This looks very similar to https://github.com/esphome/issues/issues/4315 except that the display previously worked on older versions of esphome.

I have tried both `M5STACK` and `ili9342` for the model. The documentation for the board is https://docs.m5stack.com/en/core/basic_v2.6

```
display:
  - platform: ili9xxx
    model: ili9342
    cs_pin: 14
    dc_pin: 27
    reset_pin: 33
```

The device works otherwise (buttons, sensors, wifi), just the screen is not working (which is somewhat important to my use case).

### Which version of ESPHome has the issue?

2023.3.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

M5Core (v1)

### Component causing the issue

display / ili9xxx

nielsnl68, 2023-03-23:
This error is a bit misleading, the ili9xxx tries to create an 16bit buffer first and when it fails (and that is what this message says) it will go back to a 8bit mode.

Just wondering did you also setup the backlight?

nielsnl68, 2023-03-23:
> This looks very similar to #4315 except that the display previously worked on older versions of esphome.

The issue with this issue referenced above is with a bigger screen which will not work without extra PSRAM.
We need to find the solution of your issue somewhere else.

Please show me the USB log from the startup of your device.

ax42, 2023-03-24:
Is this the USB log you are asking for?

```
[W][api.connection:083ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
E (236) psram: PSRAM ID read error: 0xffffffff
[I][logger:258]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:175]: Performing I2C bus recovery
[C][spi:023]: Setting up SPI bus...
[D][binary_sensor:034]: 'Study Display button a': Sending initial state OFF
[D][binary_sensor:034]: 'Study Display button b': Sending initial state OFF
[D][binary_sensor:034]: 'Study Display button c': Sending initial state OFF
[E][display:109]: Could not allocate buffer for display!
[D][binary_sensor:034]: 'Study Display Status': Sending initial state OFF
[D][text_sensor:067]: 'Study Display ESPHome version': Sending state '2023.3.0 Mar 23 2023, 22:33:38'
[D][text_sensor:067]: 'Study Display Mac Address': Sending state '24:6F:28:8F:23:08'
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: 24:6F:28:8F:23:08
```
Setting up the backlight led to:
 - The error still occurring
 - The display coming alive, but it's black on white, and mirrored

Logs with backlight:
```
[D][ota:316]: OTA in progress: 95.2%
[I][ota:349]: OTA update finished!
[I][app:134]: Rebooting safely...
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[W][wifi_esp32:494]: Event: Disconnected ssid='AntHeap' bssid=[redacted] reason='Association Leave'
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
E (412) psram: PSRAM ID read error: 0xffffffff
[I][logger:258]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:175]: Performing I2C bus recovery
[C][spi:023]: Setting up SPI bus...
[D][binary_sensor:034]: 'Study Display button a': Sending initial state OFF
[D][binary_sensor:034]: 'Study Display button b': Sending initial state OFF
[D][binary_sensor:034]: 'Study Display button c': Sending initial state OFF
[D][ledc.output:051]: Calculating resolution bit-depth for frequency 1000.000000
[D][ledc.output:056]: Resolution calculated as 16
[E][display:109]: Could not allocate buffer for display!
[C][light:036]: Setting up light 'Display Backlight'...
[D][light:035]: 'Display Backlight' Setting:
[D][light:040]:   Color mode: 
[D][light:046]:   State: ON
[D][light:084]:   Transition length: 1.0s
[D][binary_sensor:034]: 'Study Display Status': Sending initial state OFF
[D][text_sensor:067]: 'Study Display ESPHome version': Sending state '2023.3.0 Mar 24 2023, 13:30:00'
[D][text_sensor:067]: 'Study Display Mac Address': Sending state '24:6F:28:8F:23:08'
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: 24:6F:28:8F:23:08
```

ax42, 2023-03-24:
Followup (and for future reference), changing the model to M5Stack fixed the issue for me (once the backlight was added). There is still the error on boot (`Cannot allocate display buffer`) but everything functions fine.

Working YAML for m5stack core:

```
spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

# Define a monochromatic, dimmable light for the backlight
light:
  - platform: monochromatic
    output: gpio_32_backlight_pwm
    name: "Display Backlight"
    id: back_light
    restore_mode: ALWAYS_ON

display:
  - platform: ili9xxx
    model: M5STACK
    cs_pin: 14
    dc_pin: 27
    reset_pin: 33
    rotation: 0
```

nielsnl68, 2023-03-24:
What i miss in the above log is the Log dump of the M5STACK / display component.
When you can't find it could you try with `logging: very_verbose` and show me the whole log file?

thanks for your patience.

ax42, 2023-03-24:
very_verbose logfile attached -- thanks for looking into this and please let me know if I can help any more.

[logs.txt](https://github.com/esphome/issues/files/11065828/logs.txt)

nielsnl68, 2023-03-25:
> There is still the error on boot (`Cannot allocate display buffer`) but everything functions fine.

For now you can ignore this error. I will remove it for the next release

jbouwh, 2025-02-21:
I have this issue after updating to esphome 2025.2.0

jorblad, 2025-02-27:
I also have this or a very simliar issue after upgrading to 2025.2.0.