esphome/issues #4343: Unable to compile anything - sh: 1: xtensa-esp32-elf-g++: not found

workingmanrob, 2023-03-24:
### The problem

Compiling any new firmware fails with an error about missing files

`sh: 1: xtensa-esp32-elf-g++: not found`

I can find that file in the docker container though.
```
root@5c53de3b-esphome:/# find / -name xtensa-esp32-elf-g++
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++
```

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

all

### Example YAML snippet

```yaml
esphome:
  name: $hostname
  platform: ESP32
  board: esp32doit-devkit-v1
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/office-led-strip.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing office-led-strip (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 5.2.0
INFO Installing platformio/espressif32 @ 5.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Platform Manager: espressif32@5.2.0 has been installed!
INFO espressif32@5.2.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r2-patch3 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
INFO Installing platformio/framework-arduinoespressif32 @ ~3.20005.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32@3.20005.220925 has been installed!
INFO framework-arduinoespressif32@3.20005.220925 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40201.0
INFO Installing platformio/tool-esptoolpy @ ~1.40201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-esptoolpy@1.40201.0 has been installed!
INFO tool-esptoolpy@1.40201.0 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%          
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%          
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%          
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [/data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
```

### Additional information

I recently updated to a new kernel on my home assistant pi4.  I do use the official raspberry pi kernel.

`ii  raspberrypi-kernel                1:1.20230317-1                   armhf`

```
$ uname -r
6.1.19-v8+
$ uname -m
aarch64
```
6.x? OK...

Honestly I don't even know if I was running 64bit before or not but I feel like this kernel is the root cause.  Whether it is just something new in 6.x or if it's something 64 vs 32.  My home assistant supervisord / docker install is using 32bit containers.

I first noticed the problem with esp32 board but just had essentially the same failure on esp8266.

I have uninstalled and reinstalled the addon.  Rebooted the system multiple times.  Cleaned build folders.  Crossed every digit on my body and rattled a bunch of chicken bones... no difference.

Also checked the three similar but unfortunately not identical systems I have access to and none of them have this problem.  None of those systems have this new kernel though.

workingmanrob, 2023-03-24:
On a test system I have here which is now running current home assistant and esphome (2023.3.6 and 2023.3.1 respectively) 

```
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_connection.cpp.o
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_server.cpp.o
```
And we're cooking with gas again.

I did not do a system update yet, just home assistant.  And I have to assume I use to be running 32bit on my main system cause this.
```
$ uname -r
5.15.84-v7l+
$ uname -m
armv7l
```
¯\\\_(ツ)_/¯

Which version of Home Assistant?
2023.3.6

Which version of ESPHome?
2023.3.1

workingmanrob, 2023-03-24:
Short version..

Problem solved after adding:
```
$ tail -n 1 /boot/config.txt 
arm_64bit=0
```
Now it's back to
```
$ uname -r
6.1.19-v7l+
$ uname -m
armv7l
```
and my main machine is coming in hot once again - huzzah!
```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_connection.cpp.o
Compiling /data/office-led-strip/.pioenvs/office-led-strip/src/esphome/components/api/api_frame_helper.cpp.o
```

closed 2023-03-24 by workingmanrob

AdmiralStipe, 2023-04-22:
Thx for the tip, worked perfectly in my case also :).