esphome/issues #4344: Failure to compile due to C header file

aaronpowell, 2023-03-24:
### The problem

I went to install an update to one of my ESP8266 devices but when it starts compiling the payload there's an error that starts with:

```
src/esphome/core/helpers.h:550:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
```

Here's the full log:

```
INFO Reading configuration /config/esphome/1921681179.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing aaron-office-background (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- FastLED @ 3.3.2
Compiling /data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_connection.cpp.o
Compiling /data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:550:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:550:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/api_frame_helper.cpp:5:0:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:550:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:550:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_connection.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/aaron-office-background/.pioenvs/aaron-office-background/src/esphome/components/api/api_pb2.cpp.o] Error 1
========================== [FAILED] Took 7.22 seconds ==========================
```

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

core or FastLED

### Example YAML snippet

```yaml
esphome:
  name: aaron-office-background

esp8266:
  board: nodemcuv2
  framework:
    version: 2.7.4

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<snip>"

ota:
  password: "<snip>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "<snip>"
    password: "<snip>"

captive_portal:

light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: D2
    num_leds: 100
    rgb_order: GRB
    name: "Aaron's office background"
    effects:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 50
      - addressable_color_wipe:
          name: Color Wipe Effect With Custom Values
          colors:
            - red: 100%
              green: 100%
              blue: 100%
              num_leds: 1
            - red: 0%
              green: 0%
              blue: 0%
              num_leds: 1
          add_led_interval: 100ms
          reverse: false
```

### Anything in the logs that might be useful for us?

```txt
I updated HA Core (from 2023.3.5) yesterday, and ESPHome update was 2 days ago (from 2023.3.0), and I hadn't installed any updates on this board for quite a while so it was possible a few "revisions" behind.
```

aaronpowell, 2023-03-24:
Doing some more testing, it looks like the problem is using the `board` of `nodemcuv2` and the legacy framework together that's the problem.

petergr20, 2023-03-25:
Same issue on board esp01_1m

[10:07:51][I][app:102]: ESPHome version 2023.2.4 compiled on Feb 28 2023, 17:23:39 --> OK

When try to compile with 

[10:08:29][I][app:102]: ESPHome version 2023.3.1 compiled on Mar 25 2023, 10:01:22 --> KO

bonzai86, 2023-03-25:
I have the same issue on 2023.3.1 (introducing https://github.com/esphome/esphome/pull/4602)
Hope we're not out of luck, because we're using old arduino version due support of fastled.

sparkydave1981, 2023-03-28:
The same problem has been reported in [this thread](https://github.com/esphome/issues/issues/3084):

https://github.com/esphome/issues/issues/3084#issuecomment-1473933233

sparkydave1981, 2023-04-03:
I just tested out trying to compile the code for one of my devices which uses the older framework  on ESPhome version 2023.3.2 and it compiled without error.

```
esp8266:
  board: esp01_1m
  framework:
    version: 2.7.4
```

petergr20, 2023-04-03:
with esphome 2023.3.2 it's ok

thx a lot

bonzai86, 2023-04-04:
It is compiling again for me too. Well done.

jesserockz, 2023-04-05:
This was fixed with https://github.com/esphome/esphome/pull/4610 and released in 2023.3.2

closed 2023-04-05 by jesserockz

mhlavac, 2023-04-27:
@jesserockz huge thanks for fixing the issue. It looks like this is broken again with `2023.4.0` release and the following releases. For now reverting to `2023.3.2` works like a charm.

Here is the compile errors:

```
In file included from src/esphome/core/entity_base.h:5:0,
                 from src/esphome/core/entity_base.cpp:1:
src/esphome/core/string_ref.h:25:21: error: expected type-specifier
   using size_type = std::allocator_traits<allocator_type>::size_type;
                     ^
src/esphome/core/string_ref.h:26:27: error: expected type-specifier
   using difference_type = std::allocator_traits<allocator_type>::difference_type;
                           ^
src/esphome/core/string_ref.h:68:13: error: 'size_type' does not name a type
   constexpr size_type size() const { return len_; }
             ^
src/esphome/core/string_ref.h:70:40: error: 'size_type' has not been declared
   constexpr const_reference operator[](size_type pos) const { return *(base_ + pos); }
                                        ^
src/esphome/core/string_ref.h:79:3: error: 'size_type' does not name a type
   size_type len_;
   ^
src/esphome/core/string_ref.h: In constructor 'constexpr esphome::StringRef::StringRef()':
src/esphome/core/string_ref.h:32:38: error: class 'esphome::StringRef' does not have any field named 'len_'
   constexpr StringRef() : base_(""), len_(0) {}
                                      ^
src/esphome/core/string_ref.h: In constructor 'esphome::StringRef::StringRef(const string&)':
src/esphome/core/string_ref.h:33:64: error: class 'esphome::StringRef' does not have any field named 'len_'
   explicit StringRef(const std::string &s) : base_(s.c_str()), len_(s.size()) {}
                                                                ^
src/esphome/core/string_ref.h: In constructor 'esphome::StringRef::StringRef(const char*)':
src/esphome/core/string_ref.h:34:49: error: class 'esphome::StringRef' does not have any field named 'len_'
   explicit StringRef(const char *s) : base_(s), len_(strlen(s)) {}
                                                 ^
src/esphome/core/string_ref.h: In constructor 'constexpr esphome::StringRef::StringRef(const char*, size_t)':
src/esphome/core/string_ref.h:35:60: error: class 'esphome::StringRef' does not have any field named 'len_'
   constexpr StringRef(const char *s, size_t n) : base_(s), len_(n) {}
                                                            ^
src/esphome/core/string_ref.h: In constructor 'constexpr esphome::StringRef::StringRef(const CharT*, size_t)':
src/esphome/core/string_ref.h:37:93: error: class 'esphome::StringRef' does not have any field named 'len_'
   constexpr StringRef(const CharT *s, size_t n) : base_(reinterpret_cast<const char *>(s)), len_(n) {}
                                                                                             ^
src/esphome/core/string_ref.h: In constructor 'esphome::StringRef::StringRef(InputIt, InputIt)':
src/esphome/core/string_ref.h:40:57: error: class 'esphome::StringRef' does not have any field named 'len_'
       : base_(reinterpret_cast<const char *>(&*first)), len_(std::distance(first, last)) {}
                                                         ^
src/esphome/core/string_ref.h: In constructor 'esphome::StringRef::StringRef(InputIt*, InputIt*)':
src/esphome/core/string_ref.h:43:55: error: class 'esphome::StringRef' does not have any field named 'len_'
       : base_(reinterpret_cast<const char *>(first)), len_(std::distance(first, last)) {}
                                                       ^
src/esphome/core/string_ref.h: In member function 'constexpr const value_type* esphome::StringRef::end() const':
src/esphome/core/string_ref.h:58:57: error: 'len_' was not declared in this scope
   constexpr const_iterator end() const { return base_ + len_; };
                                                         ^
src/esphome/core/string_ref.h:58:63: error: body of constexpr function 'constexpr const value_type* esphome::StringRef::end() const' not a return-statement
   constexpr const_iterator end() const { return base_ + len_; };
                                                               ^
src/esphome/core/string_ref.h: In member function 'constexpr const value_type* esphome::StringRef::cend() const':
src/esphome/core/string_ref.h:59:58: error: 'len_' was not declared in this scope
   constexpr const_iterator cend() const { return base_ + len_; };
                                                          ^
src/esphome/core/string_ref.h:59:64: error: body of constexpr function 'constexpr const value_type* esphome::StringRef::cend() const' not a return-statement
   constexpr const_iterator cend() const { return base_ + len_; };
                                                                ^
src/esphome/core/string_ref.h: In member function 'esphome::StringRef::const_reverse_iterator esphome::StringRef::rbegin() const':
src/esphome/core/string_ref.h:61:81: error: 'len_' was not declared in this scope
   const_reverse_iterator rbegin() const { return const_reverse_iterator{base_ + len_}; }
                                                                                 ^
src/esphome/core/string_ref.h:61:85: error: no matching function for call to 'std::reverse_iterator<const char*>::reverse_iterator(<brace-enclosed initializer list>)'
   const_reverse_iterator rbegin() const { return const_reverse_iterator{base_ + len_}; }
[... 229 lines cut ...]
[... cut ...]
src/esphome/core/string_ref.h: In member function 'constexpr bool esphome::StringRef::empty() const':
src/esphome/core/string_ref.h:69:52: warning: control reaches end of non-void function [-Wreturn-type]
   constexpr bool empty() const { return len_ == 0; }
                                                    ^
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/artur-bed-light/src/esphome/core/entity_base.cpp.o] Error 1
```

ssieb, 2023-04-27:
@mhlavac that's a different error than this issue, so you need to file a new one.  But if you're using an old Arduino version, it's not an esphome issue anyway.

sparkydave1981, 2023-04-27:
I removed the Arduino framework and my light compiled and seems to work fine. I will try to confirm full functionality soon